pikvm/ustreamer #100: Syslog Entry for Connected/Disconnected Clients

jdolence999, 2021-03-19:
Presently ustreamer creates a syslog entry for connected and disconnected clients such as the one below for a connection:

ustreamer[4330]: -- INFO  [194149.051      http] -- HTTP: Registered client: [127.0.0.1]:41242, id=7740de1d7d0b8a8f; clients now: 1
 
I am running ustreamer on the local host as seen in the log entry.  I have a couple of programs that act as a reverse proxy for client connections but all client connections will log with the localhost IP address.  I'd like to request a modification for the ustreamer code to look for a pre-defined header such as the industry standard x-forwarded-for or a user defined header value (e.g. true-client-ip, actual-ip, etc.) and use the associated IP address in the log entry?  if no header exists, ustreamer should log the connecting source ip as it does today.

mdevaev, 2021-03-19:
Okay, I'll do it.

mdevaev, 2021-03-21:
Done.

closed 2021-03-21 by mdevaev

jdolence999, 2021-03-22:
Awesome!  Thanks!  I'll check it out as soon as I get a chance!