pikvm/ustreamer #102: webrtc-streamer

engelant, 2021-04-01:
I don't know if this already came up, but you may want to have a look at this [Rpi WebRTC Streamer Demo Video](https://www.youtube.com/watch?v=I1E8MrA5lhw), more precice the link to the [github repo](https://github.com/kclyu/rpi-webrtc-streamer) in the description.

I don't know if/how well it works on a PI4, but it leverages the hardware h264 encoder and gives 1080p@30fps without a problem.
I tried it out in my octoprint, running on an older pi2 or 3 (can't remember which one it was), disabeling the "normal" mjpeg.
And it's like really nice.

But as always with a grain of salt, at least when I tested it:
 * unable to establish more than a single connection at the time
 * video sometimes freezes and needs manual reconnect

Although, this doesn't seem too hard of an issue, as the underlying tech (webrtc, I use it a lot with meet.jit.si) seems quiet mature and in the worst case it needs a reconnect button in the ui and in the best case some js logic will catch the broken connection.

mdevaev, 2021-04-01:
In fact, ustreamer can compress an h264 stream with minimal latency, and it can be routed somewhere using ustreamer-dump. Or read from shared memory. I am currently writing a plugin for janus-gateway. In the first approach, it works very well. With many clients also.

mdevaev, 2021-09-09:
It is done and works perfectly. Unfortunately, I don't have much time to document this, I can only offer links to an example configuration from Janus. I hope this helps.
- https://github.com/pikvm/ustreamer/tree/master/janus
- https://github.com/pikvm/kvmd/tree/master/configs/janus

closed 2021-09-09 by mdevaev