pikvm/ustreamer #103: ./ustreamer: No such file or directory

Friday8229, 2021-04-02:
I'm getting this error after running the make file.

pi@raspberrypi:~/ustreamer $ ./ustreamer --help
bash: ./ustreamer: No such file or directory

What i did:

Installed all the packages:
sudo apt install libevent-dev libjpeg8-dev libbsd-dev libraspberrypi-dev libgpiod

Cloned the GitHub Repo:
git clone --depth=1 https://github.com/pikvm/ustreamer

Ran these commands:
 cd ustreamer

 Output of "make" command:

pi@raspberrypi:~/ustreamer $ sudo make
make apps
make[1]: Entering directory '/home/pi/ustreamer'
make -C src
make[2]: Entering directory '/home/pi/ustreamer/src'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/pi/ustreamer/src'
make[1]: Leaving directory '/home/pi/ustreamer'
make python
make[1]: Entering directory '/home/pi/ustreamer'
make[1]: Leaving directory '/home/pi/ustreamer'

Now, when i run  "./ustreamer --help" I get this error.

pi@raspberrypi:~/ustreamer $ ./ustreamer --help
bash: ./ustreamer: No such file or directory

What am i doing wrong?

krasin, 2021-04-02:
Can you please post the output of the following commands?

```
ls -l
ldd ./ustreamer
```

Friday8229, 2021-04-02:
Thanks for the quick response. Here are the outputs.

`pi@raspberrypi:~/ustreamer $ ls -l
total 92
-rw-r--r-- 1 pi   pi   35147 Apr  2 07:32 LICENSE
drwxr-xr-x 2 pi   pi    4096 Apr  2 07:32 linters
-rw-r--r-- 1 pi   pi    1930 Apr  2 07:32 Makefile
drwxr-xr-x 2 pi   pi    4096 Apr  2 07:32 man
drwxr-xr-x 6 pi   pi    4096 Apr  2 07:32 pkg
drwxr-xr-x 3 pi   pi    4096 Apr  2 07:58 python
-rw-r--r-- 1 pi   pi    9219 Apr  2 07:32 README.md
-rw-r--r-- 1 pi   pi   14265 Apr  2 07:32 README.ru.md
drwxr-xr-x 6 pi   pi    4096 Apr  2 07:34 src
drwxr-xr-x 2 pi   pi    4096 Apr  2 07:32 tools
lrwxrwxrwx 1 root root    17 Apr  2 08:06 ustreamer.bin -> src/ustreamer.bin
lrwxrwxrwx 1 root root    22 Apr  2 08:06 ustreamer-dump.bin -> src/ustreamer-dump.bin`

Running the command from inside the ustreamer folder:
`pi@raspberrypi:~/ustreamer $ ldd ./ustreamer
ldd: ./ustreamer: No such file or directory`

Running the command from outside the ustreamer folder:

`pi@raspberrypi:~ $ ldd ./ustreamer
ldd: ./ustreamer: not regular file`

krasin, 2021-04-02:
Full disclosure: I am not a contributor to this project. Take my advice as a drive-by comment of a curious bystander.

It appears to me that the program name is ustreamer.bin. Can you please try to run it as ```./ustreamer.bin``` and report the results?

Friday8229, 2021-04-02:
Thanks for all your support and quick help, it's working great now.

> pi@raspberrypi:~/ustreamer $ ./ustreamer.bin --help

uStreamer - Lightweight and fast MJPG-HTTP streamer
═══════════════════════════════════════════════════
Version: 3.23; license: GPLv3
Copyright (C) 2018-2021 Maxim Devaev <mdevaev@gmail.com>

Capturing options:
══════════════════
    -d|--device </dev/path>  ───────────── Path to V4L2 device. Default: /dev/video0.

    -i|--input <N>  ────────────────────── Input channel. Default: 0.

    -r|--resolution <WxH>  ─────────────── Initial image resolution. Default: 640x480.

    -m|--format <fmt>  ─────────────────── Image format.
                                           Available: YUYV, UYVY, RGB565, RGB24, MJPEG, JPEG; default: YUYV.

    -a|--tv-standard <std>  ────────────── Force TV standard.
                                           Available: PAL, NTSC, SECAM; default: disabled.

    -I|--io-method <method>  ───────────── Set V4L2 IO method (see kernel documentation).
                                           Changing of this parameter may increase the performance. Or not.
                                           Available: MMAP, USERPTR; default: MMAP.

    -f|--desired-fps <N>  ──────────────── Desired FPS. Default: maximum possible.

    -z|--min-frame-size <N>  ───────────── Drop frames smaller then this limit. Useful if the device
                                           produces small-sized garbage frames. Default: 128 bytes.

    -n|--persistent  ───────────────────── Don't re-initialize device on timeout. Default: disabled.

    -t|--dv-timings  ───────────────────── Enable DV timings querying and events processing
                                           to automatic resolution change. Default: disabled.

    -b|--buffers <N>  ──────────────────── The number of buffers to receive data from the device.
                                           Each buffer may processed using an independent thread.
                                           Default: 5 (the number of CPU cores (but not more than 4) + 1).

    -w|--workers <N>  ──────────────────── The number of worker threads but not more than buffers.
                                           Default: 4 (the number of CPU cores (but not more than 4)).

    -q|--quality <N>  ──────────────────── Set quality of JPEG encoding from 1 to 100 (best). Default: 80.
                                           Note: If HW encoding is used (JPEG source format selected),
                                           this parameter attempts to configure the camera
                                           or capture device hardware's internal encoder.
                                           It does not re-encode MJPG to MJPG to change the quality level
                                           for sources that already output MJPG.

    -c|--encoder <type>  ───────────────── Use specified encoder. It may affect the number of workers.
                                           Available:
                                             * CPU  ── Software MJPG encoding (default);
                                             * HW  ─── Use pre-encoded MJPG frames directly from camera hardware.
                                             * NOOP  ─ Don't compress MJPG stream (do nothing).

    -k|--blank <path>  ─────────────────── Path to JPEG file that will be shown when the device is disconnected
                                           during the streaming. Default: black screen 640x480 with 'NO SIGNAL'.

    -K|--last-as-blank <sec>  ──────────── Show the last frame received from the camera after it was disconnected,
                                           but no more than specified time (or endlessly if 0 is specified).
                                           If the device has not yet been online, display 'NO SIGNAL' or the image
                                           specified by option --blank. Default: disabled.
                                           Note: currently this option has no effect on memory sinks.

    -l|--slowdown  ─────────────────────── Slowdown capturing to 1 FPS or less when no stream or sink clients
                                           are connected. Useful to reduce CPU consumption. Default: disabled.

    --device-timeout <sec>  ────────────── Timeout for device querying. Default: 1.
[... 112 lines cut ...]

Logging options:
════════════════
    --log-level <N>  ──── Verbosity level of messages from 0 (info) to 3 (debug).
                          Enabling debugging messages can slow down the program.
                          Available levels: 0 (info), 1 (performance), 2 (verbose), 3 (debug).
                          Default: 0.

    --perf  ───────────── Enable performance messages (same as --log-level=1). Default: disabled.

    --verbose  ────────── Enable verbose messages and lower (same as --log-level=2). Default: disabled.

    --debug  ──────────── Enable debug messages and lower (same as --log-level=3). Default: disabled.

    --force-log-colors  ─ Force color logging. Default: colored if stderr is a TTY.

    --no-log-colors  ──── Disable color logging. Default: ditto.

Help options:
═════════════
    -h|--help  ─────── Print this text and exit.

    -v|--version  ──── Print version and exit.

    --features  ────── Print list of supported features.

krasin, 2021-04-02:
Glad to be helpful!

@mdevaev this issue can be closed, I believe.

mdevaev, 2021-04-02:
I was rewriting Makefiles and this is a small flaw. It is better that the files are without the .bin suffix in the root directory. Fixed it in master. Thanks!

commit dd6dc866 (2021-04-02): Fixed #103: symlinks without .bin suffix
closed 2021-04-02 by mdevaev