pikvm/ustreamer #104: Is there a better way to fed h264 stream with rtsp?

xmmwc, 2021-04-02:
at beginning of using ustreamer.I just pull mjpeg stream from RASPI to NAS.but it's very large.
I really like the feature that slowdown capturing to 1fps when no client.it's very useful for me.
I tried to look through the code difficult yesterday(I'm FE programmer and not good at English) .I found "h264" in code. so after hours tossing.
I got this solution:
camera(HW MJPEG) => ustreamer H264 sink => ffmpeg copy => v4l2loopback => v4l2rtspserver
but it can't use slowdown feature and keep capturing when NAS not recording.
Is there a better way to fed h264 stream with rtsp
btw ustreamer is amazing!

mdevaev, 2021-04-02:
Try connecting the ustreamer-dump output via pipe to the gstreamer input. This way you can avoid v4l2loopback and unnecessary copying. Unfortunately, slowdown doesn't work for h264 yet (but it will).

mdevaev, 2021-04-02:
UPD: I lied, slowdown is also working for h264.

mdevaev, 2021-05-27:
I think I have answered these questions. Another related task: #102.

closed 2021-05-27 by mdevaev

xmmwc, 2021-08-23:
What about h264 compressed stream from v4l2 device?
I just want stream video from device(h264 compressed) when service is turn on, that camera not capturing when service is off. or slowdown capturing when no client connected. slowdown capturing is so nice that i can't not use it.

Is ustreamer just aim to handle MJPG video?
I'm not sure, if ustreamer accept h264 compressed video stream(if supported) can small the size of files that client side save?(I don't know much about video codec things)
Any suggestion?

mdevaev, 2021-08-23:
This is not possible now. Ustreamer can accept either raw or jpeg as input.