pikvm/ustreamer #106: 3.24: warning: 'WITH_PTHREAD_NP' macro redefined ; warning: implicit declaration of function 'strerror_l' is invalid in C99

yurivict, 2021-04-06:
Build fails:
```
-- CC ustreamer/blank.c
In file included from <built-in>:339:
<command line>:3:9: warning: 'WITH_PTHREAD_NP' macro redefined [-Wmacro-redefined]
In file included from <built-in>:#define WITH_PTHREAD_NP 1In file included from In file included from In file included from <built-in>339
:<built-in>:
        ^:
In file included from 339<command line>:
<built-in>:<command line>339<built-in>:::3<command line>339:2:
:9<command line>3:339::
:
:<command line>:93<command line>9:::   :9warning3:::3note : 9warning:'WITH_PTHREAD_NP' macro redefined [-Wmacro-redefined]:: 9: 
 warning#define WITH_PTHREAD_NP 1:warningprevious definition is here'WITH_PTHREAD_NP' macro redefined [-Wmacro-redefined]: : 

 #define WITH_PTHREAD_NP 1
warning'WITH_PTHREAD_NP' macro redefined [-Wmacro-redefined]        ^#define WITH_PTHREAD_NP 0
'WITH_PTHREAD_NP' macro redefined [-Wmacro-redefined]

#define WITH_PTHREAD_NP 1
<command line>        ^: 
#define WITH_PTHREAD_NP 1
        ^:'WITH_PTHREAD_NP' macro redefined [-Wmacro-redefined]

2        ^
        ^:
<command line>#define WITH_PTHREAD_NP 19<command line>:

:2        ^:<command line>
:2: 9<command line>note2:: ::9previous definition is here :9:note:2
 :  :#define WITH_PTHREAD_NP 0note9note
: ::         ^ previous definition is hereprevious definition is here

previous definition is herenote
#define WITH_PTHREAD_NP 0
: 
#define WITH_PTHREAD_NP 0#define WITH_PTHREAD_NP 0previous definition is here        ^

        ^        ^#define WITH_PTHREAD_NP 0

        ^
In file included from libs/base64.c:23:
In file included from libs/base64.h:31:
libs/tools.h:156:25: warning: implicit declaration of function 'strerror_l' is invalid in C99 [-Wimplicit-function-declaration]
        strncpy(buf, (locale ? strerror_l(error, locale) : str), size - 1);
                               ^
libs/tools.h:156:23: warning: pointer/integer type mismatch in conditional expression ('int' and 'char *') [-Wconditional-type-mismatch]
        strncpy(buf, (locale ? strerror_l(error, locale) : str), size - 1);
                             ^ ~~~~~~~~~~~~~~~~~~~~~~~~~   ~~~
In file included from libs/logging.c:23:
In file included from libs/logging.h:36:
libs/tools.h:156:25: warning: implicit declaration of function 'strerror_l' is invalid in C99 [-Wimplicit-function-declaration]
        strncpy(buf, (locale ? strerror_l(error, locale) : str), size - 1);
                               ^
libs/tools.h:156:23: warning: pointer/integer type mismatch in conditional expression ('int' and 'char *') [-Wconditional-type-mismatch]
        strncpy(buf, (locale ? strerror_l(error, locale) : str), size - 1);
                             ^ ~~~~~~~~~~~~~~~~~~~~~~~~~   ~~~
In file included from libs/memsink.c:23:
In file included from libs/memsink.h:37:
libs/tools.h:156:25: warning: implicit declaration of function 'strerror_l' is invalid in C99 [-Wimplicit-function-declaration]
        strncpy(buf, (locale ? strerror_l(error, locale) : str), size - 1);
                               ^
libs/tools.h:156:23: warning: pointer/integer type mismatch in conditional expression ('int' and 'char *') [-Wconditional-type-mismatch]
        strncpy(buf, (locale ? strerror_l(error, locale) : str), size - 1);
                             ^ ~~~~~~~~~~~~~~~~~~~~~~~~~   ~~~
3 warnings generated.
-- CC ustreamer/data/blank_jpeg.c
In file included from <built-in>:339:
<command line>:3:9: warning: 'WITH_PTHREAD_NP' macro redefined [-Wmacro-redefined]
#define WITH_PTHREAD_NP 1
[... 343 lines cut ...]
>>>               build/libs/memsink.o:(memsink_init)
>>> referenced by memsink.c
>>>               build/libs/memsink.o:(memsink_destroy)
>>> referenced by memsink.c
>>>               build/libs/memsink.o:(memsink_destroy)
>>> referenced by memsink.c
>>>               build/libs/memsink.o:(memsink_destroy)
>>> referenced by memsink.c
>>>               build/libs/memsink.o:(memsink_server_check)
>>> referenced by memsink.c
>>>               build/libs/memsink.o:(memsink_server_check)
>>> referenced 4 more times
>>> did you mean: strerror_r
>>> defined in: /lib/libc.so.7
cc: error: linker command failed with exit code 1 (use -v to see invocation)
make[3]: *** [Makefile:99: ustreamer-dump.bin] Error 1
make[3]: *** Waiting for unfinished jobs....
3 warnings generated.
make[3]: Leaving directory '/wrkdirs/usr/ports/multimedia/ustreamer/work/ustreamer-3.24/src'
make[2]: *** [Makefile:37: apps] Error 2
make[2]: Leaving directory '/wrkdirs/usr/ports/multimedia/ustreamer/work/ustreamer-3.24'
gmake[1]: *** [Makefile:28: all] Error 2
gmake[1]: Leaving directory '/wrkdirs/usr/ports/multimedia/ustreamer/work/ustreamer-3.24'
===> Compilation failed unexpectedly.
```

mdevaev, 2021-04-06:
I already told you that you are using build options incorrectly. You override macros, but you need to override Makefile variables. The code does not check the value of the WITH_PTHREAD_NP macro, but its existence, these are fundamentally different things:
```c
#ifdef WITH_PTHREAD_NP
// NOT THE SAME AS
#if WITH_PTHREAD_NP
```
The first statement does not care what is inside the macro, it checks whether it is defined in any way. This is the usual way to use macros, for example, this is how _GNU_SOURCE works and the many libraries I use.

Passing parameters to make enables or disables the definition of these macros. If you do `make WITH_PTHREAD_NP=1`, then `-DWITH_PTHREAD_NP` will be added to CFLAGS **without any value**. By default, this parameter is enabled. To disable it, pass 0: `make WITH_PTHREAD_NP=0`.

You passed **second define option** that overrides a macro that was already passed to make: `#define WITH_PTHREAD_NP 0`. The macro value doesn't make sense. It is not checked. You don't need to do this, just pass the parameter to make when called.

How to fix it:
1. Stop doing [this](https://svnweb.freebsd.org/ports/head/multimedia/ustreamer/Makefile?view=markup#l22).
2. Add [MAKE_ARGS](https://svnweb.freebsd.org/ports/head/Mk/bsd.port.mk?revision=523641&view=markup#l791) to the port's Makefile: `MAKE_ARGS="WITH_PTHREAD_NP=0`.

---
The second thig with `strerror_l()` is really bug, I'll fix it.

yurivict, 2021-04-06:
Sorry, I must have overlooked that when you told me this about ```MAKE_ARGS="WITH_PTHREAD_NP```.

What about the error with ```strerror_l```? ```#include <string.h>``` seems to be missing in ```src/libs/tools.h```.

mdevaev, 2021-04-06:
It was [here](https://github.com/pikvm/ustreamer/issues/12).

string.h is in place. It seems to be a [FreeBSD bug](https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=251651), and it was you who reported it :) What's the best thing to do here?

yurivict, 2021-04-06:
I see, this function has been added last year but my jail was too late and didn't have the fix yet.

Thank you for your help!

closed 2021-04-06 by yurivict

mdevaev, 2021-04-06:
Let me know if there are any problems

mdevaev, 2021-04-06:
Made a new release for this fix.