pikvm/ustreamer #108: octoprint interface ,output_http.so plugin

benjjyman, 2021-04-18:
please disregard my previous issues, it appears I was looking at a different octoprint server on another system.

reading the docs i see ustreamer is capable of interfacing with output_http.so plugin, but for all my inexperience cant figure out how.

the window in the webpage shows a link which will open another tab with the stream, but I cant display the stream in the window, I believe its how octoprint uses the output_http.so plugin, Im only just learning to write plugins for octoprint and dont have the necessary skills yet to change this, so if anyone could help explain how to get ustreamer to interface with output_http.so plugin i would be very greatfull.

mdevaev, 2021-04-27:
Ustreamer can't interact with output_http.so, it has own http server. Sorry, I don't understand what you're doing :(

benjjyman, 2021-04-27:
I was trying to setup an instance of octoprint for monitoring a 3d printer using a low powered orangepi zero running armbian (raspberry pi ulternative) which I think is debian based. Any way the original uses mjpeg streamer which is resource hungry on my little 512mb board. So I used ustreamer inplace, I didn't understand what output_http. So was as I have very little real world coding experience, so I got stuck for a couple of days. But I have the system working just fine.

If you like I can list the changes to get the two to interface on armbian?

Thanks for sharing your hard work

mdevaev, 2021-05-27:
OMG I completely forgot to answer you, I'm sorry. Yes, list the necessary things, I'll try to find the time to do something.

panic2k, 2022-11-25:
@benjjyman I had nearly no troubles to setup ustreamer for mainsail as ustreamer serves a mjpeg stream with http. I written a systemd service file after trying out a a little bit. I'm sure this works for Octoprint too.

[Unit]
Description=ustream server
After=network-online.target

[Service]
Type=simple
User=myusername
RemainAfterExit=yes
WorkingDirectory=/home/myusername/ustreamer
ExecStart=/home/myusername/ustreamer/ustreamer -d /dev/video1 -f 15 -l -s 127.0.0.1 -p 8123 -q 90 --allow-origin "*"
Restart=always
RestartSec=30

[Install]
WantedBy=multi-user.target

My stream can be accessed at http://127.0.0.1:8123/stream

Maybe you could use the ExecStart command to figure out your needed command line
Find your device by "v4l2-ctl --list-devices"
then run the modified ExecStart command on command line with your device instead and maybe -s changed to 0.0.0.0 so you could test the stream from any other host.