pikvm/ustreamer #110: Stream crashes

vgropp, 2021-04-27:
Hi,

i see stream crashes when using a `PCI bridge: Broadcom Inc. and subsidiaries BCM2711 PCIe Bridge` with ustreamer on a rpi 4, the BCM2711 is connected via HDMI to a ASRock Z370 Professional Gaming i7  with i915 module loaded and used to a framebuffer console.

Starting and watching `ustreamer --device=/dev/video0 -r 1280x720 -f 30 -m UYVY --host 0.0.0.0 --port 8888 --device-timeout=5 --debug 2>&1` in chrome results in multiple crashes every 5 seconds

Do you have any advice what to investigate/change further when it occurs again? The rpi was working fine with other systems and is again on this system without any reboot or changes.

Debug log from ustreamer:
```
-- DEBUG [424.188    stream] -- Waiting for worker ...
-- DEBUG [424.188    stream] -- Calling select() on video device ...
-- VERB  [424.188      jw-1] -- Compressed new JPEG: size=180507, time=0.133, worker=jw-1, buffer=0
-- DEBUG [424.188      jw-1] -- Worker jw-1 waiting for a new job ...
-- VERB  [424.235      jw-0] -- Compressed new JPEG: size=181108, time=0.130, worker=jw-0, buffer=3
-- DEBUG [424.235      jw-0] -- Worker jw-0 waiting for a new job ...
-- VERB  [424.279      jw-2] -- Compressed new JPEG: size=171831, time=0.124, worker=jw-2, buffer=4
-- DEBUG [424.279      jw-2] -- Worker jw-2 waiting for a new job ...
-- DEBUG [425.188      http] -- HTTP: Repeating exposed ...
-- DEBUG [426.204      http] -- HTTP: Repeating exposed ...
-- DEBUG [427.206      http] -- HTTP: Repeating exposed ...
-- DEBUG [428.214      http] -- HTTP: Repeating exposed ...
-- DEBUG [429.190    stream] -- Device select() --> 0
-- ERROR [429.190    stream] -- Mainloop select() error: Inappropriate ioctl for device
-- INFO  [429.190    stream] -- Destroying workers pool JPEG ...
-- DEBUG [429.190      jw-0] -- Bye-bye (worker jw-0)
-- DEBUG [429.191      jw-1] -- Bye-bye (worker jw-1)
-- DEBUG [429.191      jw-2] -- Bye-bye (worker jw-2)
-- DEBUG [429.191      jw-3] -- Bye-bye (worker jw-3)
-- DEBUG [429.191    stream] -- Calling ioctl(VIDIOC_STREAMOFF) ...
-- DEBUG [429.217      http] -- HTTP: Repeating exposed ...
-- INFO  [429.481    stream] -- Capturing stopped
-- DEBUG [429.481    stream] -- Releasing device buffers ...
-- DEBUG [429.481    stream] -- Closing device ...
-- INFO  [429.482    stream] -- Device fd=8 closed
-- DEBUG [429.482    stream] -- _stream_init_loop: stream->run->stop=0
-- INFO  [429.482    stream] -- Changed video frame to BLANK
```

dmesg:
```
[  429.232319] ------------[ cut here ]------------
[  429.238234] WARNING: CPU: 2 PID: 865 at drivers/media/common/videobuf2/videobuf2-core.c:1945 __vb2_queue_cancel+0x1d0/0x24c [videobuf2_common]
[  429.253514] Modules linked in: usb_f_mass_storage usb_f_hid btsdio bluetooth ecdh_generic ecc brcmfmac brcmutil cfg80211 raspberrypi_hwmon bcm2835_codec(C) i2c_mux_pinctrl bcm2835_v4l2(C) bcm2835_isp(C) bcm2835_unicam i2c_mux v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 i2c_bcm2835 videobuf2_common iproc_rng200 rng_core rpivid_mem nvmem_rmem uio_pdrv_genirq uio sch_fq_codel snd_bcm2835(C) snd_pcm snd_timer snd tc358743 v4l2_dv_timings v4l2_fwnode videodev mc cec libcomposite dwc2 udc_core roles fuse drm drm_panel_orientation_quirks ip_tables x_tables ipv6
[  429.315371] CPU: 2 PID: 865 Comm: stream Tainted: G        WC        5.10.23-2-ARCH #1
[  429.324647] Hardware name: BCM2711
[  429.328727] [<c020f10c>] (unwind_backtrace) from [<c020aed0>] (show_stack+0x10/0x14)
[  429.337805] [<c020aed0>] (show_stack) from [<c0de95c4>] (dump_stack+0xa4/0xc4)
[  429.346370] [<c0de95c4>] (dump_stack) from [<c022ae9c>] (__warn+0xec/0x148)
[  429.354010] [<c022ae9c>] (__warn) from [<c0de1868>] (warn_slowpath_fmt+0x7c/0xac)
[  429.362837] [<c0de1868>] (warn_slowpath_fmt) from [<bf290bf0>] (__vb2_queue_cancel+0x1d0/0x24c [videobuf2_common])
[  429.374537] [<bf290bf0>] (__vb2_queue_cancel [videobuf2_common]) from [<bf291548>] (vb2_core_streamoff+0x18/0xac [videobuf2_common])
[  429.387827] [<bf291548>] (vb2_core_streamoff [videobuf2_common]) from [<bf1d5474>] (__video_do_ioctl+0x238/0x490 [videodev])
[  429.400447] [<bf1d5474>] (__video_do_ioctl [videodev]) from [<bf1d6368>] (video_usercopy+0x184/0x5d8 [videodev])
[  429.411981] [<bf1d6368>] (video_usercopy [videodev]) from [<c043e5bc>] (sys_ioctl+0x34c/0x904)
[  429.421931] [<c043e5bc>] (sys_ioctl) from [<c0200040>] (ret_fast_syscall+0x0/0x4c)
[  429.430841] Exception stack(0xc3791fa8 to 0xc3791ff0)
[  429.436566] 1fa0:                   b6c47aa0 00000004 00000008 40045613 b6885980 00e3a6a8
[  429.446088] 1fc0: b6c47aa0 00000004 00000000 00000036 40045613 0052306c b6885980 00000008
[  429.455631] 1fe0: 00522e10 b6885954 004cb81c b6bcc9fc
[  429.461373] ---[ end trace 15db264193a78920 ]---
[  429.466673] videobuf2_common: driver bug: stop_streaming operation is leaving buf b8798e68 in active state
[  429.477669] videobuf2_common: driver bug: stop_streaming operation is leaving buf 2f955132 in active state
[  434.618808] ------------[ cut here ]------------
```

mdevaev, 2021-04-27:
Hello. What video capture device do you use? Also show `uname -a`.

vgropp, 2021-04-27:
uname `Linux pikvm 5.10.23-2-ARCH #1 SMP Thu Mar 18 18:33:35 MSK 2021 armv7l GNU/Linux`

capture is a `Lusya Standard HDMI-Compatible To CSI-2 Adapter Board Input `
```
/dev/video0
Driver Info:
        Driver name      : unicam
        Card type        : unicam
        Bus info         : platform:fe801000.csi
        Driver version   : 5.10.23
        Capabilities     : 0x85a00001
                Video Capture
                Metadata Capture
                Read/Write
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x05200001
                Video Capture
                Read/Write
                Streaming
                Extended Pix Format
Media Driver Info:
        Driver name      : unicam
        Model            : unicam
        Serial           : 
        Bus info         : platform:fe801000.csi
        Media version    : 5.10.23
        Hardware revision: 0x00000000 (0)
        Driver version   : 5.10.23
Interface Info:
        ID               : 0x03000005
        Type             : V4L Video
Entity Info:
        ID               : 0x00000003 (3)
        Name             : unicam-image
        Function         : V4L2 I/O
        Flags         : default
        Pad 0x01000004   : 0: Sink
          Link 0x02000007: from remote pad 0x1000002 of entity 'tc358743 10-000f': Data, Enabled, Immutable
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'RGB3' (24-bit RGB 8-8-8)
        [1]: 'UYVY' (UYVY 4:2:2)
```

does this help?

mdevaev, 2021-04-27:
This bug has already been fixed, strange. Try adding options `--dv-timings --persistent`.

Is it Pi-KVM or something else?

vgropp, 2021-04-27:
yes, its pi-kvm

adding `--dv-timings --persistent` to above ustreamer statement works for some time and then results in multiple `-- ERROR [11105.889    stream] -- Persistent device timeout (unplugged)` and the same error in dmesg. 

Is there any update i can apply to fix it in pi-kvm?

mdevaev, 2021-04-27:
Contact me via discord.

vgropp, 2021-04-28:
It was or is some kind of hardware problem, thanks again for the support. Adding `over_voltage=2` to /boot/config.txt does actually help and the stream is stable now

closed 2021-04-28 by vgropp

mdevaev, 2021-04-28:
:ok_hand: