pikvm/ustreamer #111: Mainloop select() error: Success

LewisSpring, 2021-05-09:
Hi there! If I run
`./ustreamer --host=0.0.0.0 --port=8080 --device=/dev/video0`
I get "NO SIGNAL" through the feed and
`================================================================================
-- INFO  [396.666    stream] -- Device fd=8 opened
-- INFO  [396.666    stream] -- Using input channel: 0
-- INFO  [396.670    stream] -- Using resolution: 640x480
-- INFO  [396.670    stream] -- Using pixelformat: MJPEG
-- INFO  [396.675    stream] -- Using HW FPS: 0 -> 30 (coerced)
-- ERROR [396.675    stream] -- Device does not support setting of HW encoding quality parameters
-- INFO  [396.675    stream] -- Using IO method: MMAP
-- INFO  [396.692    stream] -- Requested 5 device buffers, got 5
-- INFO  [396.704    stream] -- Capturing started
-- INFO  [396.704    stream] -- Switching to HW encoder: the input is (M)JPEG ...
-- INFO  [396.704    stream] -- Using JPEG quality: encoder default
-- INFO  [396.704    stream] -- Creating pool JPEG with 1 workers ...
-- INFO  [396.704    stream] -- Capturing ...
-- ERROR [397.705    stream] -- Mainloop select() error: Inappropriate ioctl for device
-- INFO  [397.705    stream] -- Destroying workers pool JPEG ...
-- INFO  [397.710    stream] -- Capturing stopped
-- INFO  [397.718    stream] -- Device fd=8 closed
================================================================================
-- INFO  [397.721    stream] -- Device fd=8 opened
-- INFO  [397.722    stream] -- Using input channel: 0
-- INFO  [397.727    stream] -- Using resolution: 640x480
-- INFO  [397.728    stream] -- Using pixelformat: MJPEG
-- INFO  [397.733    stream] -- Using HW FPS: 0 -> 30 (coerced)
-- ERROR [397.733    stream] -- Device does not support setting of HW encoding quality parameters
-- INFO  [397.734    stream] -- Using IO method: MMAP
-- INFO  [397.754    stream] -- Requested 5 device buffers, got 5
-- INFO  [397.767    stream] -- Capturing started
-- INFO  [397.767    stream] -- Switching to HW encoder: the input is (M)JPEG ...
-- INFO  [397.768    stream] -- Using JPEG quality: encoder default
-- INFO  [397.768    stream] -- Creating pool JPEG with 1 workers ...
-- INFO  [397.768    stream] -- Capturing ...
^C-- INFO  [398.302      main] -- ===== Stopping by SIGINT =====
-- INFO  [398.302      http] -- HTTP eventloop stopped
-- ERROR [398.769    stream] -- Mainloop select() error: Inappropriate ioctl for device
-- INFO  [398.769    stream] -- Destroying workers pool JPEG ...
-- INFO  [398.773    stream] -- Capturing stopped
-- INFO  [398.780    stream] -- Device fd=8 closed
-- INFO  [398.781      main] -- Bye-bye
`

`pi@octopi:~/ustreamer $ v4l2-ctl --all
Driver Info:
        Driver name      : uvcvideo
        Card type        : WEB CAM: WEB CAM
        Bus info         : usb-0000:01:00.0-1.2
        Driver version   : 5.10.17
        Capabilities     : 0x84a00001
                Video Capture
                Metadata Capture
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x04200001
                Video Capture
                Streaming
                Extended Pix Format
Media Driver Info:
        Driver name      : uvcvideo
        Model            : WEB CAM: WEB CAM
        Serial           :
        Bus info         : usb-0000:01:00.0-1.2
        Media version    : 5.10.17
        Hardware revision: 0x00000100 (256)
        Driver version   : 5.10.17
Interface Info:
        ID               : 0x03000002
        Type             : V4L Video
Entity Info:
        ID               : 0x00000001 (1)
        Name             : WEB CAM: WEB CAM
        Function         : V4L2 I/O
        Flags         : default
        Pad 0x01000007   : 0: Sink
          Link 0x02000013: from remote pad 0x100000a of entity 'Extension 4': Data, Enabled, Immutable
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
        Width/Height      : 640/480
        Pixel Format      : 'MJPG' (Motion-JPEG)
        Field             : None
        Bytes per Line    : 0
        Size Image        : 1843200
        Colorspace        : sRGB
        Transfer Function : Rec. 709
        YCbCr/HSV Encoding: ITU-R 601
        Quantization      : Default (maps to Full Range)
        Flags             :
Crop Capability Video Capture:
        Bounds      : Left 0, Top 0, Width 640, Height 480
        Default     : Left 0, Top 0, Width 640, Height 480
        Pixel Aspect: 1/1
Selection: crop_default, Left 0, Top 0, Width 640, Height 480, Flags:
Selection: crop_bounds, Left 0, Top 0, Width 640, Height 480, Flags:
Streaming Parameters Video Capture:
        Capabilities     : timeperframe
        Frames per second: 30.000 (30/1)
        Read buffers     : 0
                     brightness 0x00980900 (int)    : min=1 max=255 step=1 default=128 value=128
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=2 value=1
`

If I can provide anything more, please let me know.

LewisSpring, 2021-05-09:
Using this [issue](https://github.com/pikvm/ustreamer/issues/52), I've seemed to fix it.
Of course, new issue, I can't go above 800*480
Thanks!

LewisSpring, 2021-05-09:
Fixed that too by using --format=mjpeg.
Thanks!

closed 2021-05-09 by LewisSpring

mdevaev, 2021-05-09:
:)

erjoalgo, 2024-06-25:
Adding the `--format=MJPEG` fixed the issue for me. It wasn't obvious because the same exact command appeared to work fine outside the context of a system service, I assume there must have been a difference in the environment.

It would be very helpful if the error message could be updated to something more user-friendly, like "unable to determine stream format".