pikvm/ustreamer #113: Chromium-Based browser GPU Process high RAM usage

LewisSpring, 2021-06-09:
While I'm MJPEG streaming in Edge the GPU Process memory usage increases up to somewhere around 1,200,000K, and then goes back down when streaming stops.

My options: `--host=0.0.0.0 --port=8080 --device-timeout=5 -f 15 -l -q 80 -r 1920x1080 --format=mjpeg`

Barebones Chrome takes about 900,000K

mdevaev, 2021-06-09:
Isn't this the same problem as #30? As far as I remember, Edge is a blink-based browser.

LewisSpring, 2021-06-09:
> Isn't this the same problem as #30? As far as I remember, Edge is a blink-based browser.

It is sort of, but I figured it would be better to create a new issue.
The difference is that the high RAM usage is in GPU Process for me, while RGL's is in the tab itself.

> This is not a ustreamer problem

Ah, my apologies.

> Please report this issue to the chrome/blink developers, as I don't even have this issue reproducing.

I'll try talking to them. I tried talking to Microsoft, but their "Tier 1" support wanted me to Defrag/Optimise my drives and do an in-place upgrade.

mdevaev, 2021-06-09:
Believe me, I really want this problem solved. MJPEG is a very simple format, and if it works, it works. My software doesn't do anything special, it just sends a lot of JPEGs as prescribed by the format. For example Epiphany it works terribly and the image flickers - I think this is due to the glitches of the renderer. I think your browser is trying to decode every JPEG in the stream on the GPU. This is a good idea, but probably again due to either driver problems or incorrect browser code.

LewisSpring, 2021-06-09:
> Believe me, I really want this problem solved. MJPEG is a very simple format, and if it works, it works. My software doesn't do anything special, it just sends a lot of JPEGs as prescribed by the [...]

I've just finished speaking to Bitdefender. They had a memory leak issue in their software relating to MJPEG.
I almost can't believe it's simple with the amount of issues there is with handling MJPEG!

mdevaev, 2021-06-09:
If you can reproduce this problem in Chromium, then just let the blink developers know about it. They will fix it, then it will be fixed in Edge.

mdevaev, 2022-07-07:
Is it still relevant?

closed 2022-07-07 by mdevaev