pikvm/ustreamer #115: Supporting h264/WebRTC for separate capture devices?

aggieNick02, 2021-06-18:
I was interested in working on and contributing a PR to allow the new WebRTC functionality to be used in Pi-KVM when the video capture and encoding is occurring on another machine. If this sounds ok, just let me know. You could either assign the issue to me or close it and I'll come back when I have the PR ready. Here are the details:

I use Pi-KVM in a couple ways. The first is the fairly standard setup with capture via CSI device. The second is a bit different. I use the Pi and kvmd to do the keyboard and mouse control, but have a custom capture server running on a different machine (allows for capture at higher resolutions and framerates than CSI can do).

For mjpeg, all I had to do was override the streamer config to:
```
    streamer:
        unix: ''
        host: {{ capture_host_ip }}
        port: {{ capture_host_ustreamer_port }}
        cmd:
            - "/bin/sleep"
            - "infinity"
```

and then make sure my capture host implemented the same web API as ustreamer. The modularity of your system is nice and being able to have capture on one machine and keyboard/mouse on another could provide some nice upgrade paths in the future where the device doing capture would not need to also have USB OTG.

Anyways, with the new h264 webrtc support, doing this isn't as easy. I was able to prototype getting it working... the basic steps are:

1. Build janus with the streaming plugin enabled (https://janus.conf.meetecho.com/docs/streaming.html)
2. Replace the jcfg file for ustreamer with a basic jcfg file for a simple janus streamer that takes an rtp stream on a port
3. Modify stream.js to use the janus streaming plugin (and associated id) instead of the janus ustreamer plugin
4. Have the capture device transmit an rtp stream to the given port on the kvmd/janus device

Only step 3 seems a bit hacky. I"m imagining the name of the janus plugin and other data needed could be placed in the config file with defaults for the normal ustreamer case, and then overriding would cause stream.js to behave differently as needed. Thoughts?

aggieNick02, 2021-06-18:
Thanks for the advice - I think I should be able to write a sample proof-of-concept pretty quickly just building against the ustreamer code as all I then really have to write is receiving the frames. I'll share once I have something running.

I meant to ask, I noticed the python files for the memory sink. Did you create a python wrapper around that, or was it just something you were playing with?

mdevaev, 2021-06-18:
This is the client side (i.e. read-only wrapper) to receive frames from ustreamer for the vnc server in kvmd

aggieNick02, 2021-06-24:
Ah, ok, thanks. Had some other stuff interrupt my work with the shared memory, but expect to get back to that now.

aggieNick02, 2021-07-16:
Sorry I got a bit delayed on this. I finally got back around to it recently. It's been years since I had to write strict C, and I'd never used libevent before, so it was fun to get into.

I settled on using libevent and its chunked functionality to ingest an "mh264" stream from my capture machine. It's basically the same as mjpeg but with h264 frames instead of jpegs. We can all wish browsers actually supported that. :-)

Right now I'm dealing with an issue where the picture is coming through but garbled. Usually it will have a large horizontal green rectangle covering 1/3 of the image. It's probably on my end, but thought I'd mention in case you hit similar while you were developing. If I use the janus streaming plugin and send the frames over rtp by using the ffmpeg API, everything works right. But the same frames sent over http and then memsink to the janus ustreamer plugin are having the issue.

I logged checksums before the frames are sent over http on the capture machine and before they are sent to the memsink, so it's not a transport corruption. Digging into `_rtp_process_nalu`, etc at the moment in case there is something about my h264 encoding that is somehow not compatible with it.

mdevaev, 2021-07-16:
This sounds strange because I don't have any similar problems. I use a similar transport for VNC and do not see something like this.

aggieNick02, 2021-07-16:
Thanks for the quick response. That is useful to know; my best guess is incompatibility between the encoder output (generated by nvenc on an nVidia gpu) and the plugin, if that makes sense.  But maybe I've done something else wrong. I'll keep playing on my end.

Just as a sanity check, all I'm doing is this (capture is a 4k capture, but frame width,height,stride are not used by the ustreamer plugin so I don't think they actually matter, and I don't yet bother with marking keyframes in the frame struct, which afaict is a performance optimization when used with memsink_server_check).

```
memsink_s *sink = memsink_init("H264", "kvmd::ustreamer::h264",true, 0660, false, 10, 1);
uint8_t *buffer = malloc(CFG_MEMSINK_MAX_DATA);
int content_length = 0;
frame_s *frame = frame_init();
frame->format = V4L2_PIX_FMT_H264;
frame->width=3840;
frame->height=2160;
frame->stride=3840*3;
while(true)
{
    //Read next frame into buffer with length content_length
    frame_set_data(frame, buffer, content_length);
    memsink_server_put(sink, frame);
}
```

mdevaev, 2021-07-16:
I don't see any problems in this code :thinking:

aggieNick02, 2021-07-17:
Thanks for checking it so quickly for something I might have missed. I'm hoping to get to look a little more tomorrow but then will be away from work for a week. Either way I'll keep digging.

Proabably the best way forward for me is write out a raw h264 file that I can play back from the capture machine (right now every run is with a real capture), and then verify the exact same bitstream is sent to the ffmpeg rtp stream and the memsink source stream in the janus ustreamer plugin. If that checks out I'll dig into any differences in how the rtp processors deal with the bitstream. I'll update when I have more.

mdevaev, 2021-07-18:
Maybe your h264 uses a different profile? I used baseline.

aggieNick02, 2021-07-29:
That was a good thought. The nvidia encoder uses a "default" profile by default (doesn't say what that actually is). I forced it to baseline but still have the same result. I'm finally back after my break so digging again.

aggieNick02, 2021-08-03:
I figured out the problem and now know more about rtp :-). I looked at the difference between the output of ffmpeg's rtp code and ustreamer's. They use different packet sizes by default, and ffmpeg employs STAP-A to combine smaller NALs (like type 7 and 8) which introduces some noise in the diffs but are not the cause of the issue. Forcing ffmpeg to use the same packet size eliminated most of the noise.

The issue is in the marker bit in the rtp header. It is to be set for the very last rtp packet of the access unit, but rtp.c sets it for the last rtp packet of each type 1-5 NAL. I'm encoding 4k video which ends up a NAL type 7, NAL type 8, and 4 NAL type 5 for a single frame. I'm guessing (and will verify shortly) that most/all of the time the h264 generated by capture/encoding on the PI only has a single NAL type 5, which is why the issue isn't seen there.

I can send a PR or let you make the change if you prefer. I think all that needs to happen is eliminating the local bool `marked`, along with setting in the `switch`, in `_rtp_process_nalu`, replacing it with a parameter to `_rtp_process_nalu`. Then in `rtp_wrap_h264`, the new parameter should be passed as `false` to the call in the `while` loop, and `true` to the call at the end of the function.

mdevaev, 2021-08-03:
Shoe the patch here please

aggieNick02, 2021-08-03:
```
diff --git a/pikvm/ustreamer/janus/src/rtp.c b/pikvm/ustreamer/janus/src/rtp.c
index 90c06526..acc2a8ec 100644
--- a/pikvm/ustreamer/janus/src/rtp.c
+++ b/pikvm/ustreamer/janus/src/rtp.c
@@ -30,7 +30,7 @@
 #define PRE 3 // Annex B prefix length

-void _rtp_process_nalu(rtp_s *rtp, const uint8_t *data, size_t size, uint32_t pts, rtp_callback_f callback);
+void _rtp_process_nalu(rtp_s *rtp, const uint8_t *data, size_t size, uint32_t pts, bool marked, rtp_callback_f callback);
 static void _rtp_write_header(rtp_s *rtp, uint32_t pts, bool marked);

 static ssize_t _find_annexb(const uint8_t *data, size_t size);
@@ -118,7 +118,7 @@ void rtp_wrap_h264(rtp_s *rtp, const frame_s *frame, rtp_callback_f callback) {
                        if (data[size - 1] == 0) { // Check for extra 00
                                --size;
                        }
-                       _rtp_process_nalu(rtp, data, size, pts, callback);
+                       _rtp_process_nalu(rtp, data, size, pts, false, callback);
                }

                last_offset = offset;
@@ -127,18 +127,16 @@ void rtp_wrap_h264(rtp_s *rtp, const frame_s *frame, rtp_callback_f callback) {
        if (last_offset >= 0) {
                const uint8_t *data = frame->data + last_offset + PRE;
                size_t size = frame->used - last_offset - PRE;
-               _rtp_process_nalu(rtp, data, size, pts, callback);
+               _rtp_process_nalu(rtp, data, size, pts, true, callback);
        }
 }

-void _rtp_process_nalu(rtp_s *rtp, const uint8_t *data, size_t size, uint32_t pts, rtp_callback_f callback) {
+void _rtp_process_nalu(rtp_s *rtp, const uint8_t *data, size_t size, uint32_t pts, bool marked, rtp_callback_f callback) {
        const unsigned ref_idc = (data[0] >> 5) & 3;
        const unsigned type = data[0] & 0x1F;

-       bool marked = false; // M=1 if this is an access unit
        frame_s *ps = NULL;
        switch (type) {
-               case 1 ... 5: marked = true; break;
                case 7: ps = rtp->sps; break;
                case 8: ps = rtp->pps; break;
        }
```

mdevaev, 2021-08-03:
Aha, I see. I don't know too much about NAL and took this code from another project. If you can be 100 percent sure that this will not break anything and test it (including with raspberry), then I will accept the patch. I will also test it, of course.

aggieNick02, 2021-08-03:
Ah, ok. I can't ever be 100% but I'll do the testing and validating to get as 99.999 :-) .  I'm working on testing the normal capture setup (and verifying that those frames indeed only have one type 5 NAL) and I'll post back when that's done. Based on reading the spec, I'm pretty confident this is the way it should be. Verifying that there is only a single type 5 NAL will actually mean that there's no difference in output bits when the pi is the encoder, which should give us as close to 100% confidence as possible that nothing breaks.

I also don't know alot about NAL (just what I learned to debug this), but having the ffmpeg source and behavior was really valuable. Also the relevant bit from the spec is here (search for marker bit): https://datatracker.ietf.org/doc/html/rfc6184#page-10:
"Marker bit (M): 1 bit
      Set for the very last packet of the access unit indicated by the
      RTP timestamp,"
An access unit is a set of NALs that contain a primary coded picture - e.g., a video frame

mdevaev, 2021-08-03:
The author of the [original code](https://github.com/catid/kvm/blob/master/kvm_pipeline/src/kvm_video.cpp) said that this is really a bug:
> My code is designed for encoder output that isn't producing strips but whole frames at a time, so probably it's just a bug.

Okay, please send PR, I'll test it and apply tomorrow.

aggieNick02, 2021-08-04:
Created the PR here:
https://github.com/pikvm/ustreamer/pull/119

I had some issues getting setup to test that I fjust resolved, but am out of time for the day., So I haven't validated as planned for regular capture and verifying regular pi capture has only 1 type 1-5 NALU. That's first on my list for the morning, and I'll post as soon as that's done.

Thanks!

aggieNick02, 2021-08-04:
Was able to test the fix running with standard pi capture. Everything still works as expected, and I verified the pi generated encoding indeed only has 1 type 1-5 NALU per frame, even when the captured source is the max 1080p with a looping video playing and the h264 bitrate is turned up to the maximum. So this bug seems like it would never have affected the standard use case.

Just let me know if you'd like anything else tested. Thanks!

mdevaev, 2021-08-04:
Thanks! I will test it myself today and merge.

mdevaev, 2021-08-04:
I was just thinking. What happens if the last packet is the one that was in the loop and the condition after the loop is not met? Then marked will be false for the last package from the series. It seems that this will cause a new bug.

https://github.com/pikvm/ustreamer/blob/master/janus/src/rtp.c#L121

aggieNick02, 2021-08-05:
I thought that at first too, but that situation can't actually happen. I imagine the code could be written a little bit differently to make that more clear. The gist is that the loop and subsequent `if` are structured so that the last packet is always processed by the `if` on line [127](https://github.com/pikvm/ustreamer/blob/master/janus/src/rtp.c#L127). The only reason for that `if` is to handle the degenerate case gracefully when there are no NALUs at all.

Here's a more careful breakdown:

The while loop iterates until it hits the break on line [111](https://github.com/pikvm/ustreamer/blob/master/janus/src/rtp.c#L111) when `offset < 0`. There are two possibilities when that happens:

1. This was actually the first iteration of the while loop. In this case, last_offset = -PRE, there are no NALUs at all, and `_rtp_process_nalu` is never called anywhere in the `rtp_wrap_h264`
2. This is not the first iteration of the while loop. Therefore line [124](https://github.com/pikvm/ustreamer/blob/master/janus/src/rtp.c#L124) has been executed at least once, setting `last_offset` to `offset` which is `>=0` since it was `>=0` after the check on line [110](https://github.com/pikvm/ustreamer/blob/master/janus/src/rtp.c#L110), and then had a non-negative number added to it on line [113](https://github.com/pikvm/ustreamer/blob/master/janus/src/rtp.c#L113). Since line [124 ](https://github.com/pikvm/ustreamer/blob/master/janus/src/rtp.c#L124)is the only place that sets `last_offset` other than its initial value, this means when the loop exits after more than 1 iteration, `last_offset` is non-negative and `_rtp_process_nalu` will be called with `true` for `marked` on line [130](https://github.com/pikvm/ustreamer/blob/master/janus/src/rtp.c#L130)

mdevaev, 2021-08-05:
It's disgusting. I also looked at it again.

If you want, you can try to refactor this function. I'm open to it.

aggieNick02, 2021-08-05:
Haha, sure. I'll give it a go. The ffmpeg code for the same thing does it more naturally, but it also has some "fun" code too that isn't obvious until you figure out what it is doing. It's very sharp and efficient C code, but takes some staring to understand. Like this code that skips the NALU start bytes in a buffer:
`while (!*(r++));`
(NALU start bytes can be `00 00 01` or `00 00 00 01`, so if you know you are on the beginning of start bytes, it's a nice compact way to advance past them. But I stared for a few minutes without a comment before I realized.)
And then the code that searches for the next start code is super-optimized to be able to generally advance 4 bytes at a time, regardless of endianness, doing 3 boolean operations and a subtraction on the integer formed by the 4 bytes - it's something else. :-)

But yeah, I'll see what I can do with the function and send a PR likely today.

mdevaev, 2021-08-05:
But please do it without `while (!*(r++));` :D I would prefer that the changes relate only to rtp_wrap_h264, because the rest seems more or less clear to me.

Take your time.

aggieNick02, 2021-08-05:
Haha, no worries, I won't do anything like that. :-) I'll just be playing with that single function to make it more clear when/how it terminates, trying to keep the actual behavior identical.

aggieNick02, 2021-08-05:
I started playing a bit more with the fix, and while it improves behavior, there is still at least one issue I'll need to figure out. Every few seconds, some or all of the frame will go back to a frame from a few seconds before for a short duration. The issue doesn't show up when I use the janus streaming plugin.

However, the janus streaming plugin isn't perfect either, so that's not a clear alternative - right now it stutters as bitrate climbs. So I've got a problem to overcome either way.

I'm sending the same stream over both rtp via ffmpeg to the janus plugin, and over tcp/memsink to the ustreamer plugin, with a pi receiving each, so I can see the behavior in both.

aggieNick02, 2021-08-05:
Meant to add - the lag is comparable with both plugins. Definitely not as good as the native pi capture, which is to be expected with another network hop. Makes me hopeful that when there is a pi 5 the specs are enough to pull off even a higher resolution/framerate capture. :-)

aggieNick02, 2021-08-06:
Well, I know more about the janus streaming plugin issue. It actually has nothing to do with janus. Rather, it appears the pi4 has serious issues with UDP traffic: https://www.raspberrypi.org/forums/viewtopic.php?t=264106

I actually stumbled onto this two days ago when I forgot and left my rtp sender running, so it was throwing udp packets constantly (about 1MB/s) at the pi. The end result was that running things like apt update or wget would run at 30KB/s - it was awful. Took me a few hours before I realized what the cause was, and then could easily reproduce by turning on/off the udp traffic.

Looking at the webrtc-internals in chrome, the janus streaming plugin reports over 10% packet loss on the incoming udp stream. Janus is sending a slowlink message every second or so. Following some of the suggestions in the pi forums improves things significantly - not perfect, but not the mess I see by default.

I guess even if I wanted to use the janus streaming plugin, I'd have to have my tcp sender, as the janus streaming plugin doesn't support tcp incoming streams. That assumes that sending to udp on localhost doesn't have issues like this. :-P

Anyways, onto investigating the remaining ustreamer plugin behavior I'm seeing. Just had to share the UDP issue. The ustreamer plugin doesn't require sending over UDP. :-)

mdevaev, 2021-08-06:
Try this sysctls:
```
net.core.rmem_default = 500000
net.core.wmem_default = 500000
net.core.rmem_max = 1000000
net.core.wmem_max = 1000000
```

aggieNick02, 2021-08-06:
Thanks for the suggestion. Unfortunately it doesn't seem to help here.

Adding cpu isolation for cpu 0 and fixing frequency to 1500 and forcing turbo (what was suggested by one person in the forums I linked) actually makes it significantly better, but still not perfect.

mdevaev, 2021-08-06:
@xornet-sl what do you think about it (UDP)?

aggieNick02, 2021-08-06:
A little good news for the tcp workaround option. Using the same "mh264" tcp transport I use for memsink, I can dump the h264 frames to stdout instead of the memsink, and pipe that to ffmpeg locally to do the rtp packetization. There's no noticeable lag compared to the memsink option. The command is just something like this (where streamer.bin is my little program that reads h264 over tcp and then either sends to memsink or stdout).

`./streamer.bin | ffmpeg -f h264 -i pipe: -an -c:v copy -f rtp rtp://127.0.0.1:8004`

Even with memsink being the preferred option, having the janus streaming plugin is helpful for debugging possible packetization issues, so I'm excited there's a way to still have it working without having to worry about the udp issues. Might also be useful for the regular capture for similar debugging scenarios.

aggieNick02, 2021-08-06:
Discovered something that has me at a bit of a loss on the UDP issue. Was trying to reproduce on a second pi and couldn't. It appears the bad behavior only happens when connected to certain ports of my consumer netgear 8-port switch. Port 4 results in the bad behavior. Ports 6 and 7 do not.

No clue at this point if it is actually the switch or just some sort of timing jitter that exposes the pi issue.

aggieNick02, 2021-08-09:
Have concluded that the udp issue is real. Changing ports just provided enough timing jitter to make the issue go away, as it is sensitive to tiny timing changes. I feel the issue is bad enough that I'll just be avoiding UDP until/if there is ever improvement on the pi. I should mention that this is with me running on raspberry pi OS. I'll play a bit later with arch to confirm the weirdness there.

mdevaev, 2021-08-10:
I'm wondering why none of the users have yet encountered this on Pi-KVM+Janus.

aggieNick02, 2021-08-10:
Well this is only in my scenario where I have the capture happening on another device and I'm sending it over UDP to the Pi-KVM running the Janus streaming plugin. I'm sorry if that wasn't clear. The normal Pi-KVM user is not going to have their pi getting hit constantly by RTP over UDP.

I still plan to use the ustreamer plugin instead of the Janus streaming plugin, but using the Janus streaming plugin has been useful for determining if issues I hit are in the ustreamer plugin or something on my capture/encoding side.

aggieNick02, 2021-08-10:
On the ustreamer plugin issue I was seeing with occasional glitching, I've figured that one out I believe. I am sending the frames to the memsink as I parse them from the network stream, and that is resulting in the ustreamer plugin missing frames. This makes sense since the normal capture sends a frame about every 16ms, but when shipped over the network timing can get altered.

By adding a sleep of 10ms after each call to `memsink_server_put` , I was able to reduce the problem but not eliminate it. (I went from tens of dropped frames out of ~1000 to 1.)

Did you have any thoughts on adding protection against this? I realize it doesn't seem to happen with normal pi capture, but if it ever does, it corrupts the video until the next keyframe is received. A boolean in the shared memory, or looking at the existing `last_client_ts`, would allow the server to realize that the client has not yet read the frame, sleep for a short duration, and try again. A reasonable limit on the number of retries (or just using client_ttl) would stop this from happening forever if the client has disappeared.

Happy to get your thoughts and if you're open to it, work on a PR.

mdevaev, 2021-08-10:
Try to reduce polling interval in `usleep()` in `flock_timedwait_monotonic()`: https://github.com/pikvm/ustreamer/blob/3558089a22451fa8b2e977942be96f64211c6653/src/libs/tools.h#L139

aggieNick02, 2021-08-10:
Sure thing. So just tweaking that to `usleep(100)` and removing my delay around calls to `memsink_server_put`, I was still dropping a good number of frames.

I added some simple timing around the `LOCK;rtp_wrap_h264(...);UNLOCK` code in `plugin.c`. The real root of the problem is that occasionally just this portion of the code will take longer than the frame duration, i.e. 0.017 seconds for 60fps source. It doesn't happen super often (once every few seconds), but when it does (or even is close to the duration), there is a nice window for a frame to be missed, and no decrease in sleep in the locking can help.

Does that make sense?

If the rtp sending was consistently over the frame duration of the source, no amount of queueing could solve the problem. But when it is an occasional thing, simply sleeping when a frame hasn't been read by the client seems like a viable strategy.

mdevaev, 2021-08-11:
I don't understand what you mean a little. Can you explain it step by step?

aggieNick02, 2021-08-11:
Sure. Do you mean the problem or possible solution, or both?

mdevaev, 2021-08-11:
The solution.

aggieNick02, 2021-08-11:
Sure. So here's a rough idea as a pull request - it isn't polished or cleaned up all the way, but gives the general idea:
https://github.com/pikvm/ustreamer/pull/120/commits/74b208a53ec76fd0b14339bd64c3f6ecbe920f8d

Basically the idea is to start treating the shared memory as a single element queue that, in certain conditions, we don't overwrite the current element. I tried this code after removing my previous hack of `usleep(10000)` around my memsink puts, and watched it send all the frames successfully, including some where overwriting would have happened but was detected and tried again 1ms later.

A few things:

1. There is a bit more cleanup that can happen if desired. The `get_now_id()`,`last_id` etc are no longer really needed. The `read_by_client` field can serve that purpose too.
2. I had a hard time understanding what purpose the `memsink_server_check` serves. It may return more quickly and tell us we have clients, but then we will just wait and lock right after that. Am I missing something? It also checks `sink->mem->last_client_ts`, which I think can be uninitialized. I haven't tried to fix `memsink_server_check` up for my idea, and am wondering if it could just be removed
3. You'll notice in my while loop in the put function, If there are no clients, we constantly overwrite any unread frame in shared memory. That way when a client does start reading, it gets the latest frame, and then doesn't have to miss any frames
4. If open to it, I'd love to merge the client code in `plugin.c` with `memsink_client_get` in `memsink.c` - I spent a good few minutes confused as I thought the plugin called `memsink_client_get` and my changes to it weren't affecting anything.

mdevaev, 2021-08-12:
Thanks.

> It may return more quickly and tell us we have clients

You're right

> but then we will just wait and lock right after that

All this is necessary in order to avoid blocking and do the check quickly.

> It also checks `sink->mem->last_client_ts`, which I think can be uninitialized

Indeed. We need to somehow handle this situation.

I understand your idea, my first implementation of memsink was exactly like this. But in this case, we lose the ability to use one sink to feed multiple clients.

aggieNick02, 2021-08-12:
Thanks for the feedback. 

On the multiple clients, when are there multiple clients? When both vnc and janus are running? Just thinking out loud, if clients had to announce themselves, then the server could set an `unread_by_clients_count` and that could get decremented as each client read. Of course, if one client disappeared, that would cause problems for the other client until the ttl passed and the client was deemed gone. Another option is multiple sinks, one per client. I know right now the code is very careful about avoiding copies of the frame, and another copy per sink would be needed for one sink per client.

On the `memsink_server_check`, I still feel like I'm missing something. If `memsink_server_check` is removed from `hs64_stream_process`, blocking still does not occur until `memsink_server_put` is called later in the function. The only time behavior changes is when `memsink_server_check` returns false, which is an error situation or when the sink is believed to have no clients *and* the frame metadata is identical. When there is a client or different frames are being generated, this function generally returns true. Basically I'm confused then I guess by what `memsink_server_check` is helping with by not just always returning true.

Sorry if it's something obvious on `memsink_server_check` - I may just be missing it.

aggieNick02, 2021-08-12:
I agree there really may not be a simple solution. Thank you for looking at the idea and answering my questions.

When you say "slowdown" you mean the option that reduces CPU load when there are no clients - that is, [line 132](https://github.com/pikvm/ustreamer/blob/master/src/ustreamer/stream.c#L132) in `stream.c`?

I'll play with it a bit to try to understand better as I'm still missing the point of the `memsink_server_check`. Almost every call to `memsink_server_check` returns true, which means it is followed by a call to `memsink_server_put`, which also determines if there are live clients. If there are no clients, `memsink_server_put` will not block and set `has_clients=false` quickly. If there are clients, there may be blocking for up to 1 second before `has_clients=true` happens. Is that possible 1 second delay in realizing there are clients the reason for having `memsink_server_check`? Or am I still thinking wrong?

aggieNick02, 2021-08-12:
I can see how with ~30fps h264 at 1080p (what Pi-KVM gets), the single sink approach with multiple clients works well. As long as the loop rate of the readers is a bit faster than the loop rate of the writer (about 30Hz/33ms per iteration) things will generally work out just fine.

The ustreamer Janus plugin seems to occasionally spike to around 20ms per loop iteration with the external capture I provide it.

When I try to do ~60fps content, it means the reader occasionally can't keep up. And due to how h264 works, that is a much bigger deal than when a jpeg is missed for mjpeg.

There might even be certain operations with regular capture Pi-KVM at 30fps that cause a reader hiccup resulting in a short visual blip in the feed. It could be fun to see if there are ways to make it happen. ;-)

The one "easy" option available to me would be to see if I could improve the loop rate of the ustreamer Janus plugin. I may do a little profiling. I mentioned many posts up that ffmpeg does some fancy bitwork sometimes - I would not be surprised if, for example, `_find_annexb` could be improved a bit if profiling shows it worthwhile.

mdevaev, 2021-08-12:
I vote that the rtp sending code takes too much time

aggieNick02, 2021-08-13:
Did some quick validation before attacking the rtp sending. With Pi-KVM capture at 1080p and max h264 quality, the biggest h264 frame we see is around 250KB and there are spikes of about 20ms. Since the capture is at 30fps this works fine.

I inserted an artificial 30ms sleep before the rtp send, which means that if the real send takes 10ms, total is 40ms and frames will be missed by this reader. And indeed the viewed stream is full of visual errors. 

Then I did some simple profiling. The hiccup comes from the call to `callback`, which is `_relay_rtp_clients`. I'm out of time for the evening, but I'll look at that function tomorrow, a little afraid that most of the time is just in the Janus callback.

mdevaev, 2021-08-13:
I fixed usage of uninitialized `last_client_ts` and added a comment about `rtp_wrap_h264()` (until we refactor it).

aggieNick02, 2021-08-13:
These both look great, thanks for fixing/explaining.

aggieNick02, 2021-08-13:
So indeed the spike is from the Janus callback `g_gw->relay_rtp` . Which means it cannot be directly improved.

Also, I did not realize fully what Janus does/doesn't do. I thought Janus handled more of the transmitting to multiple clients, but instead the plugin has to call `g_gw->relay_rtp` for each "session", and there is one session per browser client. This means another way to get the streaming to misbehave is to open enough browser windows all pointed at the same stream. I realize this isn't a real use case at all for Pi-KVM, but it is nice to have a way to make things misbehave besides inserting artificial sleeps. :-)

(For me I can get h264 corruption by playing the Big Buck Bunny trailer (nice open source ogg content - I use 480p ogg theora from https://peach.blender.org/trailer-page/) on a 1080p30 source with capture set to highest kbps. With 10 browser windows open to it, I'll see occasional corruption. With 15, constant.)

aggieNick02, 2021-08-13:
The only idea I have left for dealing with the Janus callback hiccups is to push them into their own thread with some buffering. The thread which currently gets the buffer from the memsink and sends it to the callback would instead send it to a memory buffer, with a new thread reading from that buffer and making the Janus callback. 

It does mean having to allocate memory for a buffer and copying from the memsink to the buffer in the Janus plugin, but at least the copying is there, and not in ustreamer. ustreamer would not change and still have a single memsink that can be read by multiple memsink clients. It is the Janus plugin that is unable to keep up with the memsink, so in my mind at least, it makes sense that it should adapt to read the memsink more reliably.

mdevaev, 2021-08-13:
> Also, I did not realize fully what Janus does/doesn't do.

No one lol :D

> The only idea I have left for dealing with the Janus callback hiccups is to push them into their own thread with some buffering.

This is a good idea, but I would prefer it to be optional, since any queue and copying will increase the delay. For Pi-KVM, this is significant.

aggieNick02, 2021-08-13:
> This is a good idea, but I would prefer it to be optional, since any queue and copying will increase the delay. For Pi-KVM, this is significant.

That makes sense. I'll start on it and then get your feedback if that sounds ok?

mdevaev, 2021-08-13:
Sure

aggieNick02, 2021-09-14:
I finally was able to get a separate thread and also some code to detect missed writes put together. I sent a PR to get your feedback: https://github.com/pikvm/ustreamer/pull/124

aggieNick02, 2021-09-14:
Also, I have the code for receiving h264 frames over a socket and sending to the memsink (like we talked about a while ago). Let me know if you are interested in having it here or just prefer I keep it on my own. It's structured very similar to the ustreamer "dump" program, using your options parsing, etc.

It works pretty well, but has one major issue which will require me to stop using libevent. It works great until 4GB are transferred, because libevent uses a `size_t` to track the bytes transferred. So when the 4GB are passed, the connection is killed. :P

aggieNick02, 2021-09-15:
I moved my h264 frame receiver to libcurl from libevent without too much trouble. It requires a little more management of the incoming bytes (you are given bytes instead of asking for them) so the ring buffer in this PR fit nicely for that purpose as well.

mdevaev, 2022-11-08:
Sup! Are we going to do something else here, or has your task already been solved?

[2 more replies]

closed 2022-11-10 by mdevaev