pikvm/ustreamer #116: How to pick up the stream using gstreamer?

dmacattack, 2021-07-04:
I'm interested in using this to replace gstreamer. 
Has this been benchmarked against gstreamer?
Also can this stream be picked up from a non browser. Like gstreamer ? I'm curious about the codec/ transport protocols

mdevaev, 2021-07-05:
[image]
https://fluxcoil.net/wiki/hardwarerelated/camlink_hdmi_converter
https://fluxcoil.net/wiki/hardwarerelated/raspberry_pi_4_tc358743
This is a regular MJPEG. That is, you can simply use the URL in the browser.

dmacattack, 2021-07-06:
>> Also can this stream be picked up from a non browser. Like gstreamer 

> This is a regular MJPEG. That is, you can simply use the URL in the browser.

I want to pick up this stream **without** a browser. MJPEG is the codec, but what is the transport mechanism ? Is it RTSP/RTP ? UDP ? other ? I want to use a client that doesnt have a browser. Is that possible ? Is there a gstreamer pipeline  that can be used as a test client ?

mdevaev, 2021-07-06:
MJPEG is a transport (sort of), the codec is JPEG: https://stackoverflow.com/questions/47729941/mjpeg-over-http-specification

> I want to use a client that doesnt have a browser. Is that possible?
Yes. Ustreamer can stream to shared memory, and you can use dump tool to pipe it. Like:
`ustreamer --sink channel`
`ustreamer-dump --sink channel - | gstreamer ...`

closed 2021-07-06 by dmacattack