pikvm/ustreamer #117: Can't get stream to show in octoprint

Jyers, 2021-07-05:
I've looked through all the different issues that have been brought up regarding octoprint, and while there are a few that seem to point to this exact issue, none gave any meaningful advise as to how to fix it...

I'm starting ustreamer with this command "./ustreamer --host=:: --port=8080 --device=/dev/video0 -r 1920x1080 -f 30 -m MJPEG"

It starts up beautifully (Love the polish compared to mjpg_streamer) and I can see the stream in my browser, but I can not for the life of me get it to show up in octoprint. It just gives a "stream not loaded" error with a link that brings me to a fully functional stream... Just not in the octoprint interface.

Any help would be greatly appreciated as I was having some issues with mjpg_streaming and this seems like a very nice alternative.

mdevaev, 2021-07-05:
Hello. See #72, it might help.

Jyers, 2021-07-05:
Thanks for the quick response!

I did see that, but where am I supposed to put `http://localhost:8080/?action=stream&123456789=1`? In octoprint, somewhere in the ustreamer command? I tried adding the &123 bit to my octoprint webcam url and it didn't seem to make a difference.

Jyers, 2021-07-05:
Wait, looks like I misinterpreted what the fix was. Changing the url from /?action=stream to /stream worked. Thanks for your time, you've got a really nice project here :)

closed 2021-07-05 by Jyers

mdevaev, 2021-07-05:
Cool. BTW I recommend contacting the octoprint developers and asking them to add official support for ustreamer. It seems that you are not the only person who wants to get rid of mjpeg_streamer.

Jyers, 2021-07-05:
Yeah, maybe I'll reach out. Although with the correct url it seems to work perfectly, just needs a bit of documentation from octo I suppose.

Btw, any plans to support http camera setting adjustment like mjpg_streamer has? Not super quintessential, but would certainly be a nice-to-have.

mdevaev, 2021-07-05:
If you are talking about changing parameters in real time, then this is a long-term matter/

Jyers, 2021-07-05:
Alright sounds good. 

Wait one more thing (Sorry I'm the worst) Is OpenMax IL still supported? I tried building with the option enabled and got a warning about a pointer and this error `cannot find -lopenmaxil` I saw an issue brought up about it being deprecated, but I didn't know if it's still a viable option for better performance. I'm running a pi 4.

mdevaev, 2021-07-05:
Yes, it's supported. You probably don't have any libraries installed or something.

Jyers, 2021-07-05:
I thought I installed everything mentioned in the readme... I'll try again, whatever it is it's definitely my fault, I just need to figure out what :P