pikvm/ustreamer #118: Issues with more than 2 simultaneous streams

Jyers, 2021-07-06:
So I'm a developer, but I'm still pretty new to dealing with usb devices and their interactions with linux. I was wondering if I could get some help diagnosing the issue I'm having.

I have three USB webcams hooked up to my pi 4. I would ideally like them all to stream 1080p 30 fps. The issue that I'm having only occurs when all three are going... I can have two streams working perfectly fine, and the second I start the third one the issue begins.

The issue itself is the signal going in and out. As in the webcams are disconnecting and then get re-initialized. They all do this in sink when it happens. At fairly regular intervals they will all show "No Signal" for a second or two, then spring back to life.

What I would like to know is whether this is a hardware limitation or something that I have mis-configured. And if it is a hardware limitation is there anyway to overcome it. Are there any parameters I should be setting to better optimize the streams, or maybe something I should set up in linux to help with bandwidth issue.

If it helps, running in debug mod shows this error when they lose signal `Mainloop select() error: Inappropriate ioctl for device`

Any help would be greatly appreciated. Like I said, I know my way around linux, but this area is a bit new to me.

djkraven, 2021-07-06:
@Jyers  I believe you are having the same problem I was running into with two webcams. The solution I ended up with is using a powered USB hub. It seems the Pi4 USB hub has issues supporting multiple webcams. I am not sure what the issue is with the Pi4 internal hub but using an external powered one makes the stream stable so it is a guess that power might be the issue.

This is the hub that resolved the No Signal issue.
https://www.amazon.com/gp/product/B075NMVGP7/

This is the webcam I use.
https://www.amazon.com/dp/B082PM9T4J/

Jyers, 2021-07-06:
@djkraven That was definitely one thing I considered, but the fact that it only happens when I actually stream from the third webcam is surprising. You'd think it would happen anytime all of them were plugged in.

Jyers, 2021-07-07:
Okay, I've ordered a powered hub and we'll see if it fixes the issue

brussli1, 2021-07-07:
sounds like power issues, since probably your power supply can support up to 2 cameras and when you plug the third it starts underpowering them

mdevaev, 2021-07-09:
See #99. This is a kernel bug.

Jyers, 2021-07-09:
Just got my powered hub today, still seeing the same behavior. I looked through the thread you linked but the final solution seemed to be just connecting a hub anyway.

One major difference I've seen between my issue and those I've seen from others is that my webcams don't stop working all together, they keep running but very consistently stop streaming every few seconds, leaving me with three streams all running, but simultaneously flashing no signal every few seconds for about a second at a time.

I'll keep troubleshooting and update this with any of my findings

Jyers, 2021-07-09:
Fun new information! I tried lowering the resolution of the streams and the issue went away... So this means it's definitely not a power issue (too bad I already bought the hub...) 

Anyway, what does this mean for my issue, some kind of bandwidth restriction? If so is there any way to overcome this aside from simply limiting the resolution?

djkraven, 2021-07-11:
@Jyers Did you get the Plugable hub I suggested? I had a different hub originally and it didn't solve the problem. The Plugable did. It may be an issue with the bandwidth on the hub. Plugable has been good at USB engineering. I agree that it does not seem to be a power issue and more a change in the USB technology that changed between the pi3 which works with multiple USB webcams and the Pi4 which has issues with more than on camera.

Jyers, 2021-07-12:
I'll be honest, I went for a cheap 4 port hub, but I really don't think the hub is the issue at this point. Based on the fact that changing the resolution improved reliability really indicated to me that this is a bandwidth issue. To test to make sure the bug wasn't the issue, I tried plugging the cameras into my super reliable anker 4 port usb 3.1 hub and found the same behavior. I'm really at a loss for what to try next...

djkraven, 2021-07-12:
@Jyers I understand what you are saying and agree the bandwidth is likely to be the issue. That bandwidth issue is in the USB of the Pi. Can you get two cameras running with the Anker? What I can confirm is using the Plugable products, one is the hub mentions and the other is a dock) I can run the two webcams without issues. Currently running as below. I did have them running at 1920x1080 quality 100 but reduced it when I discovered that was using 100mb/sec for the stream over the ethernet. Using the current settings reduced it to a max of 10mb/sec, an average 5mb/sec. I could try adding a third and fourth cam to see if it starts failing.

Here are my current systemd scripts

`/usr/bin/ustreamer --process-name-prefix ustreamer-0 --log-level 0 --device /dev/video0 --device-timeout=10 --quality  40 --resolution 1280x720  --desired-fps=29 --host=0.0.0.0 --port=8080 --static /var/www/html/ustreamer-0/`

`/usr/bin/ustreamer --process-name-prefix ustreamer-0 --log-level 0 --device /dev/video1 --device-timeout=10 --quality  40 --resolution 1280x720  --desired-fps=29 --host=0.0.0.0 --port=8081 --static /var/www/html/ustreamer-1/`

Jyers, 2021-07-12:
Yes, I can get two webcams running at 1080p no problem, it's just when I add the third that I get issues. I'm currently getting passable results by running all three at 1280x720. Still get occasional dropouts, but not nearly as bad. I'll try playing around with the quality setting as I haven't tried that yet. Thanks for all the input.

Also, any particular reason you're running at 29 fps and not 30? Just interested.

Jyers, 2021-08-02:
@mdevaev So I think I've come to the conclusion that this is definitely just a hardware bottleneck somewhere. One that I most likely will not be able to surpass.

Given this fact, is there any way to configure ustreamer to only enable a camera when the stream has a client? I know there is the -l option which lowers the framerate, but that doesn't seem to do anything for my issue. If there was a way to have the cameras completely stop streaming when not being viewed that could help my situation.

mdevaev, 2021-08-02:
No, there is no way to stop the capturing and it will not appear in the near future. uStreamer is designed to be used together with a certain service that will control its launch (in the case of Pi-KVM, this is KVMD, it launches uStreamer when someone opens a page and establishes a connection with KVMD via a web socket).

closed 2021-10-21 by mdevaev