pikvm/ustreamer #121: Failed to build on openwrt - encoder.c:(.text+0x2dc): undefined reference to `frame_copy_meta'

JohnTar22, 2021-08-18:
as described in the title, can't get the package to build using openwrt build system.
getting "encoder.c:(.text+0x2dc): undefined reference to `frame_copy_meta'"

i've installed the deps on host pc, ran the make WITH_PTHREAD_NP=0. 

but still getting errors.

any help will be much appreciated.

john.

mdevaev, 2021-08-20:
Try to use GNU Make

JohnTar22, 2021-08-22:
I am using GNU Make 4.2.1, the native make that comes with ubuntu 20.04, the target board is not raspberry pi, maybe that's the problem?

mdevaev, 2021-08-22:
ustreamer works on any platform, I use it on arch/x86_64.

If you can build it on ubuntu outside of openwrt, then there is something wrong with the build system. I don't know much about openwrt and I can't say what could have broken there.

First, try building on ubuntu from the README to make sure that it works at all.

aggieNick02, 2021-09-08:
Just FYI, I build on ubuntu 20 just fine

JohnTar22, 2021-09-10:
True, on x64 it works fine.