pikvm/ustreamer #122: Audio support ?

NiKiZe, 2021-09-06:
As I understood it this is something that is in the pipe for pikvm/pikvm#97
My question would be specific for HDMI capture and H264 streaming.

Would I be correct in assuming that this would be done in ustreamer ?
(searching for "audio" in the repo gave almost no results, so I hope this issue can also serve as documentation for anyone else that is looking for the same)

mdevaev, 2021-09-09:
This will be done in the Janus plugin for uStreamer

djkraven, 2021-12-01:
When will the Janus plugin be available? Is there information on how to use plugins?

mdevaev, 2022-06-07:
Audio will be available for TC358743 HDMI capture with Janus WebRTC. All info will be posted here: https://github.com/pikvm/pikvm/issues/97

closed 2022-06-07 by mdevaev