pikvm/ustreamer #123: Can't build with `WITH_OMX=1` on Raspberry Pi OS Bullseye

The-EG, 2021-09-11:
It would appear that `libraspberrypi-dev` has changed recently, at least in bullseye.

Attempting to build with `make WITH_OMX=1` ends with the error:
```
In file included from ustreamer/encoder.h:45,
                 from ustreamer/encoder.c:23:
ustreamer/encoders/omx/encoder.h:33:10: fatal error: IL/OMX_Core.h: No such file or directory
   33 | #include <IL/OMX_Core.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
```

`libraspberrypi-dev` is indeed installed:

The-EG, 2021-09-11:
For comparison, the `libraspberrypi-dev` package on another RPi that is still running buster.

Note the change in source and homepage! Unfortunately, I haven't been able to work out if there's a different package to install yet or if this is an issue with the package itself.
```terminal
taylor@brewserver:~ $ apt show libraspberrypi-dev
Package: libraspberrypi-dev
Version: 1:1.20210831-1
Priority: optional
Section: libdevel
Source: raspberrypi-firmware
Maintainer: Serge Schneider <serge@raspberrypi.org>
Installed-Size: 3,324 kB
Depends: libraspberrypi0 (= 1:1.20210831-1)
Homepage: https://github.com/raspberrypi/firmware
Download-Size: 400 kB
APT-Sources: http://archive.raspberrypi.org/debian buster/main armhf Packages
Description: EGL/GLES/OpenVG/etc. libraries for the Raspberry Pi's VideoCore IV (headers)
 This package contains headers and other development files for implementations
 of EGL, OpenGL ES, OpenVG, OpenWF Composition, and others for the Raspberry
 Pi's VideoCore IV multimedia processor.
```

```terminal
taylor@brewserver:~ $ dpkg-query -L libraspberrypi-dev
/.
/opt
/opt/vc
/opt/vc/include
/opt/vc/include/EGL
/opt/vc/include/EGL/egl.h
/opt/vc/include/EGL/eglext.h
/opt/vc/include/EGL/eglext_android.h
/opt/vc/include/EGL/eglext_brcm.h
/opt/vc/include/EGL/eglext_nvidia.h
/opt/vc/include/EGL/eglplatform.h
/opt/vc/include/GLES
/opt/vc/include/GLES/gl.h
/opt/vc/include/GLES/glext.h
/opt/vc/include/GLES/glplatform.h
/opt/vc/include/GLES2
/opt/vc/include/GLES2/gl2.h
/opt/vc/include/GLES2/gl2ext.h
/opt/vc/include/GLES2/gl2platform.h
/opt/vc/include/IL
/opt/vc/include/IL/OMX_Audio.h
/opt/vc/include/IL/OMX_Broadcom.h
/opt/vc/include/IL/OMX_Component.h
/opt/vc/include/IL/OMX_Core.h
/opt/vc/include/IL/OMX_ILCS.h
/opt/vc/include/IL/OMX_IVCommon.h
/opt/vc/include/IL/OMX_Image.h
/opt/vc/include/IL/OMX_Index.h
/opt/vc/include/IL/OMX_Other.h
/opt/vc/include/IL/OMX_Types.h
/opt/vc/include/IL/OMX_Video.h
/opt/vc/include/KHR
/opt/vc/include/KHR/khrplatform.h
/opt/vc/include/VG
/opt/vc/include/VG/openvg.h
/opt/vc/include/VG/vgext.h
/opt/vc/include/VG/vgplatform.h
/opt/vc/include/VG/vgu.h
/opt/vc/include/WF
/opt/vc/include/WF/wfc.h
/opt/vc/include/WF/wfcplatform.h
/opt/vc/include/bcm_host.h
/opt/vc/include/interface
/opt/vc/include/interface/debug_sym
/opt/vc/include/interface/debug_sym/debug_sym.h
/opt/vc/include/interface/mmal
[... 175 lines cut ...]
/opt/vc/include/vcinclude/common.h
/opt/vc/include/vcinclude/vc_image_types.h
/opt/vc/include/vcinclude/vcore.h
/opt/vc/lib
/opt/vc/lib/libEGL_static.a
/opt/vc/lib/libGLESv2_static.a
/opt/vc/lib/libdebug_sym_static.a
/opt/vc/lib/libkhrn_client.a
/opt/vc/lib/libkhrn_static.a
/opt/vc/lib/libvchostif.a
/opt/vc/lib/libvcilcs.a
/opt/vc/lib/pkgconfig
/opt/vc/lib/pkgconfig/bcm_host.pc
/opt/vc/lib/pkgconfig/brcmegl.pc
/opt/vc/lib/pkgconfig/brcmglesv2.pc
/opt/vc/lib/pkgconfig/brcmvg.pc
/opt/vc/lib/pkgconfig/mmal.pc
/opt/vc/lib/pkgconfig/vcsm.pc
/usr
/usr/share
/usr/share/doc
/usr/share/doc/libraspberrypi-dev
/usr/share/doc/libraspberrypi-dev/changelog.Debian.gz
/usr/share/doc/libraspberrypi-dev/copyright
```

gaetan-puleo, 2021-11-09:
I got the same error

mdevaev, 2021-11-09:
Maintainers have removed the OMX library from the package. I wrote them a letter asking them to return it back.

mdevaev, 2021-11-09:
Bullseye deprecated OMX and MMAL so it won't work anymore. They [recommend](https://forums.raspberrypi.com/viewtopic.php?t=323281&start=100) switching to V4L2 hardware encoder, I will do it, but it will take time.
The work will be done here: #83

closed 2021-11-09 by mdevaev

The-EG, 2021-11-09:
Good to know, thanks!