pikvm/ustreamer #130: "Can't set input channel" on RaspberryPI 4 with USB cam

etwasmitbaum, 2021-11-07:
Hello,
I used an old laptop to run OctoPrint and stream a WebCam with ustreamer. I now decided to use a raspberryPI 4 und and installing ustreamer and testing it and I ran into a problem.
This is the command I am using this command to stream my webacm:
`/home/pi/ustreamer/ustreamer --device /dev/video0 --resolution 1280x720 --format MJPEG --desired-fps 25 --buffers 1 --workers 1 --encoder HW --host :: --port 8000`

I tried leaving out every unnecessary part of the command, same results.

And the output is this:
```
-- INFO  [335.655      main] -- Using internal blank placeholder
-- INFO  [335.656      main] -- Listening HTTP on [::]:8000
-- INFO  [335.657    stream] -- Using V4L2 device: /dev/video0
-- INFO  [335.657    stream] -- Using desired FPS: 25
-- INFO  [335.657      http] -- Starting HTTP eventloop ...
================================================================================
-- INFO  [335.659    stream] -- Device fd=8 opened
-- INFO  [335.659    stream] -- Using input channel: 0
-- ERROR [335.659    stream] -- Can't set input channel
-- INFO  [335.659    stream] -- Device fd=8 closed
-- INFO  [335.659    stream] -- Sleeping 1 seconds before new stream init ...
================================================================================
-- INFO  [336.660    stream] -- Device fd=8 opened
-- INFO  [336.661    stream] -- Using input channel: 0
-- ERROR [336.661    stream] -- Can't set input channel
-- INFO  [336.661    stream] -- Device fd=8 closed
-- INFO  [336.661    stream] -- Sleeping 1 seconds before new stream init ...
================================================================================
```
The Output repeats itself every second.

Here is the WebCam info from `v4l2-ctl --all -d /dev/video0`:
```
Driver Info:
        Driver name      : uvcvideo
        Card type        : USB PHY 2.0: USB CAMERA
        Bus info         : usb-0000:01:00.0-1.1
        Driver version   : 5.10.63
        Capabilities     : 0x84a00001
                Video Capture
                Metadata Capture
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x04200001
                Video Capture
                Streaming
                Extended Pix Format
Media Driver Info:
        Driver name      : uvcvideo
        Model            : USB PHY 2.0: USB CAMERA
        Serial           :
        Bus info         : usb-0000:01:00.0-1.1
        Media version    : 5.10.63
        Hardware revision: 0x00000100 (256)
        Driver version   : 5.10.63
Interface Info:
        ID               : 0x03000002
        Type             : V4L Video
Entity Info:
        ID               : 0x00000001 (1)
        Name             : USB PHY 2.0: USB CAMERA
        Function         : V4L2 I/O
        Flags         : default
        Pad 0x01000007   : 0: Sink
          Link 0x02000013: from remote pad 0x100000a of entity 'Extension 4': Data, Enabled, Immutable
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
        Width/Height      : 640/480
        Pixel Format      : 'MJPG' (Motion-JPEG)
        Field             : None
        Bytes per Line    : 0
        Size Image        : 4177920
        Colorspace        : sRGB
        Transfer Function : Rec. 709
        YCbCr/HSV Encoding: ITU-R 601
        Quantization      : Default (maps to Full Range)
        Flags             :
Crop Capability Video Capture:
        Bounds      : Left 0, Top 0, Width 640, Height 480
        Default     : Left 0, Top 0, Width 640, Height 480
        Pixel Aspect: 1/1
Selection: crop_default, Left 0, Top 0, Width 640, Height 480, Flags:
Selection: crop_bounds, Left 0, Top 0, Width 640, Height 480, Flags:
Streaming Parameters Video Capture:
        Capabilities     : timeperframe
        Frames per second: 25.000 (25/1)
        Read buffers     : 0
                     brightness 0x00980900 (int)    : min=0 max=255 step=1 default=128 value=128
                       contrast 0x00980901 (int)    : min=0 max=255 step=1 default=128 value=128
                     saturation 0x00980902 (int)    : min=0 max=255 step=1 default=128 value=128
                            hue 0x00980903 (int)    : min=-256 max=-32513 step=1 default=-32640 value=128
 white_balance_temperature_auto 0x0098090c (bool)   : default=0 value=0
    white_balance_red_component 0x0098090e (int)    : min=0 max=255 step=0 default=128 value=128
   white_balance_blue_component 0x0098090f (int)    : min=0 max=255 step=1 default=128 value=128
                          gamma 0x00980910 (int)    : min=0 max=255 step=10 default=128 value=128
                           gain 0x00980913 (int)    : min=0 max=255 step=1 default=128 value=128
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=0 value=0
                       hue_auto 0x00980919 (bool)   : default=0 value=0
      white_balance_temperature 0x0098091a (int)    : min=0 max=255 step=1 default=128 value=128
                      sharpness 0x0098091b (int)    : min=0 max=255 step=1 default=128 value=128
         backlight_compensation 0x0098091c (int)    : min=0 max=255 step=1 default=128 value=128
                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=0 value=0
              exposure_absolute 0x009a0902 (int)    : min=0 max=0 step=0 default=0 value=0 flags=inactive
         exposure_auto_priority 0x009a0903 (bool)   : default=0 value=0
                   pan_absolute 0x009a0908 (int)    : min=0 max=0 step=0 default=0 value=0
                  tilt_absolute 0x009a0909 (int)    : min=0 max=0 step=0 default=0 value=0
                 focus_absolute 0x009a090a (int)    : min=0 max=0 step=0 default=0 value=0
                     focus_auto 0x009a090c (bool)   : default=0 value=0
                  zoom_absolute 0x009a090d (int)    : min=0 max=0 step=0 default=0 value=0
                zoom_continuous 0x009a090f (int)    : min=0 max=0 step=0 default=0 value=0 flags=write-only
                        privacy 0x009a0910 (bool)   : default=0 value=0
                  iris_absolute 0x009a0911 (int)    : min=0 max=0 step=0 default=0 value=0
                  iris_relative 0x009a0912 (int)    : min=0 max=0 step=0 default=0 value=0 flags=write-only
                      pan_speed 0x009a0920 (int)    : min=0 max=0 step=0 default=0 value=0
                     tilt_speed 0x009a0921 (int)    : min=0 max=0 step=0 default=0 value=0
```

mdevaev, 2021-11-07:
Try disabling input selection in the code, rebuild it and run.
Comment these lines in `srv/ustreamer/device.c`:
https://github.com/pikvm/ustreamer/blob/master/src/ustreamer/device.c#L452-L457

etwasmitbaum, 2021-11-07:
This fixed the issue but got a new one.
```
================================================================================
-- INFO  [480.179    stream] -- Device fd=8 opened
-- ERROR [480.179    stream] -- Unable to set pixelformat=MJPEG, stride=2560, resolution=1280x720: Device or resource busy
-- INFO  [480.179    stream] -- Device fd=8 closed
-- INFO  [480.180    stream] -- Sleeping 1 seconds before new stream init ...
================================================================================
```
My Raspberry PI is fresh booted and I did not try to access the camera in any way.

mdevaev, 2021-11-07:
It doesn't look like this camera was working at all. It looks like a driver glitch.

etwasmitbaum, 2021-11-07:
I think the same.
Since this is no more related to this project I will close this issue.
Thanks for your help.

closed 2021-11-07 by etwasmitbaum

mdevaev, 2021-11-07:
If you find Linux software that works with it, let me know, I'll try to figure out what's wrong here.

etwasmitbaum, 2021-11-08:
Everything is working now. I used the default Raspberry Pi OS (32 bit) instead of OctoPI to run the stream.
Something was wrong with my installation of OctoPI.