pikvm/ustreamer #134: make WITH_OMX=1 WITH_JANUS=1 error plugin.h

TelDragon, 2021-12-03:
```
root@pi:~/ustreamer-master # make WITH_OMX=1 WITH_JANUS=1
make apps
make[1]: Entering directory '/root/ustreamer-master'
make -C src
make[2]: Entering directory '/root/ustreamer-master/src'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/root/ustreamer-master/src'
make[1]: Leaving directory '/root/ustreamer-master'
make janus
make[1]: Entering directory '/root/ustreamer-master'
make -C janus
make[2]: Entering directory '/root/ustreamer-master/janus'
-- CC src/base64.c
-- CC src/frame.c
-- CC src/plugin.c
In file included from src/plugin.c:38:
/usr/include/janus/plugins/plugin.h:160:10: fatal error: refcount.h: No such file or directory
 #include "refcount.h"
          ^~~~~~~~~~~~
compilation terminated.
make[2]: *** [Makefile:40: build/src/plugin.o] Error 1
make[2]: Leaving directory '/root/ustreamer-master/janus'
make[1]: *** [Makefile:49: janus] Error 2
make[1]: Leaving directory '/root/ustreamer-master'
make: *** [Makefile:29: all] Error 2
root@pi:~/ustreamer-master # apt-get install janus-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
janus-dev is already the newest version (0.6.1-1).
0 upgraded, 0 newly installed, 0 to remove and 55 not upgraded.
root@pi:~/ustreamer-master # cd /usr/include/janus/plugins/
root@pi:/usr/include/janus/plugins # ls
plugin.h
root@pi:/usr/include/janus/plugins # 

root@pi:/usr/include/janus # ls
apierror.h  config.h  debug.h  events  ip-utils.h  log.h  mutex.h  plugins  record.h  refcount.h  rtcp.h  rtp.h  rtpsrtp.h  sdp-utils.h  text2pcap.h  transports  utils.h
```

TelDragon, 2021-12-03:
Is this possible?

```
cp /usr/include/janus/ refcount.h log.h debug.h /usr/include/janus/plugins/
```

I tried, but the result is still an error. I can't make normally

TelDragon, 2021-12-03:
perhaps

```
sed -i -e 's|^#include "refcount.h"$|#include "../refcount.h"|g' "/usr/include/janus/plugins/plugin.h"
```
The effect is the same

```
src/plugin.c: In function ‘_relay_rtp_clients’:
src/plugin.c:144:2: error: unknown type name ‘janus_plugin_rtp’; did you mean ‘janus_plugin’?
  janus_plugin_rtp packet = {
  ^~~~~~~~~~~~~~~~
  janus_plugin
src/plugin.c:145:3: error: field name not in record or union initializer
   .video = true,
   ^
src/plugin.c:145:3: note: (near initialization for ‘packet’)
src/plugin.c:146:3: error: field name not in record or union initializer
   .buffer = (char *)datagram,
   ^
src/plugin.c:146:3: note: (near initialization for ‘packet’)
src/plugin.c:146:13: warning: excess elements in scalar initializer
   .buffer = (char *)datagram,
             ^
src/plugin.c:146:13: note: (near initialization for ‘packet’)
src/plugin.c:147:3: error: field name not in record or union initializer
   .length = size,
   ^
src/plugin.c:147:3: note: (near initialization for ‘packet’)
src/plugin.c:147:13: warning: excess elements in scalar initializer
   .length = size,
             ^~~~
src/plugin.c:147:13: note: (near initialization for ‘packet’)
src/plugin.c:149:2: warning: implicit declaration of function ‘janus_plugin_rtp_extensions_reset’; did you mean ‘janus_plugin_result_new’? [-Wimplicit-function-declaration]
  janus_plugin_rtp_extensions_reset(&packet.extensions);
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  janus_plugin_result_new
src/plugin.c:149:43: error: request for member ‘extensions’ in something not a structure or union
  janus_plugin_rtp_extensions_reset(&packet.extensions);
                                           ^
In file included from src/plugin.c:43:
src/plugin.c:152:38: warning: passing argument 2 of ‘_g_gw->relay_rtp’ makes integer from pointer without a cast [-Wint-conversion]
    _g_gw->relay_rtp(client->session, &packet);
                                      ^~~~~~~
src/list.h:35:4: note: in definition of macro ‘LIST_ITERATE’
    __VA_ARGS__ \
    ^~~~~~~~~~~
src/plugin.c:152:38: note: expected ‘int’ but argument is of type ‘int *’
    _g_gw->relay_rtp(client->session, &packet);
                                      ^~~~~~~
src/list.h:35:4: note: in definition of macro ‘LIST_ITERATE’
    __VA_ARGS__ \
    ^~~~~~~~~~~
src/plugin.c:152:4: error: too few arguments to function ‘_g_gw->relay_rtp’
    _g_gw->relay_rtp(client->session, &packet);
    ^~~~~
src/list.h:35:4: note: in definition of macro ‘LIST_ITERATE’
    __VA_ARGS__ \
    ^~~~~~~~~~~
make[2]: *** [Makefile:40: build/src/plugin.o] Error 1
make[2]: Leaving directory '/root/ustreamer-master/janus'
make[1]: *** [Makefile:49: janus] Error 2
make[1]: Leaving directory '/root/ustreamer-master'
make: *** [Makefile:29: all] Error 2
```

mdevaev, 2021-12-03:
You need to use at least janus version 0.11

TelDragon, 2021-12-03:
Thank you very much

I'll install it according to this library

https://github.com/meetecho/janus-gateway

```
libtool: finish: PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/sbin" ldconfig -n /usr/lib/janus/transports
----------------------------------------------------------------------
Libraries have been installed in:
   /usr/lib/janus/transports

If you ever happen to want to link against installed libraries
in a given directory, LIBDIR, you must either use libtool, and
specify the full pathname of the library, or use the '-LLIBDIR'
flag during linking and do at least one of the following:
   - add LIBDIR to the 'LD_LIBRARY_PATH' environment variable
     during execution
   - add LIBDIR to the 'LD_RUN_PATH' environment variable
     during linking
   - use the '-Wl,-rpath -Wl,LIBDIR' linker flag
   - have your system administrator add LIBDIR to '/etc/ld.so.conf'

See any operating system documentation about shared libraries for
more information, such as the ld(1) and ld.so(8) manual pages.
----------------------------------------------------------------------
 /usr/bin/mkdir -p '/usr/include/janus/transports'
 /usr/bin/install -c -m 644 transports/transport.h '/usr/include/janus/transports'
make[3]: Leaving directory '/root/janus-gateway-master'
make[2]: Leaving directory '/root/janus-gateway-master'
make[1]: Leaving directory '/root/janus-gateway-master'
```

Then modify it again

```
#include "../refcount.h"
```

Will get it perfectly
```
root@pi:~/ustreamer-master # make WITH_OMX=1 WITH_JANUS=1
make apps
make[1]: Entering directory '/root/ustreamer-master'
make -C src
make[2]: Entering directory '/root/ustreamer-master/src'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/root/ustreamer-master/src'
make[1]: Leaving directory '/root/ustreamer-master'
make janus
make[1]: Entering directory '/root/ustreamer-master'
make -C janus
make[2]: Entering directory '/root/ustreamer-master/janus'
-- CC src/plugin.c
-- CC src/rtp.c
== SO libjanus_ustreamer.so
make[2]: Leaving directory '/root/ustreamer-master/janus'
make[1]: Leaving directory '/root/ustreamer-master'
root@pi:~/ustreamer-master # ls
janus  libjanus_ustreamer.so  LICENSE  linters  Makefile  man  pkg  python  README.md  README.ru.md  src  tools  ustreamer  ustreamer-dump
root@pi:~/ustreamer-master # ls -ll
total 96
drwxr-xr-x 4 root root  4096 Dec  3 16:12 janus
lrwxrwxrwx 1 root root    27 Dec  3 16:12 libjanus_ustreamer.so -> janus/libjanus_ustreamer.so
-rw-r--r-- 1 root root 35147 Nov 28 13:43 LICENSE
drwxr-xr-x 2 root root  4096 Nov 28 13:43 linters
-rw-r--r-- 1 root root  2266 Nov 28 13:43 Makefile
drwxr-xr-x 2 root root  4096 Nov 28 13:43 man
drwxr-xr-x 6 root root  4096 Nov 28 13:43 pkg
drwxr-xr-x 3 root root  4096 Nov 28 13:43 python
-rw-r--r-- 1 root root  9385 Nov 28 13:43 README.md
-rw-r--r-- 1 root root 14514 Nov 28 13:43 README.ru.md
drwxr-xr-x 6 root root  4096 Dec  3 16:12 src
drwxr-xr-x 2 root root  4096 Nov 28 13:43 tools
lrwxrwxrwx 1 root root    17 Dec  3 16:12 ustreamer -> src/ustreamer.bin
lrwxrwxrwx 1 root root    22 Dec  3 16:12 ustreamer-dump -> src/ustreamer-dump.bin
```

mdevaev, 2021-12-03:
Glad to hear it.

closed 2021-12-03 by mdevaev