pikvm/ustreamer #135: Can't make it on raspberry pi zero

thomiz, 2021-12-12:
I am pretty shure it worked a year ago, but when I try to install ustreamer on another pi-zero today I only get this trying to make ustreamer:

~~~bash
In file included from ustreamer/http/bev.c:23:
ustreamer/http/bev.h:28:10: fatal error: event2/util.h: Ingen slik fil eller filkatalog
   28 | #include <event2/util.h>
      |          ^~~~~~~~~~~~~~~
~~~

mdevaev, 2021-12-12:
Apparently, some packages are missing

thomiz, 2021-12-12:
I did the usual git clone so everything should be there right?

mdevaev, 2021-12-12:
No. Please follow README guide and install all necessary packages.

thomiz, 2021-12-12:
Will do, probably OMX og GPIO missing then.

mdevaev, 2021-12-12:
libevent also

thomiz, 2021-12-12:
Got it, after installing all the dependencies the make goes OK. Sorry.

closed 2021-12-12 by thomiz