pikvm/ustreamer #145: I have a question like this, is but I don't know what "vcsm", Can you help me out

waveng, 2022-02-24:
Hi,

I have a question like this, is but I don't know what "vcsm", Can you help me out

```
In file included from ustreamer/device.c:23:
ustreamer/device.h:44:11: fatal error: interface/vcsm/user-vcsm.h: No such file or directory
   44 | # include <interface/vcsm/user-vcsm.h>
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [Makefile:112: build/ustreamer/device.o] Error 1
make[2]: Leaving directory '/home/pi/ustreamer-4.13/src'
make[1]: *** [Makefile:38: apps] Error 2
make[1]: Leaving directory '/home/pi/ustreamer-4.13'
make: *** [Makefile:28: all] Error 2

````

mdevaev, 2022-02-24:
Hello. [Bullseye threw out some libraries and broke compatibility](https://github.com/pikvm/ustreamer/issues/123). I'm working on a 5.0 release that will fix this problem. For now, I recommend that you don't use the WITH_OMX flag.

closed 2022-02-24 by mdevaev

waveng, 2022-02-24:
yes goody, Looking forward to 5.0