pikvm/ustreamer #146: Device doesn't support setting of HW encoding quality parameters

dicamarques14, 2022-02-24:
I'm trying to stream a usb hdmi capture card that outputs in MJPEG.
I'm using a pi B+ with dietPi OS.

I built the m2m branch and got these errors:

` ./ustreamer --format=mjpeg  -r 1280x720 -c M2M-VIDEO --device=/dev/video0 --host=0.0.0.0 --port=8080 --persistent --drop-same-frames=30 -f 30

-- INFO  [4828.692      main] -- Using internal blank placeholder
-- INFO  [4828.696      main] -- Listening HTTP on [0.0.0.0]:8080
-- INFO  [4828.699      http] -- Starting HTTP eventloop ...
-- INFO  [4828.702    stream] -- Using V4L2 device: /dev/video0
-- INFO  [4828.705    stream] -- Using desired FPS: 30
================================================================================
-- INFO  [4828.719    stream] -- Device fd=8 opened
-- INFO  [4828.725    stream] -- Using resolution: 1280x720
-- INFO  [4828.725    stream] -- Using format: MJPEG
-- ERROR [4828.730    stream] -- Invalid HW FPS numerator: 333333 != 1
-- ERROR [4828.731    stream] -- Device doesn't support setting of HW encoding quality parameters
-- INFO  [4828.732    stream] -- Using IO method: MMAP
-- INFO  [4828.896    stream] -- Requested 2 device buffers, got 2
-- ERROR [4828.899    stream] -- Can't map device buffer=0: No such device
-- INFO  [4828.949    stream] -- Device fd=8 closed
-- INFO  [4828.949    stream] -- Sleeping 1 seconds before new stream init ...
`

Also, should i use the HW encoder or the M2M-ViDEO. Not sure since my input is already in MJPEG

mdevaev, 2022-02-25:
Try `./ustreamer --format=uyvy --dv-timings -c M2M-VIDEO --device=/dev/video0 --host=0.0.0.0 --port=8080 --persistent --drop-same-frames=30 -f 30`

dicamarques14, 2022-02-25:
I tried with both --format= mjpeg and uyvy ( I must use the mjpeg mode) and i got:

``` 
-- INFO  [495.760    stream] -- Using desired FPS: 0
================================================================================
-- INFO  [495.773    stream] -- Device fd=8 opened
-- INFO  [495.773    stream] -- Using input channel: 0
-- ERROR [495.776    stream] -- Can't subscribe to DV-timings events: Invalid argument
-- INFO  [495.778    stream] -- Device fd=8 closed
-- INFO  [495.779    stream] -- Sleeping 1 seconds before new stream init ...

```

mdevaev, 2022-03-15:
What a device are you using?

dicamarques14, 2022-03-16:
> What a device are you using?

Raspberry PI 1 B+

mdevaev, 2022-03-16:
Try `./ustreamer --format=jpeg --device=/dev/video0 --host=0.0.0.0 --port=8080 --drop-same-frames=30`

dicamarques14, 2022-03-17:
Sorry to say but it didn't work
```
 INFO  [499.663      main] -- Using internal blank placeholder
-- INFO  [499.666      main] -- Listening HTTP on [0.0.0.0]:8080
-- INFO  [499.669      http] -- Starting HTTP eventloop ...
-- INFO  [499.671    stream] -- Using V4L2 device: /dev/video0
-- INFO  [499.672    stream] -- Using desired FPS: 0
================================================================================
-- INFO  [499.690    stream] -- Device fd=8 opened
-- INFO  [499.692    stream] -- Using input channel: 0
-- ERROR [499.693    stream] -- Can't set device format: Invalid argument
-- INFO  [499.696    stream] -- Device fd=8 closed
-- INFO  [499.697    stream] -- Sleeping 1 seconds before new stream init ...

```

mdevaev, 2022-03-17:
I think this is a driver issue.

closed 2022-07-20 by mdevaev