pikvm/ustreamer #147: FEAT: Ability to exec command when client joins

aurimasniekis, 2022-03-04:
At the moment because ustreamer doesn't support multiple different formats from v4l2 supported, I am running akvcam with FFmpeg to convert to a suitable one and everything is fine, but it means that FFmpeg is running nonstop even if there are no clients.

It would be nice if there was a possibility to have some exec functions on when there are clients aka the first client joined and then all clients left, so I could boot/kill the FFmpeg without wasting resources in the background.

mdevaev, 2022-03-15:
Do you put ffmpeg after ustreamer? I'm not sure that I quite understant the chain.

mdevaev, 2022-04-01:
Sup?

mdevaev, 2022-04-16:
No answer, so I'm closing it. Let me know if you're still interested.

closed 2022-04-16 by mdevaev