pikvm/ustreamer #151: My Stream isn´t starting! -- ERROR [1717.524    stream] -- Mainloop select() error: Success

eliasstepanik, 2022-04-17:
My stream is not starting with usb camera!

./ustreamer --device=/dev/v4l/by-id/usb-Generic_HD_camera-video-index0 --host=0.0.0.0 --port=8081

-- INFO  [1901.680    stream] -- Device fd=8 opened
-- INFO  [1901.680    stream] -- Using input channel: 0
-- INFO  [1901.681    stream] -- Using resolution: 640x480
-- INFO  [1901.682    stream] -- Using format: YUYV
-- INFO  [1901.683    stream] -- Using HW FPS: 0 -> 30 (coerced)
-- INFO  [1901.683    stream] -- Using IO method: MMAP
-- INFO  [1901.689    stream] -- Requested 5 device buffers, got 5
-- INFO  [1901.694    stream] -- Capturing started
-- INFO  [1901.694    stream] -- Using JPEG quality: 80%
-- INFO  [1901.694    stream] -- Creating pool JPEG with 4 workers ...
-- INFO  [1901.695    stream] -- Capturing ...
-- ERROR [1902.696    stream] -- Mainloop select() error: Input/output error
-- INFO  [1902.696    stream] -- Destroying workers pool JPEG ...
-- INFO  [1902.699    stream] -- Capturing stopped
-- INFO  [1902.702    stream] -- Device fd=8 closed

SirCipherz, 2022-04-21:
i have the same error

mdevaev, 2022-04-21:
Please try option `-m jpeg` or `-m mjpeg`.

eliasstepanik, 2022-04-21:
I can only use yuvv

jdfalk, 2022-05-13:
Same here. I am using a Razor Kiyo Pro. 

I've tried:
`ustreamer -s 0.0.0.0 -p 10008 -d /dev/v4l/by-id/usb-Razer_Inc_Razer_Kiyo_Pro-video-index0 -f 60 -m MJPEG -q 100 -c HW -n` -- and every variation of this, leaving off `-c -q -n`
`ustreamer -s 0.0.0.0 -p 10008 -d /dev/v4l/by-id/usb-Razer_Inc_Razer_Kiyo_Pro-video-index0 -f 30 -m YUYV`
`ustreamer -s 0.0.0.0 -p 10008 -d /dev/v4l/by-id/usb-Razer_Inc_Razer_Kiyo_Pro-video-index0 -f 30 -m JPEG`

`v4l2-ctl --all -d /dev/v4l/by-id/usb-Razer_Inc_Razer_Kiyo_Pro-video-index0:`
```
Driver Info:
	Driver name      : uvcvideo
	Card type        : Razer Kiyo Pro
	Bus info         : usb-0000:04:00.0-2.3
	Driver version   : 5.4.181
	Capabilities     : 0x84a00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Media Driver Info:
	Driver name      : uvcvideo
	Model            : Razer Kiyo Pro
	Serial           :
	Bus info         : usb-0000:04:00.0-2.3
	Media version    : 5.4.181
	Hardware revision: 0x00000821 (2081)
	Driver version   : 5.4.181
Interface Info:
	ID               : 0x03000002
	Type             : V4L Video
Entity Info:
	ID               : 0x00000001 (1)
	Name             : Razer Kiyo Pro
	Function         : V4L2 I/O
	Flags         : default
	Pad 0x01000007   : 0: Sink
	  Link 0x02000013: from remote pad 0x100000a of entity 'Extension 2': Data, Enabled, Immutable
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
	Width/Height      : 1920/1080
	Pixel Format      : 'MJPG' (Motion-JPEG)
	Field             : None
	Bytes per Line    : 0
	Size Image        : 4147200
	Colorspace        : sRGB
	Transfer Function : Default (maps to sRGB)
	YCbCr/HSV Encoding: Default (maps to ITU-R 601)
	Quantization      : Default (maps to Full Range)
	Flags             :
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 1920, Height 1080
	Default     : Left 0, Top 0, Width 1920, Height 1080
	Pixel Aspect: 1/1
Selection Video Capture: crop_default, Left 0, Top 0, Width 1920, Height 1080, Flags:
Selection Video Capture: crop_bounds, Left 0, Top 0, Width 1920, Height 1080, Flags:
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 60.000 (60/1)
	Read buffers     : 0
                     brightness 0x00980900 (int)    : min=0 max=255 step=1 default=128 value=128
                       contrast 0x00980901 (int)    : min=0 max=255 step=1 default=128 value=128
                     saturation 0x00980902 (int)    : min=0 max=255 step=1 default=128 value=128
 white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
                           gain 0x00980913 (int)    : min=0 max=255 step=1 default=0 value=0
[... 1100 lines cut ...]
-- INFO  [78415.184    stream] -- Capturing started
-- INFO  [78415.184    stream] -- Using JPEG quality: 80%
-- INFO  [78415.184    stream] -- Creating pool JPEG with 4 workers ...
-- DEBUG [78415.184      jw-0] -- Hello! I am a worker jw-0 ^_^
-- DEBUG [78415.184      jw-0] -- Worker jw-0 waiting for a new job ...
-- DEBUG [78415.184      jw-1] -- Hello! I am a worker jw-1 ^_^
-- DEBUG [78415.184      jw-2] -- Hello! I am a worker jw-2 ^_^
-- DEBUG [78415.184      jw-2] -- Worker jw-2 waiting for a new job ...
-- DEBUG [78415.184      jw-3] -- Hello! I am a worker jw-3 ^_^
-- DEBUG [78415.184      jw-3] -- Worker jw-3 waiting for a new job ...
-- DEBUG [78415.184      jw-1] -- Worker jw-1 waiting for a new job ...
-- INFO  [78415.184    stream] -- Capturing ...
--------------------------------------------------------------------------------
-- DEBUG [78415.184    stream] -- Waiting for worker ...
-- DEBUG [78415.184    stream] -- Calling select() on video device ...
-- DEBUG [78415.952      http] -- HTTP: Repeating exposed ...
-- DEBUG [78416.953      http] -- HTTP: Repeating exposed ...
-- DEBUG [78417.968      http] -- HTTP: Repeating exposed ...
-- DEBUG [78418.969      http] -- HTTP: Repeating exposed ...
-- DEBUG [78419.985      http] -- HTTP: Repeating exposed ...
-- DEBUG [78421.001      http] -- HTTP: Repeating exposed ...
-- DEBUG [78422.019      http] -- HTTP: Repeating exposed ...
-- DEBUG [78423.035      http] -- HTTP: Repeating exposed ...
-- DEBUG [78424.051      http] -- HTTP: Repeating exposed ...
```

charlesrg, 2022-05-16:
Same issue here with a Logitech Brio 4K, similar to the Razor camera
Tried many options, including increasing device timeout -m mjpeg / MJPEG.
The strange part is that ffmpeg and stream also fail.

 ffmpeg -y -f v4l2 -video_size 1280x720 -i /dev/video1 -r 0.2 -qscale:v 2 -update 1 /tmp/webcam.jpg
 streamer  -c /dev/video1 -f rgb24 -r 3 -t 00:30:00 -o ./outfile.avi

`root@octoprint:~# ustreamer --device=/dev/video1 --device-timeout 5 --slowdown -r 640x480   --host 0.0.0.0
-- INFO  [1949.188      main] -- Using internal blank placeholder
-- INFO  [1949.189      main] -- Listening HTTP on [0.0.0.0]:8080
-- INFO  [1949.189    stream] -- Using V4L2 device: /dev/video1
-- INFO  [1949.190    stream] -- Using desired FPS: 0
-- INFO  [1949.190      http] -- Starting HTTP eventloop ...
================================================================================
-- INFO  [1949.191    stream] -- Device fd=8 opened
-- INFO  [1949.191    stream] -- Using input channel: 0
-- INFO  [1949.193    stream] -- Using resolution: 640x480
-- INFO  [1949.193    stream] -- Using format: YUYV
-- INFO  [1949.196    stream] -- Using HW FPS: 0 -> 30 (coerced)
-- INFO  [1949.196    stream] -- Using IO method: MMAP
-- INFO  [1949.198    stream] -- Requested 5 device buffers, got 5
-- INFO  [1950.035    stream] -- Capturing started
-- INFO  [1950.035    stream] -- Using JPEG quality: 80%
-- INFO  [1950.035    stream] -- Creating pool JPEG with 4 workers ...
-- INFO  [1950.036    stream] -- Capturing ...
-- ERROR [1955.542    stream] -- Mainloop select() error: Success
-- INFO  [1955.542    stream] -- Destroying workers pool JPEG ...
-- INFO  [1955.671    stream] -- Capturing stopped
-- INFO  [1955.674    stream] -- Device fd=8 closed
================================================================================
-- INFO  [1955.675    stream] -- Device fd=8 opened
-- INFO  [1955.675    stream] -- Using input channel: 0
-- INFO  [1955.678    stream] -- Using resolution: 640x480
-- INFO  [1955.678    stream] -- Using format: YUYV
-- INFO  [1955.680    stream] -- Using HW FPS: 0 -> 30 (coerced)
-- INFO  [1955.680    stream] -- Using IO method: MMAP
-- INFO  [1955.683    stream] -- Requested 5 device buffers, got 5
-- INFO  [1956.534    stream] -- Capturing started
-- INFO  [1956.535    stream] -- Using JPEG quality: 80%
-- INFO  [1956.535    stream] -- Creating pool JPEG with 4 workers ...
-- INFO  [1956.535    stream] -- Capturing ...
-- ERROR [1962.041    stream] -- Mainloop select() error: Success
-- INFO  [1962.041    stream] -- Destroying workers pool JPEG ...
-- INFO  [1962.171    stream] -- Capturing stopped
-- INFO  [1962.174    stream] -- Device fd=8 closed
================================================================================
-- INFO  [1962.175    stream] -- Device fd=8 opened
`

mdevaev, 2022-05-16:
> The strange part is that ffmpeg and stream also fail.

A driver issue maybe?

jdfalk, 2022-05-17:
Well that's incredibly frustrating as I got it explicitly for it's USB 3 capabilities as I'm currently flooding the usb2 bus when I use it. Guess I'm just going to have to setup a pi to host this single camera. Thanks anyways I appreciate you looking into it. If you happen to know any USB3 cameras that do work well with ustreamer I'd appreciate the info. Thanks again.

closed 2022-05-17 by mdevaev