pikvm/ustreamer #154: Adding h264 sink causing issues (hanging on exit, no signal on regular stream)

cp2004, 2022-05-03:
Thanks for creating ustreamer! It is working very well for me to stream regular mjpg video to OctoPrint at the moment.

I'm trying to setup webrtc streaming for the efficient bandwidth usage that comes with it, but hitting a few issues. I've tried to follow the new [h264 guide](https://github.com/pikvm/ustreamer/blob/master/docs/h264.md) and everything is built and setup correctly.

When I run this command:

```
./ustreamer -m MJPEG -d /dev/video0 
```
Everything is beautiful and I can access the stream for mjpg fine.

When trying to add the h264 sink options:
```
./ustreamer -m MJPEG -d /dev/video0 --h264-sink=demo::ustreamer::h264 --h264-sink-mode=660
```

Results in some errors:
```
-- INFO  [33623.629      main] -- Using internal blank placeholder
-- INFO  [33623.629      main] -- Using H264-sink: demo::ustreamer::h264
-- INFO  [33623.630      main] -- Listening HTTP on [127.0.0.1]:8080
-- INFO  [33623.631    stream] -- Using V4L2 device: /dev/video0
-- INFO  [33623.631    stream] -- Using desired FPS: 0
-- INFO  [33623.631      http] -- Starting HTTP eventloop ...
-- INFO  [33623.631    stream] -- H264: Initializing encoder ...
-- INFO  [33623.659    stream] -- H264: Configuring encoder: DMA=0 ...
================================================================================
-- INFO  [33624.372    stream] -- Device fd=10 opened
-- INFO  [33624.372    stream] -- Using input channel: 0
-- INFO  [33624.395    stream] -- Using resolution: 640x480
-- INFO  [33624.396    stream] -- Using format: MJPEG
-- INFO  [33624.419    stream] -- Using HW FPS: 0 -> 30 (coerced)
-- ERROR [33624.419    stream] -- Device doesn't support setting of HW encoding quality parameters
-- INFO  [33624.419    stream] -- Using IO method: MMAP
-- INFO  [33624.423    stream] -- Requested 5 device buffers, got 5
-- INFO  [33624.437    stream] -- Capturing started
-- INFO  [33624.437    stream] -- Switching to HW encoder: the input is (M)JPEG ...
-- INFO  [33624.437    stream] -- Using JPEG quality: encoder default
-- INFO  [33624.437    stream] -- Creating pool JPEG with 1 workers ...
-- INFO  [33624.438    stream] -- Capturing ...
-- ERROR [33625.439    stream] -- Mainloop select() error: Inappropriate ioctl for device
-- INFO  [33625.439    stream] -- Destroying workers pool JPEG ...
-- INFO  [33625.442    stream] -- Capturing stopped
-- INFO  [33625.444    stream] -- Device fd=10 closed
================================================================================
-- INFO  [33625.477    stream] -- Device fd=10 opened
-- INFO  [33625.477    stream] -- Using input channel: 0
-- INFO  [33626.239    stream] -- Using resolution: 640x480
-- INFO  [33626.240    stream] -- Using format: MJPEG
-- INFO  [33626.271    stream] -- Using HW FPS: 0 -> 30 (coerced)
-- ERROR [33626.271    stream] -- Device doesn't support setting of HW encoding quality parameters
-- INFO  [33626.271    stream] -- Using IO method: MMAP
-- INFO  [33626.275    stream] -- Requested 5 device buffers, got 5
-- INFO  [33626.288    stream] -- Capturing started
-- INFO  [33626.288    stream] -- Switching to HW encoder: the input is (M)JPEG ...
-- INFO  [33626.288    stream] -- Using JPEG quality: encoder default
-- INFO  [33626.288    stream] -- Creating pool JPEG with 1 workers ...
-- INFO  [33626.288    stream] -- Capturing ...
-- ERROR [33626.849    stream] -- Can't decompress JPEG: Huffman table 0x00 was not defined
[..... many more ......]
-- ERROR [33630.897    stream] -- Can't decompress JPEG: Huffman table 0x00 was not defined
^C-- INFO  [33630.954      main] -- ===== Stopping by SIGINT =====
-- INFO  [33630.954      http] -- HTTP eventloop stopped
-- ERROR [33630.961    stream] -- Can't decompress JPEG: Huffman table 0x00 was not defined
-- INFO  [33630.961    stream] -- Destroying workers pool JPEG ...
-- INFO  [33630.964    stream] -- Capturing stopped
-- INFO  [33630.966    stream] -- Device fd=10 closed
-- INFO  [33630.966    stream] -- H264: Destroying encoder ...
-- INFO  [33630.971      main] -- Bye-bye
``` 
[... 581 lines cut ...]
                Size: Discrete 1184x656
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 1280x960
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
    ```
    </details>

Let me know what you think, if you want me to run anything else or answer any questions I will do what I can.

mdevaev, 2022-07-05:
Hello. Sorry for the long answer, I'll try to figure it out, this is an unusual case.

cp2004, 2022-09-23:
> Hi @cp2004 , sorry to hijack this issue. 

You are correct, this is not the right place to discuss random user questions.

If you need support with OctoPrint, I suggest you seek it on the [community forums](https://community.octoprint.org) or [Discord server](https://discord.octoprint.org), the official support channels.

aguinaldoabbj, 2022-09-23:
> You are correct, this is not the right place to discuss random user questions.
>
> If you need support with OctoPrint, I suggest you seek it on the [community forums](https://community.octoprint.org) [...]

How polite and helpful of you!

mdevaev, 2024-02-03:
I've never been able to reproduce it, and there have been no new messages, so I'm closing it. Let me know if you still have this problem reproducing.

mdevaev, 2024-03-14:
I've reworked sinks logic so it should be fixed.

closed 2024-03-14 by mdevaev