pikvm/ustreamer #158: Fatal Error on Raspbian Bullseye

gallitim, 2022-06-09:
make apps
make[1]: Entering directory '/home/pi/ustreamer'
make -C src
make[2]: Entering directory '/home/pi/ustreamer/src'
-- CC ustreamer/http/bev.c
In file included from ustreamer/http/bev.c:23:
ustreamer/http/bev.h:28:10: fatal error: event2/util.h: No such file or directory
   28 | #include <event2/util.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [Makefile:100: build/ustreamer/http/bev.o] Error 1
make[2]: Leaving directory '/home/pi/ustreamer/src'
make[1]: *** [Makefile:38: apps] Error 2
make[1]: Leaving directory '/home/pi/ustreamer'
make: *** [Makefile:28: all] Error 2

Getting this error when running the make command, just want to make sure that this is a compatibility issue with the 64-bit OS rather than a problem with my raspberry pi itself. I am running 64-bit Bullseye on a pi 4, and looking for an http streamer that is compatible with the new camera stack. Is ustreamer compatible or is there a certain parameter I need to specify when running the make command to get it to work? Any help would be greatly appreciated.

gallitim, 2022-06-09:
My mistake this is the error I wanted to know about:

-- INFO  [1170.329    stream] -- Device fd=8 opened
-- INFO  [1170.330    stream] -- Using input channel: 0
-- ERROR [1170.330    stream] -- Can't set device format: Device or resource busy
-- INFO  [1170.330    stream] -- Device fd=8 closed
-- INFO  [1170.330    stream] -- Sleeping 1 seconds before new stream init ...
^C-- INFO  [1171.200      main] -- ===== Stopping by SIGINT =====
-- INFO  [1171.200      http] -- HTTP eventloop stopped
-- INFO  [1171.331      main] -- Bye-bye

Is this because I don't have the legacy stack activated?

mdevaev, 2022-06-11:
Hello. Which camera do you use?

gallitim, 2022-06-13:
I use the CSI Raspberry Pi Camera V2, IMX 219 sensor.

mdevaev, 2022-06-16:
Please try the ustreamer fork from this thread: https://github.com/pikvm/ustreamer/issues/139

ustreamer does not work with libcamera and is designed for regular v4l2 devices. At some point I will merge the code with what is presented in the neighboring issue, but I don't have time for that right now. Please try the solutions described there and let us know in that issue if it helped