pikvm/ustreamer #160: Exposure, white balance, etc only being set once per stream

jmattingley23, 2022-07-04:
First off I just want to say thank you for making this service.  The performance improvements over mjpg-streamer are remarkable and I recommend it to everyone.

The issue I'm having is that certain image parameters - namely the brightness, gain, and white balance - only seem to be getting set once on cold boot of the service and are not adjusted ever again.  I'm using the service to remotely monitor a 3D printer and take timelapses, and sometimes these sessions can go on for 15+ hours across many different times of day, resulting in a big difference in image quality.  If I boot up the device in the morning, by the time it gets to night time the image is almost completely black and unusable, since the fixed exposure was set for a bright sunny morning.  The opposite is also true, if I start in the evening and let it run overnight, the image is completely blown and color shifted once it becomes light outside again.  As soon as I kill the stream and restart it, the image is always perfect, even in low light conditions.

Here are some example images to illustrate this problem.  The first image was taken from an image stream that was started the night before and became blown out in the morning.  The second image was taken after rebooting the stream - the two images were taken less than a minute apart.
[image]
[image]

I'm using a picam connected to a RPI4 with the latest release of ustreamer.  My launch parameters are:
`ustreamer --process-name-prefix ustreamer --log-level 0 --device /dev/video0 --drop-same-frames=30 --brightness=auto --hue=auto --white-balance=auto --gain=auto --encoder=HW --resolution=2592x1944`

Is it possible to have ustreamer continuously re-evaluate the image parameters to always ensure a good looking capture?  Not sure if this would have too much of a performance impact or if this is impossible due to some hardware limitation with the camera module.

mdevaev, 2022-07-05:
This is interesting, it seemed to me that setting the AUTO value for the camera parameters means that the driver or hardware itself will monitor the optimal settings for the image. That is, AUTO means switching the mode, and not the constant selection of values by the software. Actually, it seems to me that this problem should be solved in the driver, but I don't want to leave you without a solution, so I have one idea. You can use v4l2-ctl to refresh the camera settings, so no need to restart ustreamer at all. Check the list of controls (`v4l2-ctl -l -d /dev/video0`) and set it periodically using systemd-timer or something like this (`v4l2-ctl --set-ctrl=white_balance_automatic=1`). Add the other controls and your problem will be solved I guess.

jmattingley23, 2022-07-05:
Gotcha, yeah I wasn't quite sure if this issue was in scope or not but I figured I would ask just in case.  Appreciate you brainstorming some more potential solutions - I'll try these out and see if I can get something working.  Thanks!

closed 2022-07-05 by jmattingley23

mdevaev, 2022-07-05:
You're welcome!