pikvm/ustreamer #162: Can't build with Ubuntu 20.04

thermodyn, 2022-07-29:
Good afternoon

I am trying to install on ubuntu 20.04 and I am running into the following error when I try to compile.

I think I am missing a depenancy but for the life of me I cant figure out which one.

```
make apps
make[1]: Entering directory '/opt/ustreamer'
make -C src
make[2]: Entering directory '/opt/ustreamer/src'
== LD ustreamer.bin
/usr/bin/ld: build/ustreamer/main.o: in function `_signal_handler':
main.c:(.text+0x15b): undefined reference to `sigabbrev_np'
/usr/bin/ld: main.c:(.text+0x251): undefined reference to `sigabbrev_np'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:89: ustreamer.bin] Error 1
make[2]: Leaving directory '/opt/ustreamer/src'
make[1]: *** [Makefile:38: apps] Error 2
make[1]: Leaving directory '/opt/ustreamer'
make: *** [Makefile:28: all] Error 2
```

mdevaev, 2022-07-29:
Please try to build the master branch

thermodyn, 2022-07-30:
Build fine now .. thank you for the superfast fix :)

mdevaev, 2022-07-30:
No problem)

commit 3d7fb8c8 (2022-07-29): fixed #162: optional sigabbrev_np()
closed 2022-07-30 by mdevaev