pikvm/ustreamer #164: "Unable to start capturing: Input/output error" with generic USB webcam

pyunramura, 2022-08-06:
I'm getting an error when testing ustreamer. Specifically the error `Unable to start capturing: Input/output error`. The error doesn't seem to tell me much, so I'm hoping I can get some insight into the problem, or at least some corrective steps I can try. Any help would be greatly appreciated.

Ustreamer version: 4.8

Video device: Generic "Amazon special" 1080p webcam

Host: Raspberry Pi 4

OS: Raspbian 10 "Buster"

With command: `./ustreamer --device /dev/video2 --resolution 640x480 --format YUYV --desired-fps 30 --encoder CPU --host 0.0.0.0 --port 8080 --debug`

Debug Logs:

```
-- INFO  [2139.616    stream] -- Device fd=8 opened
-- DEBUG [2139.617    stream] -- Calling ioctl(VIDIOC_QUERYCAP) ...
-- INFO  [2139.617    stream] -- Using input channel: 0
-- DEBUG [2139.617    stream] -- Using TV standard: DEFAULT
-- DEBUG [2139.617    stream] -- Calling ioctl(VIDIOC_S_FMT) ...
-- DEBUG [2139.619      http] -- HTTP: Updating exposed frame (online=0) ...
-- VERB  [2139.619      http] -- HTTP: Exposed frame: online=0, exp_time=0.000000
-- INFO  [2139.625    stream] -- Using resolution: 640x480
-- INFO  [2139.625    stream] -- Using pixelformat: YUYV
-- DEBUG [2139.625    stream] -- Calling ioctl(VIDIOC_G_PARM) ...
-- INFO  [2139.634    stream] -- Using HW FPS: 30
-- INFO  [2139.634    stream] -- Using IO method: MMAP
-- DEBUG [2139.634    stream] -- Calling ioctl(VIDIOC_REQBUFS) for V4L2_MEMORY_MMAP ...
-- INFO  [2139.642    stream] -- Requested 5 device buffers, got 5
-- DEBUG [2139.642    stream] -- Allocating device buffers ...
-- DEBUG [2139.642    stream] -- Calling ioctl(VIDIOC_QUERYBUF) for device buffer 0 ...
-- DEBUG [2139.642    stream] -- Mapping device buffer 0 ...
-- DEBUG [2139.643    stream] -- Calling ioctl(VIDIOC_QUERYBUF) for device buffer 1 ...
-- DEBUG [2139.643    stream] -- Mapping device buffer 1 ...
-- DEBUG [2139.644    stream] -- Calling ioctl(VIDIOC_QUERYBUF) for device buffer 2 ...
-- DEBUG [2139.644    stream] -- Mapping device buffer 2 ...
-- DEBUG [2139.644    stream] -- Calling ioctl(VIDIOC_QUERYBUF) for device buffer 3 ...
-- DEBUG [2139.644    stream] -- Mapping device buffer 3 ...
-- DEBUG [2139.645    stream] -- Calling ioctl(VIDIOC_QUERYBUF) for device buffer 4 ...
-- DEBUG [2139.645    stream] -- Mapping device buffer 4 ...
-- DEBUG [2139.646    stream] -- Calling ioctl(VIDIOC_QBUF) for buffer 0 ...
-- DEBUG [2139.646    stream] -- Calling ioctl(VIDIOC_QBUF) for buffer 1 ...
-- DEBUG [2139.646    stream] -- Calling ioctl(VIDIOC_QBUF) for buffer 2 ...
-- DEBUG [2139.646    stream] -- Calling ioctl(VIDIOC_QBUF) for buffer 3 ...
-- DEBUG [2139.647    stream] -- Calling ioctl(VIDIOC_QBUF) for buffer 4 ...
-- DEBUG [2139.647    stream] -- Device fd=8 initialized
-- DEBUG [2139.647    stream] -- Calling ioctl(VIDIOC_STREAMON) ...
-- ERROR [2139.650    stream] -- Unable to start capturing: Input/output error
-- DEBUG [2139.650    stream] -- Releasing device buffers ...
-- DEBUG [2139.651    stream] -- Closing device ...
-- INFO  [2139.653    stream] -- Device fd=8 closed
-- INFO  [2139.653    stream] -- Sleeping 1 seconds before new stream init ...
```

Command: `v4l2-ctl --list-formats-ext -d /dev/video2`
```
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'MJPG' (Motion-JPEG, compressed)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                Size: Discrete 480x272
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                Size: Discrete 640x360
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                Size: Discrete 1920x1080
[... 76 lines cut ...]
                Size: Discrete 1024x576
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
        [2]: 'YUYV' (YUYV 4:2:2)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                Size: Discrete 480x272
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                Size: Discrete 640x360
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                Size: Discrete 800x600
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
```

mdevaev, 2022-08-06:
Try `--format=MJPEG --encoder=HW`.

pyunramura, 2022-08-06:
Just gave that a shot, but no change unfortunately.

Command: `./ustreamer --device /dev/video2 --resolution 640x480 --format MJPEG --encoder=HW --desired-
fps 30 --host 0.0.0.0 --port 8080 --debug`

Debug logs:
```
-- INFO  [57164.194    stream] -- Device fd=8 opened
-- DEBUG [57164.194    stream] -- Calling ioctl(VIDIOC_QUERYCAP) ...
-- INFO  [57164.194    stream] -- Using input channel: 0
-- DEBUG [57164.194    stream] -- Using TV standard: DEFAULT
-- DEBUG [57164.194    stream] -- Calling ioctl(VIDIOC_S_FMT) ...
-- INFO  [57164.201    stream] -- Using resolution: 640x480
-- INFO  [57164.201    stream] -- Using pixelformat: MJPEG
-- DEBUG [57164.201    stream] -- Calling ioctl(VIDIOC_G_PARM) ...
-- DEBUG [57164.206      http] -- HTTP: Updating exposed frame (online=0) ...
-- VERB  [57164.206      http] -- HTTP: Exposed frame: online=0, exp_time=0.000000
-- INFO  [57164.212    stream] -- Using HW FPS: 30
-- ERROR [57164.212    stream] -- Device does not support setting of HW encoding quality parameters
-- INFO  [57164.212    stream] -- Using IO method: MMAP
-- DEBUG [57164.212    stream] -- Calling ioctl(VIDIOC_REQBUFS) for V4L2_MEMORY_MMAP ...
-- INFO  [57164.219    stream] -- Requested 5 device buffers, got 5
-- DEBUG [57164.219    stream] -- Allocating device buffers ...
-- DEBUG [57164.219    stream] -- Calling ioctl(VIDIOC_QUERYBUF) for device buffer 0 ...
-- DEBUG [57164.219    stream] -- Mapping device buffer 0 ...
-- DEBUG [57164.220    stream] -- Calling ioctl(VIDIOC_QUERYBUF) for device buffer 1 ...
-- DEBUG [57164.220    stream] -- Mapping device buffer 1 ...
-- DEBUG [57164.221    stream] -- Calling ioctl(VIDIOC_QUERYBUF) for device buffer 2 ...
-- DEBUG [57164.221    stream] -- Mapping device buffer 2 ...
-- DEBUG [57164.221    stream] -- Calling ioctl(VIDIOC_QUERYBUF) for device buffer 3 ...
-- DEBUG [57164.221    stream] -- Mapping device buffer 3 ...
-- DEBUG [57164.222    stream] -- Calling ioctl(VIDIOC_QUERYBUF) for device buffer 4 ...
-- DEBUG [57164.222    stream] -- Mapping device buffer 4 ...
-- DEBUG [57164.223    stream] -- Calling ioctl(VIDIOC_QBUF) for buffer 0 ...
-- DEBUG [57164.223    stream] -- Calling ioctl(VIDIOC_QBUF) for buffer 1 ...
-- DEBUG [57164.223    stream] -- Calling ioctl(VIDIOC_QBUF) for buffer 2 ...
-- DEBUG [57164.223    stream] -- Calling ioctl(VIDIOC_QBUF) for buffer 3 ...
-- DEBUG [57164.223    stream] -- Calling ioctl(VIDIOC_QBUF) for buffer 4 ...
-- DEBUG [57164.223    stream] -- Device fd=8 initialized
-- DEBUG [57164.223    stream] -- Calling ioctl(VIDIOC_STREAMON) ...
-- ERROR [57164.226    stream] -- Unable to start capturing: Input/output error
-- DEBUG [57164.226    stream] -- Releasing device buffers ...
-- DEBUG [57164.227    stream] -- Closing device ...
-- INFO  [57164.229    stream] -- Device fd=8 closed
-- INFO  [2139.653    stream] -- Sleeping 1 seconds before new stream init ...
```

mdevaev, 2022-11-09:
Sorry for the long answer. I think this is a driver issue. Probably also not enough USB bandwidth if you have a lot of devices connected.

mdevaev, 2023-11-16:
Try the latest release. If the reason is not a bandwidth, it could be fixed

closed 2024-02-03 by mdevaev