pikvm/ustreamer #165: error while running make

mh2024, 2022-08-20:
Hi all
first of all, sorry for the nooby problem, I'm not so into making c/c++ files so the error is attached below.
any help would be appreciated.
OS: ubuntu 18.04

```
~/projects/ustreamer$ make
make apps
make[1]: Entering directory '/home/sitl/projects/ustreamer'
make -C src
make[2]: Entering directory '/home/sitl/projects/ustreamer/src'
-- CC libs/base64.c
cc: error: unrecognized command line option ‘-std=c17’; did you mean ‘-std=c11’?
Makefile:98: recipe for target 'build/libs/base64.o' failed
make[2]: *** [build/libs/base64.o] Error 1
make[2]: Leaving directory '/home/sitl/projects/ustreamer/src'
Makefile:38: recipe for target 'apps' failed
make[1]: *** [apps] Error 2
make[1]: Leaving directory '/home/sitl/projects/ustreamer'
Makefile:28: recipe for target 'all' failed
make: *** [all] Error 2
```

mdevaev, 2022-08-20:
Hello. Ubuntu 18.04 has very old GCC that not support C17 standard. You need to find new GCC for building ustreamer or perform dist upgrade.

mh2024, 2022-08-20:
thanks for the support

closed 2022-08-20 by mh2024