pikvm/ustreamer #166: Blurry video with ustreamer and Bullseye Lite 32 bits on Raspberry PI Zero 2 W

Denisthemalice, 2022-08-20:
Summary: ustreamer is running but the quality of the image is rather poor : hazy and blurry. 

Is it "normal" and if not, is it a problem from UV4L or from ustreamer ?

The camera is the standard RPi camera connected directly to the board using the CSI connector.

After the installation of the 32 bits Bullseye Lite version and an update of the OS, the kernel version is:

> uname -a
> Linux raspberrypi 5.15.56-v7+ #1575 SMP Fri Jul 22 20:28:11 BST 2022 armv7l GNU/Linux

The version of ustreamer (after the end of the installation) is :
```
./ustreamer --version
5.20
```

At the end of the installation the following features are indicated:
```
./ustreamer --features
- WITH_GPIO
- WITH_SYSTEMD
+ WITH_PTHREAD_NP
+ WITH_SETPROCTITLE
+ HAS_PDEATHSIG
```

The following command allows to view the video:

`./ustreamer --host 192.168.1.18 --port 8000 --resolution 1920x1080 --desired-fps 30 --workers 3 --buffers 4 
--flip-vertical 1 --encoder M2M-IMAGE `

The result displayed using Putty and SSH is the following:

```
./ustreamer --host 192.168.1.18 --port 8000 --resolution 1920x1080 --desired-fps 30 --workers 3 --buffers 4 
--flip-vertical 1 --encoder M2M-IMAGE

-- INFO  [131.720      main] -- Using internal blank placeholder
-- INFO  [131.721      main] -- Listening HTTP on [192.168.1.18]:8000
-- INFO  [131.721    stream] -- Using V4L2 device: /dev/video0
-- INFO  [131.721    stream] -- Using desired FPS: 30
-- INFO  [131.722      http] -- Starting HTTP eventloop ...
================================================================================
-- INFO  [131.726    stream] -- Device fd=8 opened
-- INFO  [131.727    stream] -- Using input channel: 0
-- INFO  [131.732    stream] -- Using resolution: 1920x1080
-- INFO  [131.732    stream] -- Using format: YUYV
-- INFO  [131.732    stream] -- Using HW FPS: 30
-- INFO  [131.732    stream] -- Using IO method: MMAP
-- INFO  [131.760    stream] -- Requested 4 device buffers, got 4
-- INFO  [131.767    stream] -- Applying control flip_vertical: 1
-- INFO  [132.319    stream] -- Capturing started
-- INFO  [132.320    stream] -- JPEG-0: Initializing encoder ...
-- INFO  [132.320    stream] -- JPEG-1: Initializing encoder ...
-- INFO  [132.320    stream] -- JPEG-2: Initializing encoder ...
-- INFO  [132.320    stream] -- Using JPEG quality: 80%
-- INFO  [132.320    stream] -- Creating pool JPEG with 3 workers ...
-- INFO  [132.321    stream] -- Capturing ...
-- INFO  [132.669      jw-0] -- JPEG-0: Configuring encoder: DMA=0 ...
-- INFO  [132.851      jw-1] -- JPEG-1: Configuring encoder: DMA=0 ...
-- INFO  [148.686      http] -- HTTP: Registered client: 
[192.168.1.11]:42160, id=318f7618c9d859bd; clients now: 1
```

Note: lower resolutions have been attempted but do not solve the problem.

I also tried several combinations of --encoder and --format  without success.

To view the stream on a Firefox browser, I used:

        http://192.168.1.18: 8000/stream`

There is an odd information for installing UV4L on Bullseye. Take a look at:

https://www.linux-projects.org/uv4l/installation/

There is the following warning in red characters:

```
For Bullseye users: before installing and using (many of) the UV4L drivers with hardware acceleration enabled, you must: 

1.	Open a terminal and run the following commands: 
o	sudo rpi-update
[A reboot is needed here to activate the new firmware].
o	echo /opt/vc/lib/ | sudo tee /etc/ld.so.conf.d/vc.conf
o	sudo ldconfig

2.	Enable the Legacy Camera support from the Interface Options menu of the following system command and reboot: 
o	sudo raspi-config
o	sudo reboot
```

     a) It is rather odd to be instructed to run rpi-update since it is usually deprecated.

     b) It is rather odd to be instructed to run raspi-config to enable the legacy camera, 
        since it is not required for Bullseye for other camera software.

The quality of the video is so poor that this combination of software make it unusable in practice.

Any thoughts to improve the quality of the pictures when running ustreamer + UV4L + Bullseye ?

Denisthemalice, 2022-08-21:
**SOLVED**: Removing the piece of plastic that protects the lens solved the issue !

The following command provides a latency around 300 mS :
./ustreamer --host 192.168.1.18 --port 8000 --resolution 1280x720 --desired-fps 25 --workers 3 --buffers 4 --flip-vertical 1 --encoder M2M-IMAGE

closed 2022-08-21 by Denisthemalice

mdevaev, 2022-08-21:
Perfect :)