pikvm/ustreamer #167: How to add a text line above the video or to add a text line overlay within the video ?

Denisthemalice, 2022-08-22:
When the Pi camera was accessible using Python in earlier versions than Bullseye, 
it was possible to add some text above the video, e.g. using the following program:

https://picamera.readthedocs.io/en/latest/recipes2.html#web-streaming

Is there a similar way to do it using ustreamer under Bullseye ?

Alternatively, is there a way to do it adding a text line overlay within the video ?

For example, using the Video Overlay options from v4l2-ctl ?

Note: There is an preliminary project there: https://www.improwis.com/projects/sw_polycam/ , but is it unclear on how to use it.

mdevaev, 2022-08-22:
I haven't implemented this (and I don't plan to at the moment), so I recommend doing it using the browser with HTML/CSS.

Denisthemalice, 2022-08-23:
Thank you for your recommendation. 

Hereafter in a solution for a Raspberry PI Zero 2 W:

```
sudo apt update
sudo apt install apache2 -y
sudo apt install php libapache2-mod-php -y
sudo rm /var/www/html/index.html
sudo nano /var/www/html/index.php

```
and place the following text into the file index.php :

```
<!DOCTYPE html>
<html lang="fr">
<head>
<meta charset="utf-8">
<title>Loco</title>
<link href="A-style.css" rel="stylesheet" media="all" type="text/css">
</head>
<body>
<H1>
<center>
<p style="font-family: Arial">
Line 1<br>Line 2
</p>
</center>
</H1>
<center>
<iframe
    style="width:1280px; height:720px;"
    src="http://192.168.1.18:8000/stream"
</center>
</iframe>
</body>
</html>
```

Launch ustreamer:

```
cd ustreamer
./ustreamer --host 192.168.1.18 --port 8000 --resolution 1280x720 --desired-fps 25 --workers 4 --buffers 5 --flip-vertical 1 --encoder M2M-IMAGE
```

See the stream in a browser using http://192.168.1.18

mdevaev, 2022-08-23:
Nice :ok_hand: 

You can serve `index.html` using ustreamer itself. See `--static` option.

closed 2022-08-23 by mdevaev