pikvm/ustreamer #170: Supper monochrome cameras

gudvinr, 2022-09-09:
I have this monochrome camera: http://withrobot.com/en/camera/ocam-1mgn-u

It is quite unusual in a sense that it doesn't provide color image formats supported by ustreamer (YUYV, UYVY, RGB565, RGB24, JPEG).  
Its only output pixelformat is GREY which is [8-bit monochrome image](https://www.cs.auckland.ac.nz/courses/compsci773s1c/lectures/YuY2_files/intro.htm#GREY)

This fact leads to ustreamer not being able to consume stream from the camera:
```
-- ERROR [2233.317    stream] -- Could not obtain the requested pixelformat=YUYV; driver gave us unsupported
-- ERROR [2233.317    stream] -- Unsupported pixelformat=GREY (fourcc)
```

`$ uvcdynctrl -f` output:
```
Listing available frame formats for device video0:
Pixel format: GREY (8-bit Greyscale)
  Frame size: 1280x960
    Frame rates: 30, 25, 15
  Frame size: 1280x720
    Frame rates: 30, 25, 15
  Frame size: 640x480
    Frame rates: 100, 90, 60, 50, 30, 25, 15
  Frame size: 320x240
    Frame rates: 180, 150, 120, 100, 60, 50
```

mdevaev, 2022-09-11:
Interesting. Btw can this camera generate mjpeg independently? Show me `v4l2 --list-formats`

gudvinr, 2022-09-11:
No, it can't. uvcdynctrl basically does same thing, output just slightly different

mdevaev, 2022-09-11:
Got it. I like the idea of adding this, but I have to think about how to do it right.

drowningblue, 2025-03-05:
I would like to have this feature as well please. I am repurposing windows hello cameras and would like to use them as night vision cameras.

drowningblue, 2025-03-06:
@gudvinr yes it does. 

Here is the v4l2 output:
```
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: GREY' (8-bit Greyscale)
                Size: Discrete 340x340
                        Interval: Discrete 0.033s (30.000 fps)
```

PR #171 (2025-03-08): Add GREY pixelformat
Fixes #170

Monochrome cameras send only Y component of YUV image
commit b380beba (2025-03-08): Add GREY pixelformat (#171)

Fixes #170

Monochrome cameras send only Y component of YUV image
closed 2025-03-08 by mdevaev