pikvm/ustreamer #172: change resolution while streaming

OliverGruenberg, 2022-09-20:
Hi there!
I just ran into the challenge, that I do want to e.g. stream at 640x480 and capture a snapshot with 4656x3496 px.

--help tells me to use --dv-timings in order to change resolution on the fly (although I don't know how to do this), but I don't get this far, because...

The command
ustreamer --device=/dev/video0 --host=0.0.0.0 --port=8080 --resolution 4656x3496 --format JPEG --quality 100 --device-timeout 2 --slowdown --drop-same-frames=30 --dv-timings --debug

results in:
-- DEBUG [17923.042      main] -- Increasing frame buffer 'blank_internal': 0 -> 524288 (+524288)
-- INFO  [17923.042      main] -- Using internal blank placeholder
-- DEBUG [17923.042      main] -- Installing SIGINT handler ...
-- DEBUG [17923.043      main] -- Installing SIGTERM handler ...
-- DEBUG [17923.043      main] -- Ignoring SIGPIPE ...
-- DEBUG [17923.043      main] -- Binding HTTP to [0.0.0.0]:8080 ...
-- INFO  [17923.044      main] -- Listening HTTP on [0.0.0.0]:8080
-- INFO  [17923.044    stream] -- Using V4L2 device: /dev/video0
-- INFO  [17923.044    stream] -- Using desired FPS: 0
-- DEBUG [17923.045    stream] -- _stream_init_loop: stream->run->stop=0
-- INFO  [17923.045      http] -- Starting HTTP eventloop ...
================================================================================
-- DEBUG [17923.061      http] -- HTTP: Updating exposed frame (online=0) ...
-- VERB  [17923.061      http] -- HTTP: Exposed frame: online=0, exp_time=0.000000
-- INFO  [17923.115    stream] -- Device fd=8 opened
-- DEBUG [17923.115    stream] -- Calling ioctl(VIDIOC_QUERYCAP) ...
-- INFO  [17923.115    stream] -- Using input channel: 0
-- DEBUG [17923.115    stream] -- Using TV standard: DEFAULT
-- DEBUG [17923.115    stream] -- Using DV timings
-- DEBUG [17923.115    stream] -- Calling ioctl(VIDIOC_QUERY_DV_TIMINGS) ...
-- DEBUG [17923.115    stream] -- Calling ioctl(VIDIOC_QUERYSTD) ...
-- DEBUG [17923.115    stream] -- Calling ioctl(VIDIOC_SUBSCRIBE_EVENT) ...
-- ERROR [17923.115    stream] -- Can't subscribe to V4L2_EVENT_SOURCE_CHANGE: Invalid argument
-- DEBUG [17923.115    stream] -- Closing device ...
-- INFO  [17923.116    stream] -- Device fd=8 closed
-- INFO  [17923.116    stream] -- Sleeping 1 seconds before new stream init ...
-- DEBUG [17924.070      http] -- HTTP: Repeating exposed ...
================================================================================
-- INFO  [17924.118    stream] -- Device fd=8 opened
-- DEBUG [17924.118    stream] -- Calling ioctl(VIDIOC_QUERYCAP) ...
-- INFO  [17924.118    stream] -- Using input channel: 0
-- DEBUG [17924.118    stream] -- Using TV standard: DEFAULT
-- DEBUG [17924.118    stream] -- Using DV timings
-- DEBUG [17924.118    stream] -- Calling ioctl(VIDIOC_QUERY_DV_TIMINGS) ...
-- DEBUG [17924.118    stream] -- Calling ioctl(VIDIOC_QUERYSTD) ...
-- DEBUG [17924.119    stream] -- Calling ioctl(VIDIOC_SUBSCRIBE_EVENT) ...
-- ERROR [17924.119    stream] -- Can't subscribe to V4L2_EVENT_SOURCE_CHANGE: Invalid argument
-- DEBUG [17924.119    stream] -- Closing device ...
-- DEBUG [17924.119      http] -- HTTP: Updating exposed frame (online=0) ...
-- VERB  [17924.119      http] -- HTTP: Exposed frame: online=0, exp_time=0.000000
-- INFO  [17924.119    stream] -- Device fd=8 closed
-- INFO  [17924.119    stream] -- Sleeping 1 seconds before new stream init ...
================================================================================

Can you please assist?

mdevaev, 2022-09-20:
Option `--dv-timings` can be used with only CSI HDMI capture device or similar. Also it is impossible to use different resolutions for streaming and screenshots, sorry.

closed 2022-09-20 by OliverGruenberg

CursingThomas, 2023-12-06:
I want to Re-open this question, since i would also like to change resolution during streaming, but without a CSI HDMI; this should be possible no?
Maybe by adjusting the state and forcing a reload?