pikvm/ustreamer #173: Streamed resolution doesn't update when server's resolution is reduced

coderthetyler, 2022-09-21:
Been having some weird issues with the streamed resolution. 

**tl;dr: the active resolution can be increased, but once it's increased, it doesn't go back down again.**

I originally had the resolution on the server OS set to 1280x720 (16:9). No issues. I then bumped it up to 1440x900 (16:10). Again, no issues. I then dropped it back down to 1280x1080, and I got black bars on the top & bottom of the streamed video (presumably due to the now-mismatched aspect ratio). I ran a `v4lt2-ctl --all` and under DV Timings I saw the Active Width/Height reported as 1440x900. Uh oh.

I then tried updating the resolution on the server to 1920x1080. No issues. DV Timings now reports an active resolution of 1920x1080. Tried dropping the server resolution back to 1440x900 and, you guessed it, DV Timings still reports 1920x1080. I also saw black bars on the sides of the stream (again, presumably due to the mismatched aspect ratio of what's streamed vs what the server is outputting). Dropping the server resolution down to 1280x720 doesn't yield the black bars anymore but only because it's streaming at the same aspect ratio, but I do see scaling artifacts as if it's upscaling from 1280x720 to 1920x1080.

Things I tried without any luck:

- Rebooting my PiKVM
- Replugging the HDMI to the server
- Rebooting the server
- Various combinations & orderings of these things
- Played with the args to `ustreamer` (my most valiant attempt: removing `--dv-timings` and setting `--resolution 1366x768`, which just caused my stream to go blue, heh)
- Tried configuring the video device with `v4l2-ctl`

Here's what I'm working with: (I bought the pre-assembled PiKVM kit as I wasn't feeling too adventurous 😅.)
[image]
[image]

thanks :)

mdevaev, 2022-09-21:
Which OS do you have on the server?

coderthetyler, 2022-09-21:
I should use the term "server" lightly. It's running Windows 10 Enterprise.

mdevaev, 2022-09-21:
I suppose this is a windows issue. Windows does not change the resolution honestly, but uses downscale in some cases. You need to disable this in the OS.