pikvm/ustreamer #174: Fix Raspi Cam video port

appleimperio, 2022-09-25:
Hi I have 2 cameras connected to my pi one is USB that i use the UDEV rules to name it /dev/CAM1 but the other camera is a Raspberry pi camera module and i can't find a way to fix the video to /dev/CAM2 every time i restart the pi the systemd services i create for each camera get mix up. Thanks

mdevaev, 2022-10-24:
You can try to bind device name in udev using serial number of the device or vid+pid.

appleimperio, 2022-10-24:
Do you know how can i get the vid+pid of the Raspi cam? Thank You

mdevaev, 2022-10-24:
Ah, sorry, I missed it. You can bind to `KERNELS`, use udev-adm to get the exact value. See the example: https://github.com/pikvm/kvmd/blob/master/configs/os/udev/v0-hdmi-rpi3.rules

appleimperio, 2022-11-03:
this is all im getting from the PI cam 
```
pi@klipperfarm:~ $ udevadm info --name=/dev/video0 --attribute-walk

Udevadm info starts with the device specified by the devpath and then
walks up the chain of parent devices. It prints for every device
found, all possible attributes in the udev rules key format.
A rule to match, can be composed by the attributes of the device
and the attributes from one single parent device.

  looking at device '/devices/virtual/video4linux/video0':
    KERNEL=="video0"
    SUBSYSTEM=="video4linux"
    DRIVER==""
    ATTR{dev_debug}=="0"
    ATTR{index}=="0"
    ATTR{name}=="camera0"
    ATTR{power/control}=="auto"
    ATTR{power/runtime_active_time}=="0"
    ATTR{power/runtime_status}=="unsupported"
    ATTR{power/runtime_suspended_time}=="0"
```
For the USB I get a lot more info and I'm sure that's the correct video0 for the camera because I manually assigned it and I can view it but on the next restart it will be lost. Thanks

mdevaev, 2022-11-03:
Try this: `udevadm info -a -p $(udevadm info -q path -n /dev/video0)`

appleimperio, 2022-11-04:
Same output. I add the list of devices maybe im pointing it  wrong. Thank you
```
pi@klipperfarm:~ $ v4l2-ctl --list-devices
bcm2835-codec-decode (platform:bcm2835-codec):
	/dev/video10
	/dev/video11
	/dev/video12
	/dev/video18
	/dev/video31
	/dev/media2

bcm2835-isp (platform:bcm2835-isp):
	/dev/video13
	/dev/video14
	/dev/video15
	/dev/video16
	/dev/video20
	/dev/video21
	/dev/video22
	/dev/video23
	/dev/media1
	/dev/media3

mmal service 16.1 (platform:bcm2835-v4l2-0):
	/dev/video0

rpivid (platform:rpivid):
	/dev/video19
	/dev/media0

USB Camera: USB Camera (usb-0000:01:00.0-1.3):
	/dev/video1
	/dev/video2
	/dev/media4

pi@klipperfarm:~ $ udevadm info -a -p $(udevadm info -q path -n /dev/video0)

Udevadm info starts with the device specified by the devpath and then
walks up the chain of parent devices. It prints for every device
found, all possible attributes in the udev rules key format.
A rule to match, can be composed by the attributes of the device
and the attributes from one single parent device.

  looking at device '/devices/virtual/video4linux/video0':
    KERNEL=="video0"
    SUBSYSTEM=="video4linux"
    DRIVER==""
    ATTR{dev_debug}=="0"
    ATTR{index}=="0"
    ATTR{name}=="camera0"
    ATTR{power/control}=="auto"
    ATTR{power/runtime_active_time}=="0"
    ATTR{power/runtime_status}=="unsupported"
    ATTR{power/runtime_suspended_time}=="0"
```

mdevaev, 2024-02-03:
Hi. I'm sorry, I missed your message. Is the question still relevant?

closed 2024-02-04 by appleimperio