pikvm/ustreamer #175: crash in libjpeg62-turbo

tomaszduda23, 2022-10-02:
There is crash inside libjpeg62. Any idea how I can fix it?

```
docker buildx build --file pkg/docker/Dockerfile.x64.native --tag ustreamer --platform linux/arm64 .
```
```
(gdb) run -t -w 1
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /ustreamer/ustreamer -t -w 1
warning: Error disabling address space randomization: Operation not permitted
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
-- INFO  [8147.333      main] -- Using internal blank placeholder
-- INFO  [8147.334      main] -- Listening HTTP on [127.0.0.1]:8080
[New Thread 0xffff929961e0 (LWP 1262)]
-- INFO  [8147.354    stream] -- Using V4L2 device: /dev/video0
-- INFO  [8147.354    stream] -- Using desired FPS: 0
================================================================================
[New Thread 0xffff921951e0 (LWP 1263)]
-- INFO  [8147.355    stream] -- Device fd=8 opened
-- INFO  [8147.355    stream] -- Using input channel: 0
-- INFO  [8147.355      http] -- Starting HTTP eventloop ...
-- INFO  [8147.362    stream] -- Got new DV-timings: 1280x720p60.00, pixclk=74250000, vsync=30, hsync=370
-- INFO  [8147.384    stream] -- Using resolution: 1280x720
-- ERROR [8147.385    stream] -- Could not obtain the requested format=YUYV; driver gave us RGB24
-- INFO  [8147.385    stream] -- Falling back to format=RGB24
-- INFO  [8147.385    stream] -- Using format: RGB24
-- INFO  [8147.385    stream] -- Querying HW FPS changing is not supported
-- INFO  [8147.385    stream] -- Using IO method: MMAP
-- INFO  [8147.393    stream] -- Requested 5 device buffers, got 5
-- INFO  [8147.398    stream] -- Capturing started
-- INFO  [8147.398    stream] -- Using JPEG quality: 80%
-- INFO  [8147.398    stream] -- Creating pool JPEG with 1 workers ...
[New Thread 0xffff90be41e0 (LWP 1264)]
-- INFO  [8147.400    stream] -- Capturing ...

Thread 4 "jw-0" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xffff90be41e0 (LWP 1264)]
0x0000ffff92db279c in jsimd_extrgb_ycc_convert_neon () at jsimd_arm64_neon.S:2133
2133	jsimd_arm64_neon.S: No such file or directory.
(gdb) bt
#0  0x0000ffff92db279c in jsimd_extrgb_ycc_convert_neon () at jsimd_arm64_neon.S:2133
#1  0x0000ffff92d944f4 in pre_process_data (cinfo=0xffff90be3478, input_buf=0xffff8c000b20, in_row_ctr=0xffff90be3304, in_rows_avail=1, output_buf=0xaaaaaf99d648 <us_g_log_mutex>, 
    out_row_group_ctr=0xffff84002ab4, out_row_groups_avail=0) at jcprepct.c:148
#2  0x0000ffff8c000fd0 in ?? ()
Backtrace stopped: Cannot access memory at address 0x412e848000000018
(gdb) quit
```

tomaszduda23, 2022-10-02:
Similar crash after using ubuntu as base image

```
(gdb) run -t -w 1
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /ustreamer/ustreamer -t -w 1
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
-- INFO  [11369.542      main] -- Using internal blank placeholder
-- INFO  [11369.542      main] -- Listening HTTP on [127.0.0.1]:8080
[New Thread 0xfffff7a4c120 (LWP 2809)]
-- INFO  [11369.544    stream] -- Using V4L2 device: /dev/video0
-- INFO  [11369.544    stream] -- Using desired FPS: 0
===============================[New Thread 0xfffff723c120 (LWP 2810)]
=================================================
-- INFO  [11369.545      http] -- Starting HTTP eventloop ...
-- INFO  [11369.545    stream] -- Device fd=8 opened
-- INFO  [11369.545    stream] -- Using input channel: 0
-- INFO  [11369.552    stream] -- Got new DV-timings: 1280x720p60.00, pixclk=74250000, vsync=30, hsync=370
-- INFO  [11369.575    stream] -- Using resolution: 1280x720
-- ERROR [11369.575    stream] -- Could not obtain the requested format=YUYV; driver gave us RGB24
-- INFO  [11369.576    stream] -- Falling back to format=RGB24
-- INFO  [11369.576    stream] -- Using format: RGB24
-- INFO  [11369.576    stream] -- Querying HW FPS changing is not supported
-- INFO  [11369.576    stream] -- Using IO method: MMAP
-- INFO  [11369.588    stream] -- Requested 5 device buffers, got 5
-- INFO  [11369.594    stream] -- Capturing started
-- INFO  [11369.594    stream] -- Using JPEG quality: 80%
-- INFO  [11369.594    stream] -- Creating pool JPEG with 1 workers ...
[New Thread 0xfffff5c7c120 (LWP 2811)]
-- INFO  [11369.595    stream] -- Capturing ...

Thread 4 "jw-0" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xfffff5c7c120 (LWP 2811)]
rgb_ycc_convert (cinfo=<optimized out>, input_buf=0xfffff5c7b1f8, output_buf=0xffffe80023d8, output_row=1, num_rows=<optimized out>) at jccolor.c:162
162	jccolor.c: No such file or directory.
(gdb) bt
#0  rgb_ycc_convert (cinfo=<optimized out>, input_buf=0xfffff5c7b1f8, output_buf=0xffffe80023d8, output_row=1, num_rows=<optimized out>) at jccolor.c:162
#1  0x0000fffff7ede868 in pre_process_data (cinfo=0xfffff5c7b2a0, input_buf=0xfffff5c7b1f0, in_row_ctr=0xfffff5c7b144, in_rows_avail=1, output_buf=0xffffe8002a38, out_row_group_ctr=0xffffe8002a2c, 
    out_row_groups_avail=8) at jcprepct.c:145
#2  0x0000fffff7edb880 in process_data_simple_main (cinfo=0xfffff5c7b2a0, input_buf=0xfffff5c7b1f0, in_row_ctr=0xfffff5c7b144, in_rows_avail=1) at jcmainct.c:123
#3  0x0000fffff7ed4edc in jpeg_write_scanlines (cinfo=0xfffff5c7b2a0, scanlines=0xfffff5c7b1f0, num_lines=<optimized out>) at jcapistd.c:109
#4  0x0000aaaaaaac7a7c in us_cpu_encoder_compress ()
#5  0x0000aaaaaaac5568 in _worker_run_job ()
#6  0x0000aaaaaaae7048 in _worker_thread ()
#7  0x0000fffff7cbd5c8 in start_thread (arg=0x0) at ./nptl/pthread_create.c:442
#8  0x0000fffff7d25d1c in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:79
(gdb) disassemble 
```

mdevaev, 2022-10-02:
Please try with libjpeg8

tomaszduda23, 2022-10-02:
I already checked libjpeg8 and libjpeg9 both crashes.

mdevaev, 2022-10-02:
Sorry, I haven't used it for a long time so I'm not sure why it doesn't work.

tomaszduda23, 2022-10-02:
Changing format solve the issue. `./ustreamer -t  -w 1 --format UYVY`.
It is somehow related to RGB24,

mdevaev, 2023-11-15:
@tomaszduda23 hello. Do you know is it fixed?

mdevaev, 2023-11-15:
Could you check it with new libs?

tomaszduda23, 2023-11-15:
It seems to be working "pikvm/ustreamer:v5.45".

closed 2023-11-15 by tomaszduda23

mdevaev, 2023-11-16:
Perfect. thank you.