pikvm/ustreamer #183: Can't get Ustreamer and Symlink to work together

BlueWings172, 2022-11-03:
Hello

I trying to use 2 webcams but they keep swapping video0 and video1 so I decided to use udev rules to create symlinks in addition the Ustreamer and I had working it for a while until i rebooted the Pi !!

Not sure what I did wrong and this is driving me crazy and Ustreamer is showing **No Signal** on the browser. If I don't use the symlinks, Ustreamer works.

Here is my rule with priority 99:

`SUBSYSTEM=="video4linux", ENV{ID_BUS}=="usb", ATTRS{idVendor}=="090c", ATTRS{idProduct}=="f37d", SYMLINK+="endoscope"`

The dev/endoscope does show when I run ls /dev and when I run udevadm info /dev/endoscope  I get the correct info of the webcam

Here is my docker-compose YAML file:

```
  ustreamer:
    image: mkuf/ustreamer:latest
    restart: unless-stopped
    ports:
      - 8091:8080
    devices:
      - "/dev/endoscope"
    command: --host=0.0.0.0 --port=8080 --slowdown --device=/dev/endoscope --resolution=1080x720 --format=MJPEG --desired-fps=20
```

when I run `docker-compose logs ustreamer` , I get repeated rows of the below:

```
iotstack-ustreamer-1  | -- INFO  [579.531    stream] -- Device fd=8 opened
iotstack-ustreamer-1  | -- INFO  [579.531    stream] -- Using input channel: 0
iotstack-ustreamer-1  | -- ERROR [579.531    stream] -- Can't set input channel
iotstack-ustreamer-1  | -- INFO  [579.531    stream] -- Device fd=8 closed
iotstack-ustreamer-1  | -- INFO  [579.531    stream] -- Sleeping 1 seconds before new stream init ...
```

That said, if I modify my docker-compose YAML to include `/dev/video0` instead of` /dev/endoscope` then it works.

So why Ustreamer doesn't want to play well with my symlink ?

I would appreciate any help.

thanks!

mdevaev, 2022-11-05:
I use symlinks with ustreamer without any problems. Are you sure it points to the right device?

BlueWings172, 2022-11-06:
@mdevaev 

I'm certain the rule points to the correct device. 

After reboot when both ustreamer containers are showing No Signal, I have to run sudo udevadm control --reload-rules && sudo udevadm trigger  to reload and trigger the rules and then restart the container with docker-compose restart ustreamer_xxxxx and then immediately that works. So if the symlinks are not pointing to the correct device, this won't work.

I don't understand this behavior. I tried to add the command sudo udevadm control --reload-rules && sudo udevadm to rc.local but that did not help. I thought maybe the containers are started before the rules are applied so I've added RUN action in my rule to restart the containers but that also did not work.

It is like the when the containers are started, they don't see the symlinks but when I reload the rules and restart the containers the same config works.

Why the containers don't see the symlinks after reboot ? I'm using another rule for my 3d printer and Octoprint can see the device without issues. I've spent countless hours trying to find a solution but i'm getting no where.

mdevaev, 2022-11-06:
Alas, I do not know what is the matter. uStreamer doesn't do any magic with device files, if it can't open it, then something is wrong with the device.

mdevaev, 2023-02-26:
Sup?

BlueWings172, 2023-02-27:
@mdevaev 
all good. I did find a solution if that's what you're asking about.

I just had to add ATTR{index}=="0" in the udev rule.

Been working great since.

closed 2023-02-27 by BlueWings172

mdevaev, 2023-02-27:
Glad to hear it!