pikvm/ustreamer #192: Feature request: Add option for timestamp

schneemaier, 2022-12-02:
Please consider adding option to include a timestamp in the video (ex year-month-day hour-min-sec).

Example use case on raspberry pi:
1. read the video in yuyv format from the webcam
2. add timestamp
3. use the hardware encoder to create an mjpeg stream
Thanks

mdevaev, 2022-12-03:
JPEG transcoding is a huge load for Raspberry and it will negatively affect framerate. Adding a timestamp is very difficult and I don't think this is what a ustreamer should do. All it does is encode the image as quickly as possible and give it to you, any video change is the task of the next component in the pipeline.

schneemaier, 2022-12-03:
@mdevaev maybe you are right.

In my use case i have a webcam connected to the RPi and the stream can be watched over the internet while a timelaps video is recorder. I created a setup to add the timestamp, which works as the following:

Webcam yuyv422 -> ffmpeg drawtext -> yuyv422 -> ustream Hardware encoding -> mjpeg stream

The problem is that the YUV stream on the webcam is limiting the framrate to ~10 frames per sec due to USB bandwidth.

This is the code I use
```
sudo modprobe v4l2loopback 2
/usr/bin/screen -d -m /usr/bin/ffmpeg -f v4l2 -framerate 8 -video_size 960x720 -i /dev/video0 -video_size 960x720 -vf drawtext=text='%{localtime}':fontsize=14:fontcolor=white:font=Arial:x=w-tw-10:y=th+10:box=1:boxcolor=black:boxborderw=15,format=yuyv422 -c:a copy -f v4l2 /dev/video2
/usr/bin/screen -d -m /usr/local/bin/ustreamer --device=/dev/video2 -f 8 --encoder=M2M-VIDEO -l --workers=1 --drop-same-frames=10 --host=0.0.0.0 --port=8080
```

This runs on a RPi B V1.2 with around 75% CPU load where ffmpeg takes the most resources. ffmpeg have hardware acceleration on RPis just does not support mjpeg encoding or decoding.

mdevaev, 2022-12-29:
If ffmpeg can read mjpeg and decode it then that would solve your problem. But I think too many resources will be spent on this.

closed 2023-02-25 by mdevaev