pikvm/ustreamer #194: Encoding issue with specific text on an HDMI-USB video capture device

billdenney, 2022-12-23:
I'm using ustreamer with this HDMI-USB video capture device: https://smile.amazon.com/gp/product/B08CXWPYQ8

Whenever I use this command to make the text below green on a black background, I get a garbled image:

```
echo -e "\e[1;32m fwupd-94d6edee-8f19-40a9-be14-d881e2e3d55c.cap \e[0m"
```

[image]

It is occurring with either the H.264 or MJPEG encoder.

I'm not sure if this is related to #169, but it seems similar.  I'm using ustreamer version 5.34 on a Raspberry Pi with the 64-bit Raspberry Pi OS based on Debian Bullseye.

Related to tiny-pilot/tinypilot#1237

If there is additional information I can provide to support debugging, please let me know.  (I tried to grab the raw output from the image capture device using `v4l2-ctl`, but I couldn't figure out the correct command to use.)

billdenney, 2022-12-23:
The command I used and logs from the output are:

```
$ /opt/ustreamer/ustreamer \
  --host :: \
  --port 8001 \
  --encoder hw \
  --format jpeg \
  --resolution 1920x1080 \
  --persistent
-- INFO  [19280.218      main] -- Starting PiKVM uStreamer 5.34 ...
-- INFO  [19280.218      main] -- Using internal blank placeholder
-- INFO  [19280.219      main] -- Listening HTTP on [::]:8001
-- INFO  [19280.219    stream] -- Using V4L2 device: /dev/video0
-- INFO  [19280.219    stream] -- Using desired FPS: 0
================================================================================
-- INFO  [19280.225      http] -- Starting HTTP eventloop ...
-- INFO  [19280.291    stream] -- Device fd=8 opened
-- INFO  [19280.291    stream] -- Using input channel: 0
-- INFO  [19280.292    stream] -- Using resolution: 1920x1080
-- ERROR [19280.292    stream] -- Could not obtain the requested format=JPEG; dr                                                                                 iver gave us MJPEG
-- INFO  [19280.292    stream] -- Falling back to format=MJPEG
-- INFO  [19280.293    stream] -- Using format: MJPEG
-- INFO  [19280.294    stream] -- Using HW FPS: 0 -> 60 (coerced)
-- ERROR [19280.294    stream] -- Device doesn't support setting of HW encoding                                                                                  quality parameters
-- INFO  [19280.294    stream] -- Using IO method: MMAP
-- INFO  [19280.319    stream] -- Requested 5 device buffers, got 5
-- INFO  [19280.328    stream] -- Capturing started
-- INFO  [19280.328    stream] -- Using JPEG quality: encoder default
-- INFO  [19280.328    stream] -- Creating pool JPEG with 1 workers ...
-- INFO  [19280.328    stream] -- Capturing ...
-- INFO  [19306.798      http] -- HTTP: NEW client (now=1): [192.168.0.156]:39510, id=e4b31ca92b6d1c77
```

mdevaev, 2022-12-23:
Hello. The encoding process for USB-HDMI is performed inside the device itself and the streamer cannot affect it in any way. It's a black box. Unfortunately, this is a hardware problem and cannot be fixed.

billdenney, 2022-12-23:
That's unfortunate.  Thanks for the quick response.

closed 2022-12-23 by billdenney

mdevaev, 2022-12-25:
Let me clarify again: does this happen even if you are watching a JPEG stream, not H.264?

billdenney, 2022-12-27:
Yes, the issue always happens whether MJPEG or H.264.  If I understand what you said and how I interpret the logs correctly, here are my thoughts (please let me know if I'm missing something).

1. There is an encoding issue with my device.  It outputs bad JPEG data in certain circumstances.
2. When ustreamer outputs MJPEG, it just takes the device output and sends it directly, without interpreting it.
3. When ustreamer outputs H.264, it takes the MJPEG and re-encodes it causing the "extraneous bytes before marker" errors that I'm seeing below.

```
$ sudo /opt/ustreamer/ustreamer \
  --host :: \
  --port 8001 \
  --encoder hw \
  --format jpeg \
  --resolution 1920x1080 \
  --persistent \
  --h264-sink ustreamer::h264 \
  --h264-sink-mode 777 \
  --h264-sink-rm
-- INFO  [341884.079      main] -- Starting PiKVM uStreamer 5.34 ...
-- INFO  [341884.080      main] -- Using internal blank placeholder
-- INFO  [341884.080      main] -- Using H264-sink: ustreamer::h264
-- INFO  [341884.081      main] -- Listening HTTP on [::]:8001
-- INFO  [341884.081    stream] -- Using V4L2 device: /dev/video0
-- INFO  [341884.081    stream] -- Using desired FPS: 0
-- INFO  [341884.081      http] -- Starting HTTP eventloop ...
-- INFO  [341884.082    stream] -- H264: Initializing encoder ...
-- INFO  [341884.094    stream] -- H264: Configuring encoder: DMA=0 ...
================================================================================
-- INFO  [341884.192    stream] -- Device fd=10 opened
-- INFO  [341884.192    stream] -- Using input channel: 0
-- INFO  [341884.193    stream] -- Using resolution: 1920x1080
-- ERROR [341884.193    stream] -- Could not obtain the requested format=JPEG; driver gave us MJPEG
-- INFO  [341884.193    stream] -- Falling back to format=MJPEG
-- INFO  [341884.193    stream] -- Using format: MJPEG
-- INFO  [341884.194    stream] -- Using HW FPS: 0 -> 60 (coerced)
-- ERROR [341884.195    stream] -- Device doesn't support setting of HW encoding quality parameters
-- INFO  [341884.195    stream] -- Using IO method: MMAP
-- INFO  [341884.213    stream] -- Requested 5 device buffers, got 5
-- INFO  [341884.218    stream] -- Capturing started
-- INFO  [341884.219    stream] -- Using JPEG quality: encoder default
-- INFO  [341884.219    stream] -- Creating pool JPEG with 1 workers ...
-- INFO  [341884.219    stream] -- Capturing ...
-- ERROR [341884.551    stream] -- Can't decompress JPEG: Not a JPEG file: starts with 0x28 0x00
-- INFO  [341884.654    stream] -- H264: Configuring encoder: DMA=0 ...
Corrupt JPEG data: 4 extraneous bytes before marker 0xd9
Corrupt JPEG data: 4 extraneous bytes before marker 0xd9
Corrupt JPEG data: 4 extraneous bytes before marker 0xd9
Corrupt JPEG data: 4 extraneous bytes before marker 0xd9
Corrupt JPEG data: premature end of data segment
Corrupt JPEG data: 6 extraneous bytes before marker 0xd9
Corrupt JPEG data: 5 extraneous bytes before marker 0xd9
Corrupt JPEG data: 5 extraneous bytes before marker 0xd9
```

When I run the below command, I get no glitches.  If I understand what I'm doing, this command grabs the raw YUYV stream from the capture device and encodes it with the CPU.  Since I don't see a glitch there, it suggests to me that the issue is when the capture device is doing the encoding.  Is that right?

```
$ sudo /opt/ustreamer/ustreamer \
  --host :: \
  --port 8001 \
  --encoder cpu \
  --format yuyv \
  --resolution 1920x1080 \
  --persistent
-- INFO  [342457.124      main] -- Starting PiKVM uStreamer 5.34 ...
-- INFO  [342457.124      main] -- Using internal blank placeholder
-- INFO  [342457.125      main] -- Listening HTTP on [::]:8001
-- INFO  [342457.125    stream] -- Using V4L2 device: /dev/video0
-- INFO  [342457.125    stream] -- Using desired FPS: 0
================================================================================
-- INFO  [342457.128      http] -- Starting HTTP eventloop ...
-- INFO  [342457.195    stream] -- Device fd=8 opened
-- INFO  [342457.195    stream] -- Using input channel: 0
-- INFO  [342457.197    stream] -- Using resolution: 1920x1080
-- INFO  [342457.197    stream] -- Using format: YUYV
-- INFO  [342457.198    stream] -- Using HW FPS: 0 -> 5 (coerced)
-- INFO  [342457.198    stream] -- Using IO method: MMAP
-- INFO  [342457.217    stream] -- Requested 5 device buffers, got 5
-- INFO  [342457.224    stream] -- Capturing started
-- INFO  [342457.224    stream] -- Using JPEG quality: 80%
-- INFO  [342457.224    stream] -- Creating pool JPEG with 4 workers ...
-- INFO  [342457.225    stream] -- Capturing ...
-- INFO  [342461.331      http] -- HTTP: NEW client (now=1): [192.168.0.156]:53176, id=4d599c76da468786
```

mdevaev, 2022-12-27:
Yes, everything is right.