pikvm/ustreamer #197: Arducam B0205 resolution issues.

gothrek22, 2023-01-14:
Hi, 

great software.

I've found a weird issue with Arducam B0205. It seems that even though the camera supports up to 30 fps 1920x1080 MJPEG streams, that stream dies on ustreamer, but works with mjpeg-streamer.

Here are supported modes:
```ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'MJPG' (Motion-JPEG, compressed)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x360
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 800x600
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 960x720
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 352x288
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 432x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x184
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 176x144
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1280x960
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1280x1024
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1024x768
                        Interval: Discrete 0.033s (30.000 fps)
        [1]: 'YUYV' (YUYV 4:2:2)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 432x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1024x768
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 352x288
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 176x144
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 160x120
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 800x600
                        Interval: Discrete 0.050s (20.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 960x720
                        Interval: Discrete 0.067s (15.000 fps)
                Size: Discrete 1280x960
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 1280x1024
                        Interval: Discrete 0.200s (5.000 fps)
```

I've tried the following: `ustreamer -d /dev/video0 -p 8081 -s 0.0.0.0 -r 1920x1080  -f 30 --debug  --format=mjpeg`.

```
-- INFO  [205120.740      main] -- Starting PiKVM uStreamer 5.34 ...
-- INFO  [205120.740      main] -- Using internal blank placeholder
-- DEBUG [205120.740      main] -- Installing SIGINT handler ...
-- DEBUG [205120.740      main] -- Installing SIGTERM handler ...
-- DEBUG [205120.740      main] -- Ignoring SIGPIPE ...
-- DEBUG [205120.740      main] -- Binding HTTP to [0.0.0.0]:8081 ...
-- INFO  [205120.740      main] -- Listening HTTP on [0.0.0.0]:8081
-- INFO  [205120.740    stream] -- Using V4L2 device: /dev/video0
-- INFO  [205120.740    stream] -- Using desired FPS: 30
-- DEBUG [205120.741    stream] -- _stream_init_loop: stream->run->stop=0
================================================================================
-- INFO  [205120.741      http] -- Starting HTTP eventloop ...
-- DEBUG [205120.758      http] -- HTTP: Updating exposed frame (online=0) ...
-- VERB  [205120.758      http] -- HTTP: Exposed frame: online=0, exp_time=0.000000
-- INFO  [205120.796    stream] -- Device fd=8 opened
-- DEBUG [205120.796    stream] -- Querying device capabilities ...
-- INFO  [205120.796    stream] -- Using input channel: 0
-- DEBUG [205120.796    stream] -- Using TV standard: DEFAULT
-- DEBUG [205120.796    stream] -- Probing device format=MJPEG, stride=3840, resolution=1920x1080 ...
-- INFO  [205120.829    stream] -- Using resolution: 1920x1080
-- INFO  [205120.829    stream] -- Using format: MJPEG
-- DEBUG [205120.829    stream] -- Querying HW FPS ...
-- INFO  [205121.154    stream] -- Using HW FPS: 30
-- ERROR [205121.154    stream] -- Device doesn't support setting of HW encoding quality parameters
-- INFO  [205121.154    stream] -- Using IO method: MMAP
-- DEBUG [205121.154    stream] -- Requesting 5 device buffers for MMAP ...
-- INFO  [205121.155    stream] -- Requested 5 device buffers, got 5
-- DEBUG [205121.155    stream] -- Allocating device buffers ...
-- DEBUG [205121.156    stream] -- Calling us_xioctl(VIDIOC_QUERYBUF) for device buffer=0 ...
-- DEBUG [205121.156    stream] -- Mapping device buffer=0 ...
-- DEBUG [205121.156    stream] -- Calling us_xioctl(VIDIOC_QUERYBUF) for device buffer=1 ...
-- DEBUG [205121.156    stream] -- Mapping device buffer=1 ...
-- DEBUG [205121.156    stream] -- Calling us_xioctl(VIDIOC_QUERYBUF) for device buffer=2 ...
-- DEBUG [205121.156    stream] -- Mapping device buffer=2 ...
-- DEBUG [205121.156    stream] -- Calling us_xioctl(VIDIOC_QUERYBUF) for device buffer=3 ...
-- DEBUG [205121.156    stream] -- Mapping device buffer=3 ...
-- DEBUG [205121.156    stream] -- Calling us_xioctl(VIDIOC_QUERYBUF) for device buffer=4 ...
-- DEBUG [205121.156    stream] -- Mapping device buffer=4 ...
-- DEBUG [205121.156    stream] -- Calling us_xioctl(VIDIOC_QBUF) for buffer=0 ...
-- DEBUG [205121.156    stream] -- Calling us_xioctl(VIDIOC_QBUF) for buffer=1 ...
-- DEBUG [205121.156    stream] -- Calling us_xioctl(VIDIOC_QBUF) for buffer=2 ...
-- DEBUG [205121.156    stream] -- Calling us_xioctl(VIDIOC_QBUF) for buffer=3 ...
-- DEBUG [205121.156    stream] -- Calling us_xioctl(VIDIOC_QBUF) for buffer=4 ...
-- DEBUG [205121.157    stream] -- Device fd=8 initialized
-- DEBUG [205121.157    stream] -- Starting device capturing ...
-- INFO  [205121.243    stream] -- Capturing started
-- INFO  [205121.243    stream] -- Switching to HW encoder: the input is (M)JPEG ...
-- INFO  [205121.243    stream] -- Using JPEG quality: encoder default
-- INFO  [205121.243    stream] -- Creating pool JPEG with 1 workers ...
-- INFO  [205121.243    stream] -- Capturing ...
--------------------------------------------------------------------------------
-- DEBUG [205121.243    stream] -- Waiting for worker ...
-- DEBUG [205121.243    stream] -- Calling select() on video device ...
-- DEBUG [205121.243      jw-0] -- Hello! I am a worker jw-0 ^_^
-- DEBUG [205121.243      jw-0] -- Worker jw-0 waiting for a new job ...
-- DEBUG [205121.775      http] -- HTTP: Repeating exposed ...
-- DEBUG [205122.243    stream] -- Device select() --> 0
-- ERROR [205122.243    stream] -- Mainloop select() error: Inappropriate ioctl for device
-- INFO  [205122.243    stream] -- Destroying workers pool JPEG ...
-- DEBUG [205122.243      jw-0] -- Bye-bye (worker jw-0)
-- DEBUG [205122.243    stream] -- Stopping device capturing ...
-- INFO  [205122.330    stream] -- Capturing stopped
-- DEBUG [205122.330    stream] -- Releasing device buffers ...
-- DEBUG [205122.330    stream] -- Closing device ...
-- INFO  [205122.332    stream] -- Device fd=8 closed
-- DEBUG [205122.332    stream] -- _stream_init_loop: stream->run->stop=0
```

It seems to be sending a single frame and boot looping. I'm running it on a Ryzen 3600 with MPG B550 mobo. So it shouldn't be the hardware.

Kinda lost here, and I'd appreciate any support.

mdevaev, 2023-02-25:
Hi. I'm sorry, but I don't have such a camera to figure out what exactly is wrong here. Are you using raspivid with mjpg-streamer or input_uvc?

gothrek22, 2023-07-11:
Oh, sorry mate, I forgot I turned off my notifications on github. I'm 70% sure that the issue is the USB Host Controller that's being used here, and it's running out of memory. I've managed to tune the settings to make it work, but they are suboptimal.

Here's the full config as a systemctl service.
```[Unit]
Description=mjpeg streamer for prusa arducam

[Service]
Type=simple
ExecStart=/bin/bash -c '/usr/bin/ustreamer --host 0.0.0.0 -r 1280x1024 -f 10  -m MJPEG --persistent -d $$(readlink -f /dev/prusacam)'

[Install]
WantedBy=default.target```

mdevaev, 2023-07-11:
Well, if it works - it works. I've seen similar problems before due to a buggy driver or a lack of USB 2.0 bandwidth. Glad it's solved.

closed 2023-07-11 by mdevaev