pikvm/ustreamer #198: Error -- Can't start capturing: Invalid argument

tecnocris, 2023-01-15:
For reference, when I use libcamera-hello on this pi, it runs just fine. So the camera works. Using the latest stock version of pios x64 lite. 

```
/pi@raspberrypi:~ $ v4l2-ctl --list-devices
bcm2835-codec-decode (platform:bcm2835-codec):
        /dev/video10
        /dev/video11
        /dev/video12
        /dev/video18
        /dev/video31
        /dev/media4

bcm2835-isp (platform:bcm2835-isp):
        /dev/video13
        /dev/video14
        /dev/video15
        /dev/video16
        /dev/video20
        /dev/video21
        /dev/video22
        /dev/video23
        /dev/media0
        /dev/media1

unicam (platform:fe801000.csi):
        /dev/video0
        /dev/media3

rpivid (platform:rpivid):
        /dev/video19
        /dev/media2
```

```
pi@raspberrypi:~/ustreamer $ ./ustreamer
-- INFO  [368.676      main] -- Starting PiKVM uStreamer 5.36 ...
-- INFO  [368.676      main] -- Using internal blank placeholder
-- INFO  [368.677      main] -- Listening HTTP on [127.0.0.1]:8080
-- INFO  [368.678    stream] -- Using V4L2 device: /dev/video0
-- INFO  [368.678    stream] -- Using desired FPS: 0
-- INFO  [368.678      http] -- Starting HTTP eventloop ...
================================================================================
-- INFO  [368.679    stream] -- Device fd=8 opened
-- INFO  [368.680    stream] -- Using input channel: 0
-- INFO  [368.680    stream] -- Using resolution: 640x480
-- INFO  [368.680    stream] -- Using format: YUYV
-- INFO  [368.680    stream] -- Querying HW FPS changing is not supported
-- INFO  [368.680    stream] -- Using IO method: MMAP
-- INFO  [368.691    stream] -- Requested 5 device buffers, got 5
-- ERROR [368.692    stream] -- Can't start capturing: Invalid argument
-- INFO  [368.694    stream] -- Device fd=8 closed
-- INFO  [368.694    stream] -- Sleeping 1 seconds before new stream init ...
```

masbando, 2023-01-16:
Same problem with the new Raspberry PI Camera Module 3 Wide with IMX708. Tested on Raspberry PI 3B and 4 with newest Raspberry PI OS 32bit and 64bit.

Hely0n, 2023-01-31:
exactly same here...

mdevaev, 2023-06-08:
I've got a camera and experimented with it. Too much work will be needed for native support, but Raspberry offers "libcamerify", which works fine with uStreamer. On Raspberry OS, install `libcamera-tools` and run as follows: `libcamerify ./ustreamer -r 1920x1080 --encoder=m2m-image`

closed 2023-06-08 by mdevaev