pikvm/ustreamer #203: Unsupported format=H264 (fourcc)

ssssmileeee, 2023-02-12:
I trying to make stream from my camera Creality CRCC-S7 and crowsnest libriary on Armbian 22.11.1 Bullseye with Linux 5.19.17-meson64, but ustreamer can't read H264 format from camera and writes to log:

> Could not obtain the requested format=YUYV;
> Unsupported format=H264 (fourcc)
> Device fd=8 closed
> Sleeping 1 seconds before new stream init ...

I tried all available formats (YUYV, UYVY, RGB565, RGB24, MJPEG, JPEG) but no effect.
In log I see that camera supports only two generic formats: H264 and HEVC, but I can't select any of this.
full log here: https://pastebin.com/HcRfun9Y

Guide me please if there is a solution for similar cases

mdevaev, 2023-02-12:
Hello. Ustreamer doesn't support reading h264 from camera at all. Only raw (yuyv/uyvy) or (m)jpeg.

SokolovJek, 2023-02-20:
And I thought it would support https://github.com/pikvm/ustreamer/blob/master/docs/h264.md?

mdevaev, 2023-02-21:
It is encoding, not capturing h264.

mdevaev, 2023-02-25:
Unfortunately there is nothing else we can do here at this stage. Ustreamer is not designed for such use. Sorry.

closed 2023-02-25 by mdevaev