pikvm/ustreamer #205: Question:  How many FPS should I expect on a Pi

mudmin, 2023-02-20:
Sorry to fill out an issue, but I've tried a Pi Zero, 3b, and 4b (the latter 2 connected by ethernet).  Plugging one of the typical Amazon HDMI to USB converters (which works great for capture in general), I'm getting 5-10 FPS of 1080p when viewing from another computer (either using VLC or the web interface.  

Does this seem about right for a PC looking at a Pi on the same switch?   Again, I apologize for the tech support question, but the description of this uses terms like "very fast" and I don't know if I should keep pressing on looking for better performance, or if my capture device is not ideal or where to even start looking.  

My default config has been 
./ustreamer --device=/dev/video0 --host=0.0.0.0 --port=8080  --workers=5 --persistent --drop-same-frames=30 --resolution=1920x1080

mdevaev, 2023-02-25:
Hello. You should have 20-30 FPS. Try `-m jpeg` or `-m mjpeg` option.

mudmin, 2023-02-27:
Ok. That's really helpful I think I had too high of expectations. I was thinking it would be 30 minimum.   I did get it into the 20fps range, but the motion blur is pretty bad. I will keep tinkering.  Thanks for your time.

mdevaev, 2023-02-27:
You're welcome.

closed 2023-02-27 by mdevaev