pikvm/ustreamer #206: Problem with setting up ustremer and janus for webrtc.

SokolovJek, 2023-02-20:
Good evening! Problem with setting up ustremer and janus for webrtc. I set up everything as in the instructions(https://github.com/pikvm/ustreamer/blob/master/docs/h264.md), but ustremer gives an error "**H264: Can't open encoder device: No such file or directory**":

Error for start ustreamer:
```
-- INFO  [1577.036      main] -- Starting PiKVM uStreamer 5.37 ...
-- INFO  [1577.036      main] -- Using internal blank placeholder
-- INFO  [1577.036      main] -- Using H264-sink: demo::ustreamer::h264
-- INFO  [1577.037      main] -- Listening HTTP on [0.0.0.0]:8080
-- INFO  [1577.038      http] -- Starting HTTP eventloop ...
-- INFO  [1577.038    stream] -- Using V4L2 device: /dev/video4
-- INFO  [1577.038    stream] -- Using desired FPS: 30
-- INFO  [1577.038    stream] -- H264: Initializing encoder ...
-- INFO  [1577.052    stream] -- H264: Configuring encoder: DMA=0 ...
-- ERROR [1577.052    stream] -- H264: Can't open encoder device: No such file or directory
-- ERROR [1577.052    stream] -- H264: Encoder destroyed due an error (prepare)
================================================================================
-- INFO  [1577.113    stream] -- Device fd=9 opened
-- INFO  [1577.114    stream] -- Using input channel: 0
-- INFO  [1577.149    stream] -- Using resolution: 1920x1080
-- INFO  [1577.149    stream] -- Using format: MJPEG
-- INFO  [1577.480    stream] -- Using HW FPS: 30
-- ERROR [1577.480    stream] -- Device doesn't support setting of HW encoding quality parameters
-- INFO  [1577.480    stream] -- Using IO method: MMAP
-- INFO  [1577.499    stream] -- Requested 5 device buffers, got 5
-- INFO  [1577.564    stream] -- Capturing started
-- INFO  [1577.564    stream] -- Switching to HW encoder: the input is (M)JPEG ...
-- INFO  [1577.565    stream] -- Using JPEG quality: encoder default
-- INFO  [1577.565    stream] -- Creating pool JPEG with 1 workers ...
-- INFO  [1577.565    stream] -- Capturing ...
-- INFO  [1577.830    stream] -- H264: Configuring encoder: DMA=0 ...
-- ERROR [1577.831    stream] -- H264: Can't open encoder device: No such file or directory
-- ERROR [1577.831    stream] -- H264: Encoder destroyed due an error (prepare)
```

**Сommand to run ustreamer** 

```
/usr/local/bin/ustreamer --log-level 0 --device=/dev/video4 --host=0.0.0.0 --port=8080 --resolution 1920x1080 --desired-fps=30 --format=MJPEG --h264-sink-rm --h264-sink-mode=660 --h264-sink=demo::ustreamer::h264
```

**Formats usb camera:(v4l2-ctl -d /dev/video4 --list-formats-ext)**
```
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'MJPG' (Motion-JPEG, compressed)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x680
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 800x600
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1024x768
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1280x1024
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
        [1]: 'YUYV' (YUYV 4:2:2)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.167s (6.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 800x600
                        Interval: Discrete 0.048s (21.000 fps)
                Size: Discrete 1024x768
                        Interval: Discrete 0.167s (6.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.111s (9.000 fps)
                Size: Discrete 1280x1024
                        Interval: Discrete 0.167s (6.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
```
I have been using ustreamer for a long time, I like it, thank you! But there was a need to set up streaming video.Can you suggest what is the problem? Thank you!

SokolovJek, 2023-02-21:
I so predict that my camera not support encoding h264. How correct is my guess? And if so, can you advise me how to set up a stream video for mjpeg?

mdevaev, 2023-02-21:
H264 is supported only on raspberry pi. Just exclude h264 options from CLI and mjpeg will work.

closed 2023-02-25 by mdevaev