pikvm/ustreamer #210: Why use libjpeg9-dev instead of libjpeg62-turbo on Raspberry OS?

yrjyrj123, 2023-04-02:
In README, "[Preconditions](https://github.com/pikvm/ustreamer#preconditions)" provides some dependencies on different operating systems.

Other systems use libjpeg-turbo directly or indirectly, but Raspberry Pi uses libjpeg9-dev instead.

Libjpeg-turbo can greatly accelerate the encoding and decoding speed of JPEG.

On the Raspberry Pi 4B, I used libjpeg62-turbo instead of libjpeg9-dev and achieved about 2 times speed improvement.

Is there a specific reason for using libjpeg9-dev in the README?

Thx

mdevaev, 2023-04-02:
I don't remember, you can try another library. It seems there were some compatibility issues, perhaps they are no longer there.

yrjyrj123, 2023-04-10:
After using it continuously for a week, I think it is stable.

Here is my environment information:

```
Package: libjpeg62-turbo-dev
Source: libjpeg-turbo
Version: 1:2.0.6-4
Architecture: armhf
Maintainer: Ondřej Surý <ondrej@debian.org>
Installed-Size: 545
Depends: libc-dev, libjpeg62-turbo (= 1:2.0.6-4)
Conflicts: libjpeg62-dev, libjpeg7-dev, libjpeg8-dev, libjpeg9-dev
Replaces: libjpeg62-dev, libjpeg7-dev, libjpeg8-dev, libjpeg9-dev
Provides: libjpeg-dev
Multi-Arch: same
Homepage: https://www.libjpeg-turbo.org/
Priority: optional
Section: libdevel
Filename: pool/main/libj/libjpeg-turbo/libjpeg62-turbo-dev_2.0.6-4_armhf.deb
Size: 240348
SHA256: bcb8383f987bc73aa66ac125b2cf239b0e4793044b963dae5aa5dfae065bd13a
SHA1: 3a1ac4e6fcde19023c0e781e8cad26192515fddf
MD5sum: 214a5c93c235d56ce86fcd96c42243cd
Description: Development files for the libjpeg-turbo JPEG library
 The libjpeg-turbo JPEG library is a library for handling JPEG files.
 .
 libjpeg-turbo is a JPEG image codec that uses SIMD instructions (MMX,
 SSE2, NEON) to accelerate baseline JPEG compression and decompression
 on x86, x86-64, and ARM systems.  The libjpeg-turbo JPEG library is
 an API/ABI compatible with the IJG JPEG library.
 .
 This package contains the static library, headers and documentation.
Description-md5: 5d090a630808f56c955fbfbe50d5c165
```

```
Operating System: Raspbian GNU/Linux 11 (bullseye)
Kernel: Linux 5.15.84-v7l+
Architecture: arm
```

mdevaev, 2023-11-15:
I've updated README, thank you.

closed 2023-11-15 by mdevaev