pikvm/ustreamer #211: OpenWRT UCI and LuCI support

relaxibus, 2023-04-22:
Hi, are you planning to add support for OpenWRT so we can create a new tab in LuCI and configure uStreamer?
Do you a,so support CSI interfaces?
GStreamer support for hardware encoding support? I’m working onan iMX8MP.

mdevaev, 2023-04-22:
Hello. I wrote OpenWRT package Makefile some times ago: https://github.com/pikvm/ustreamer/tree/master/pkg/openwrt.
It needs to be updated, it is a good point to start. Patches welcome ;)

I don't plan to add GStreamer or CSI, because uStreamer was created for one niche - V4L2 capture devices - and is not designed for something else. There is no technical possibility to add something else without rewriting half of the code, and unfortunately I don't have time to do it :/

relaxibus, 2023-04-23:
Hi, thanks!

closed 2023-04-23 by mdevaev