pikvm/ustreamer #213: ustremer and janus for webrtc Logitech c920

bliteknight, 2023-05-04:
Followed the steps here
https://github.com/pikvm/ustreamer/blob/master/docs/h264.md

I am trying to set this up for my logitech C920 webcam that supports h.264

When I run 
```
./ustreamer --log-level 0 --device=/dev/video1 --host=0.0.0.0 --port=8085 \
 --resolution=1280x720 --h264-sink=demo::ustreamer::h264 --h264-sink-mode=660 --h264-sink-rm
```
I get this output
```
-- INFO  [83748.000      main] -- Starting PiKVM uStreamer 5.38 ...
-- INFO  [83748.000      main] -- Using internal blank placeholder
-- INFO  [83748.000      main] -- Using H264-sink: demo::ustreamer::h264
-- INFO  [83748.001      main] -- Listening HTTP on [0.0.0.0]:8085
-- INFO  [83748.001    stream] -- Using V4L2 device: /dev/video1
-- INFO  [83748.001    stream] -- Using desired FPS: 0
-- INFO  [83748.001    stream] -- H264: Initializing encoder ...
-- INFO  [83748.001      http] -- Starting HTTP eventloop ...
-- INFO  [83748.008    stream] -- H264: Configuring encoder: DMA=0 ...
-- ERROR [83748.008    stream] -- H264: Can't open encoder device: No such file or directory
-- ERROR [83748.008    stream] -- H264: Encoder destroyed due an error (prepare)
================================================================================
-- INFO  [83748.075    stream] -- Device fd=9 opened
-- INFO  [83748.076    stream] -- Using input channel: 0
-- INFO  [83748.076    stream] -- Using resolution: 1280x720
-- INFO  [83748.076    stream] -- Using format: YUYV
-- INFO  [83748.077    stream] -- Using HW FPS: 0 -> 10 (coerced)
-- INFO  [83748.077    stream] -- Using IO method: MMAP
-- INFO  [83748.082    stream] -- Requested 5 device buffers, got 5
-- INFO  [83748.239    stream] -- Capturing started
-- INFO  [83748.240    stream] -- Using JPEG quality: 80%
-- INFO  [83748.240    stream] -- Creating pool JPEG with 4 workers ...
-- INFO  [83748.241    stream] -- Capturing ...
-- INFO  [83748.492    stream] -- H264: Configuring encoder: DMA=1 ...
-- ERROR [83748.492    stream] -- H264: Can't open encoder device: No such file or directory
-- ERROR [83748.492    stream] -- H264: Encoder destroyed due an error (prepare)
```

When I check
v4l2-ctl -d /dev/video1 --list-formats-ex
I get this
```
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'YUYV' (YUYV 4:2:2)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.042s (24.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.133s (7.500 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 160x90
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Dis...
                Size: Discrete 1920x1080
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 2304x1296
                        Interval: Discrete 0.500s (2.000 fps)
                Size: Discrete 2304x1536
                        Interval: Discrete 0.500s (2.000 fps)
        [1]: 'H264' (H.264, compressed)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.042s (24.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
[... 113 lines cut ...]
janus.js:1380 {janus: 'message', body: {…}, transaction: 'QpdQk5NecppK', jsep: {…}}
janus.js:772 Got a plugin event on session 2466482022428012
janus.js:773 {janus: 'event', session_id: 2466482022428012, transaction: 'QpdQk5NecppK', sender: 6758963070362546, plugindata: {…}}
janus.js:784   -- Event is coming from 6758963070362546 (janus.plugin.recordplay)
janus.js:786 {ustreamer: 'event', result: {…}}
janus.js:799 Notifying application...
janus.js:624 Got an ack on session 2466482022428012
janus.js:625 {janus: 'ack', session_id: 2466482022428012, transaction: 'QpdQk5NecppK'}
janus.js:1385 Message sent!
janus.js:1386 {janus: 'ack', session_id: 2466482022428012, transaction: 'QpdQk5NecppK'}
janus.js:1474 Sending trickle candidate (handle=6758963070362546):
janus.js:1475 {janus: 'trickle', candidate: {…}, transaction: '46aEdFLglDBE'}
janus.js:1845 End of candidates.
janus.js:1474 Sending trickle candidate (handle=6758963070362546):
janus.js:1475 {janus: 'trickle', candidate: {…}, transaction: '5vrCcM69raFk'}
janus.js:624 Got an ack on session 2466482022428012
janus.js:625 {janus: 'ack', session_id: 2466482022428012, transaction: '46aEdFLglDBE'}
janus.js:624 Got an ack on session 2466482022428012
janus.js:625 {janus: 'ack', session_id: 2466482022428012, transaction: '5vrCcM69raFk'}
janus.js:682 Got a webrtcup event on session 2466482022428012
janus.js:683 {janus: 'webrtcup', session_id: 2466482022428012, sender: 6758963070362546}
janus.js:1931 Remote track muted: Event {isTrusted: true, type: 'mute', target: MediaStreamTrack, currentTarget: MediaStreamTrack, eventPhase: 2, …}
```

Any help would be appreciated.

bliteknight, 2023-05-04:
UPDATE: So I changed my ustreamer execution to this:
```
./ustreamer --log-level 0 --device=/dev/video1 --host=0.0.0.0 --port=8085  \
--resolution=1280x720 --sink=demo::ustreamer::jpeg --sink-mode=770 --sink-rm
```
So no more h.264 errors

But on the Janus end I am now seeing this error:
```
== ustreamer/video     -- Memsink opened; reading frames ...
[ERR] [src/memsinkfd.c:us_memsink_fd_get_frame:61] == ustreamer/video     -- Got non-H264 frame from memsink
== ustreamer/video     -- Memsink closed
```

Does janus and ustreamer only support h264?

mdevaev, 2023-05-05:
Hello. Janus + ustreamer can only work with H264, but ustreamer can't use raw H264 stream as a video source. It was developed for video devices producing UYVY and I have no plans now to add support for anything else.

bliteknight, 2023-05-05:
ok thanks, I will try something else.

closed 2023-05-05 by bliteknight