pikvm/ustreamer #214: Resource temporarily unavailable (reading,eof)

UchihaYuki, 2023-05-18:
```
$ v4l2-ctl -d /dev/video0 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'MJPG' (Motion-JPEG, compressed)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
```
You can see that it supports 30 fps. but when I use 30 fps to run ustreamer, I will get `Resource temporarily unavailable (reading,eof)` after 1 second (sometimes several seconds or longer). But when I changed the fps to 5, I will get the same error after several minutes.

Following is the log
```
biqu@BTT-CB1:~/crowsnest/bin/ustreamer$ ./ustreamer -p 8080 -d /dev/video0 --device-timeout=2 -m MJPEG --encoder=HW -r 1920x1080 -f 30 --allow-origin=* --static /home/biqu/crowsnest/ustreamer-www --host=0.0.0.0
-- INFO  [1794.702      main] -- Starting PiKVM uStreamer 5.38 ...
-- INFO  [1794.702      main] -- Using internal blank placeholder
-- INFO  [1794.702      main] -- Enabling HTTP file server: /home/biqu/crowsnest/ustreamer-www
-- INFO  [1794.703      main] -- Listening HTTP on [0.0.0.0]:8080
-- INFO  [1794.703    stream] -- Using V4L2 device: /dev/video0
-- INFO  [1794.704    stream] -- Using desired FPS: 30
-- INFO  [1794.704      http] -- Starting HTTP eventloop ...
================================================================================
-- INFO  [1794.770    stream] -- Device fd=8 opened
-- INFO  [1794.770    stream] -- Using input channel: 0
-- INFO  [1794.781    stream] -- Using resolution: 1920x1080
-- INFO  [1794.781    stream] -- Using format: MJPEG
-- INFO  [1794.792    stream] -- Using HW FPS: 30
-- ERROR [1794.792    stream] -- Device doesn't support setting of HW encoding quality parameters
-- INFO  [1794.792    stream] -- Using IO method: MMAP
-- INFO  [1794.798    stream] -- Requested 5 device buffers, got 5
-- INFO  [1795.124    stream] -- Capturing started
-- INFO  [1795.124    stream] -- Using JPEG quality: encoder default
-- INFO  [1795.125    stream] -- Creating pool JPEG with 1 workers ...
-- INFO  [1795.125    stream] -- Capturing ...
-- INFO  [1797.841      http] -- HTTP: NEW client (now=1): [192.168.31.123]:58721, id=4b0b162b72916844
-- INFO  [1800.202      http] -- HTTP: DEL client (now=0): [192.168.31.123]:58721, id=4b0b162b72916844, Resource temporarily unavailable (reading,eof)
-- INFO  [1800.210      http] -- HTTP: NEW client (now=1): [192.168.31.123]:58722, id=c8683bdb942bf712
-- INFO  [1814.402      http] -- HTTP: DEL client (now=0): [192.168.31.123]:58722, id=c8683bdb942bf712, Resource temporarily unavailable (reading,eof)
-- INFO  [1814.433      http] -- HTTP: NEW client (now=1): [192.168.31.123]:58726, id=caa4d5e5f1c04ca1
```

Following is what I get from state
```
{"ok": true, "result": { "instance_id": "", "encoder": {"type": "HW", "quality": 0}, "source": {"resolution": {"width": 1920, "height": 1080}, "online": true, "desired_fps": 30, "captured_fps": 14}, "stream": {"queued_fps": 14, "clients": 1, "clients_stat": {"42df3080072e7fed": {"fps": 14, "extra_headers": false, "advance_headers": false, "dual_final_frames": false, "zero_data": false, "key": "0"}}}}}
```

mdevaev, 2023-05-18:
Hello. This is a problem from the client side. Which browser do you use?

UchihaYuki, 2023-05-18:
> Hello. This is a problem on the client side. Which browser do you use?

You are right. I just tested it. It works on Firefox, but doesn't work on Edge and Chrome.

There is another project (https://github.com/mainsail-crew/crowsnest) which uses this project. There will be no picture in their plugin after several seconds or minutes, even when I use Firefox. I guess the bug is on their side. I will close the issue. Thank you!

closed 2023-05-18 by UchihaYuki

UchihaYuki, 2023-05-18:
It seems, the problem is not on the browser side.

I just come across this issue https://github.com/pikvm/ustreamer/issues/110. My video stream is also not stable.

UchihaYuki, 2023-05-19:
The error is caused by a broken USB wire.

mdevaev, 2023-05-19:
:ok_hand: