pikvm/ustreamer #216: Mainloop select() error: Inappropriate ioctl for device

Drzet, 2023-05-25:
This happens with every resolution above 1280x720. RasPicam v2

***@octopi:~ $ ustreamer --host :: -m jpeg --device-timeout=5 --buffers=3 -r 3280x2464
-- INFO  [1258.291      main] -- Using internal blank placeholder
-- INFO  [1258.292      main] -- Listening HTTP on [::]:8080
-- INFO  [1258.292    stream] -- Using V4L2 device: /dev/video0
-- INFO  [1258.292    stream] -- Using desired FPS: 0
-- INFO  [1258.292      http] -- Starting HTTP eventloop ...
================================================================================
-- INFO  [1258.294    stream] -- Device fd=8 opened
-- INFO  [1258.294    stream] -- Using input channel: 0
-- INFO  [1258.297    stream] -- Using resolution: 3280x2464
-- INFO  [1258.297    stream] -- Using pixelformat: JPEG
-- INFO  [1258.298    stream] -- Using HW FPS: 0 -> 90 (coerced)
-- ERROR [1258.298    stream] -- Device does not support setting of HW encoding quality parameters
-- INFO  [1258.298    stream] -- Using IO method: MMAP
-- INFO  [1258.332    stream] -- Requested 3 device buffers, got 3
-- INFO  [1258.837    stream] -- Capturing started
-- INFO  [1258.837    stream] -- Switching to HW encoder: the input is (M)JPEG ...
-- INFO  [1258.838    stream] -- Using JPEG quality: encoder default
-- INFO  [1258.838    stream] -- Creating pool JPEG with 1 workers ...
-- INFO  [1258.838    stream] -- Capturing ...
**-- ERROR [1263.843    stream] -- Mainloop select() error: Inappropriate ioctl for device**
-- INFO  [1263.843    stream] -- Destroying workers pool JPEG ...
-- INFO  [1263.857    stream] -- Capturing stopped
-- INFO  [1263.869    stream] -- Device fd=8 closed

mdevaev, 2023-07-10:
Hi. I missed it. Probably it doesn't support jpeg above this resolution? Try something like this: https://github.com/pikvm/ustreamer#raspberry-pi-camera-example

tuxmobil, 2023-07-10:
I just ran into a similar issue using the RaspiCam v3. 

Bullseye 64-Bit on Pi4 with ustreamer freshly built. I tried different resolutions, with no change in behaviour.

```
$ libcamerify ./ustreamer --host :: --encoder m2m-image -r 640x480 --debug
-- INFO  [1463.545      main] -- Starting PiKVM uStreamer 5.41 ...
-- INFO  [1463.545      main] -- Using internal blank placeholder
-- DEBUG [1463.546      main] -- Installing SIGINT handler ...
-- DEBUG [1463.546      main] -- Installing SIGTERM handler ...
-- DEBUG [1463.546      main] -- Ignoring SIGPIPE ...
-- DEBUG [1463.546      main] -- Binding HTTP to [::]:8080 ...
-- INFO  [1463.547      main] -- Listening HTTP on [::]:8080
-- INFO  [1463.547    stream] -- Using V4L2 device: /dev/video0
-- INFO  [1463.547    stream] -- Using desired FPS: 0
-- INFO  [1463.547      http] -- Starting HTTP eventloop ...
-- DEBUG [1463.548    stream] -- _stream_init_loop: stream->run->stop=0
================================================================================
[0:24:23.538620735] [6003]  INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70
-- DEBUG [1463.565      http] -- HTTP: Updating exposed frame (online=0) ...
-- VERB  [1463.565      http] -- HTTP: Exposed frame: online=0, exp_time=0.000000
[0:24:23.679357672] [6005]  INFO RPI raspberrypi.cpp:1476 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media3 and ISP device /dev/media1
-- INFO  [1463.696    stream] -- Device fd=8 opened
-- DEBUG [1463.696    stream] -- Querying device capabilities ...
-- ERROR [1463.696    stream] -- Can't query device capabilities: Inappropriate ioctl for device
-- DEBUG [1463.696    stream] -- Closing device ...
-- INFO  [1463.696    stream] -- Device fd=8 closed
-- INFO  [1463.696    stream] -- Sleeping 1 seconds before new stream init ...
-- DEBUG [1464.571      http] -- HTTP: Repeating exposed ...
================================================================================
-- INFO  [1464.698    stream] -- Device fd=8 opened
-- DEBUG [1464.698    stream] -- Querying device capabilities ...
-- ERROR [1464.698    stream] -- Can't query device capabilities: Inappropriate ioctl for device
-- DEBUG [1464.698    stream] -- Closing device ...
-- INFO  [1464.698    stream] -- Device fd=8 closed
-- INFO  [1464.698    stream] -- Sleeping 1 seconds before new stream init ...
-- DEBUG [1464.700      http] -- HTTP: Updating exposed frame (online=0) ...
-- VERB  [1464.700      http] -- HTTP: Exposed frame: online=0, exp_time=0.000000
================================================================================
```

```
$ libcamera-hello --list-cameras
Available cameras
-----------------
0 : imx708_wide_noir [4608x2592] (/base/soc/i2c0mux/i2c@1/imx708@1a)
    Modes: 'SRGGB10_CSI2P' : 1536x864 [120.13 fps - (768, 432)/3072x1728 crop]
                             2304x1296 [56.03 fps - (0, 0)/4608x2592 crop]
                             4608x2592 [14.35 fps - (0, 0)/4608x2592 crop]
```

mdevaev, 2023-07-10:
Hm, this is IMX708 chip. Maybe unsupported? As I remember libcamera-hello or some other libcamera app should capture stream to the file. Try it without ustreamer, will it work at all.

tuxmobil, 2023-07-10:
The camera does work in general. However I still struggle to find a streaming solution with acceptable frame rate (tried motion first - without motion detection, at least that was my intention - which was SLOW). Also had some success with mjpg-streamer, but still with huge latency and low framerate. That's why I wanted to try whether ustreamer would be better. I didn't expect that a camera module from Raspberry would be so difficult to get running with say 25-30 fps.

mdevaev, 2023-07-13:
Do you have any special settings for the camera? Like module params, etc? Also show the `/boot/config.txt` please.

tuxmobil, 2023-07-17:
None that I'm aware of.

[config.txt](https://github.com/pikvm/ustreamer/files/12074148/config.txt)

It's basically the current raspbian bullseye 64 Bit without any modifications.
As the camera was detected by default, I didn't change anything. However, as I read the comment about
the driver arguments to probably influence the capture speed, I also tried re-loading the driver with

`modprobe bcm2835-v4l2 max_video_width=4608 max_video_height=2592`

which didn't made any difference. (rmmod-ing it first)

tuxmobil, 2023-07-18:
> I don't know what's going on :/ Try to add `cma=128M` to `/boot/cmdline.txt` and `gpu_mem=128` to `/boot/config.txt`

No change. :-/

mdevaev, 2023-07-19:
I'm sorry, but I've run out of ideas here :(

uStreamer was intended to capture HDMI and USB in the first place, I didn't think it would be used with raspicam and I really don't know how it works.

mdevaev, 2023-08-07:
Sorry again.

closed 2023-08-07 by mdevaev