pikvm/ustreamer #217: Streaming stops randomly

thomassorensen2650, 2023-05-26:
I have an issue where the video stream from a USB webcam constantly stops . If I open the /stream endpoint in google chrome or Microsoft Edge, then the video runs for a few seconds and then it changes into a gray background. I can right click the image (gray background) and select “load image" to start the stream again, but it only runs a few seconds. I have tried with another camera with the same result.

The only thing that I can see in the logs are _**Resource temporarily unavailable (reading,eof)**_ when I reload the image or refresh the browser. 

Is there are good way to troubleshoot this issue further?

ustreamer v5.38.
Raspberry PI 4 CM4
ubuntu@demo:~$ uname -a
Linux demo 5.15.0-1029-raspi #31-Ubuntu SMP PREEMPT Sat Apr 22 12:26:40 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux

Logs:
```
ubuntu@FCLabControl01:~$ ustreamer --device /dev/video0 --format=MJPEG --resolution 1280x720 --desired-fps=5 --host=0.0.0.0 --port=8080
-- INFO  [3974.629      main] -- Starting PiKVM uStreamer 5.38 ...
-- INFO  [3974.629      main] -- Using internal blank placeholder
-- INFO  [3974.630      main] -- Listening HTTP on [0.0.0.0]:8080
-- INFO  [3974.630    stream] -- Using V4L2 device: /dev/video0
-- INFO  [3974.630    stream] -- Using desired FPS: 5
-- INFO  [3974.630      http] -- Starting HTTP eventloop ...
================================================================================
-- INFO  [3974.745    stream] -- Device fd=8 opened
-- INFO  [3974.745    stream] -- Using input channel: 0
-- INFO  [3974.748    stream] -- Using resolution: 1280x720
-- INFO  [3974.748    stream] -- Using format: MJPEG
-- INFO  [3974.751    stream] -- Using HW FPS: 5 -> 30 (coerced)
-- ERROR [3974.751    stream] -- Device doesn't support setting of HW encoding quality parameters
-- INFO  [3974.751    stream] -- Using IO method: MMAP
-- INFO  [3974.753    stream] -- Requested 5 device buffers, got 5
-- INFO  [3974.757    stream] -- Capturing started
-- INFO  [3974.757    stream] -- Switching to HW encoder: the input is (M)JPEG ...
-- INFO  [3974.757    stream] -- Using JPEG quality: encoder default
-- INFO  [3974.758    stream] -- Creating pool JPEG with 1 workers ...
-- INFO  [3974.758    stream] -- Capturing ...
-- INFO  [3977.965      http] -- HTTP: NEW client (now=1): [10.18.5.58]:65441, id=9af0c0df849b167
-- INFO  [3979.815      http] -- HTTP: DEL client (now=0): [10.18.5.58]:65441, id=9af0c0df849b167, Resource temporarily unavailable (reading,eof)
-- INFO  [3980.855      http] -- HTTP: NEW client (now=1): [10.18.5.58]:65442, id=d2764bdefc3654f5
-- INFO  [3981.406      http] -- HTTP: DEL client (now=0): [10.18.5.58]:65442, id=d2764bdefc3654f5, Resource temporarily unavailable (reading,eof)
-- INFO  [3981.590      http] -- HTTP: NEW client (now=1): [10.18.5.58]:65446, id=f27c42457f8f5b00
-- INFO  [3982.247      http] -- HTTP: DEL client (now=0): [10.18.5.58]:65446, id=f27c42457f8f5b00, Resource temporarily unavailable (reading,eof)
-- INFO  [3982.440      http] -- HTTP: NEW client (now=1): [10.18.5.58]:65448, id=f9ded20df543d626
-- INFO  [3982.951      http] -- HTTP: DEL client (now=0): [10.18.5.58]:65448, id=f9ded20df543d626, Resource temporarily unavailable (reading,eof)
-- INFO  [3983.134      http] -- HTTP: NEW client (now=1): [10.18.5.58]:65449, id=1aa324244f6669
-- INFO  [3983.654      http] -- HTTP: DEL client (now=0): [10.18.5.58]:65449, id=1aa324244f6669, Resource temporarily unavailable (reading,eof)
-- INFO  [3983.840      http] -- HTTP: NEW client (now=1): [10.18.5.58]:65450, id=1800e664c6f28d8c
-- INFO  [4313.669      http] -- HTTP: DEL client (now=0): [10.18.5.58]:65450, id=1800e664c6f28d8c, Resource temporarily unavailable (reading,eof)
-- INFO  [4314.130      http] -- HTTP: NEW client (now=1): [10.18.5.58]:54002, id=9ae9b0a08caf5f2f
```

mdevaev, 2023-05-26:
This is a problem on the browser side, it breaks the connection. Look in developer tools, try incognito mode without any extensions.

thomassorensen2650, 2023-05-31:
Thanks for the feedback. I have tried multiple different computers, and they all have the same issue, no errror messages anywhere, but it looks like it works perfectly in Firefox.  I will investigate further on the client side.

closed 2023-05-31 by thomassorensen2650